zwave-js/zwave-js #5155: zwave_js_notification Does not receive manual events

rpruden, 2022-10-09:
Using the latest version of HomeAssistant Core, and zwave JS. This problem has been occurring as long as I can remember. When locking/unlocking the door using home assistant, I will see events being triggered in the zwave_js_notification event. However, if I manually lock/unlock, or use the keypad to lock/unlock, it does not log any events.

My device is identified as a Kwikset 912, though that could be incorrect as I had to submit a previous issue to have this entity ID added into the Zwave JS database. Unfortunately the lock came with the house, so I can not confirm with 100% certainty as to which exact model it is. I will reply with a copy of a debug log during a re-interview of the device. I hope you can help me figure out what is wrong.

rpruden, 2022-10-09:
[zwavejs_log.txt](https://github.com/zwave-js/node-zwave-js/files/9740692/zwavejs_log.txt)

attached log.

blhoward2, 2022-10-17:
Have you tried taking the lock off the door? Ordinarily there is a model number, serial number, etc. on the inside of the lock.

Can you point me to a timestamp where you were operating the lock manually? Or run another debug-level log off while doing so?

AlCalzone, 2023-03-06:
I finally had a chance to look at the log. The interview looks normal, but afterwards no notifications are received.
Maybe healing the node to give it a new route to the controller could help. Other than that, I have no suggestions.