zwave-js/zwave-js #5157: Zooz Outdoor Motion Sensor ZSE29 - SupervisionCCReport fails - cannot change device configuration values

robertvorthman, 2022-10-12:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

The Zooz Outdoor Motion sensor, ZSE29, does not appear to support supervision command class.  Attempting to update any value, such as setting motion sensitivity from 10 to 9, fails and the value returns to the previous value.  I see an error about SupervisionCCReport in the logs when attempting to update values.

```
2022-10-12T01:17:32.939Z DRIVER « [Node 091] [REQ] [BridgeApplicationCommand]
│ RSSI: -71 dBm
└─[Security2CCMessageEncapsulation]
│ sequence number: 188
└─[SupervisionCCReport]
session id: 4
more updates follow: false
status: Fail
duration: 0s
2022-10-12T01:17:32.951Z CNTRLR [Node 091] [setValue] result of SET_VALUE API call for ConfigurationCCAPI: (Su
pervisionResult)
status: Fail
2022-10-12T01:17:32.952Z CNTRLR [Node 091] [setValue] not updating value
```

This issue seems similar to issue #5139 and pull #5008.

### Device information

Manufacturer: Zooz
Model name: ZSE29
Node ID in your network: 92

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: Z-Wave JS UI Add-on version: 1.1.0
`node-zwave-js` branch: 10.3.0
`zwave-js-ui` branch: 8.1.0

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

A few months ago I was able to update the values on my ZSE29.

### Attach Driver Logfile

See last 80 lines.  I attempt to update parameter 3 to a 9 but it reverts back to 10.
[zwavejs_current (1).log](https://github.com/zwave-js/node-zwave-js/files/9761096/zwavejs_current.1.log)

AlCalzone, 2022-10-12:
We should let Zooz know about this.

AlCalzone, 2022-10-12:
@robertvorthman Could you open a PR with your changes please?

robertvorthman, 2022-10-12:
I just submitted PR #5159 with the updated device config file.

I started an email support ticket with Zooz yesterday, assuming it was an isolated problem with my device.  When I found the root cause, I emailed them a link to this Github issue.  I will email them back and ask them to notify their engineering team.  Would the goal be for Zooz to update their firmware?

AlCalzone, 2022-10-12:
> Would the goal be for Zooz to update their firmware?

Yeah, exactly.

PR #5159 (2022-10-12): fix(config): disable Supervision for Zooz ZSE29
fixes: #5157

The ZSE29 fails to respond to supervised commands.  Copied the "compat" attribute from the neighboring zse19.json config file.  This solved the problem when loaded as a custom device config file on my Home Assistant server.
closed 2022-10-12 by AlCalzone