zwave-js/zwave-js #5158: [feat] Assign return route during device association

joshlawless, 2022-10-12:
**Is your feature request related to a problem? Please describe.**
I have struggled to create associations between my permanently powered devices (e.g., wall-mounted scene controllers) and my battery-powered devices (e.g., Springs Window Fashions battery powered cellular shades).  The association only ever seemed to work when the battery-powered device was already "awake" (e.g., in motion and constantly sending updated position reports, or in inclusion mode).  The secondary controllers that came with the shades functioned based on group association, but must have some secret sauce for "waking up" the shades when they send a command.

**Describe the solution you'd like**
I discovered this post: [https://drzwave.blog/2021/02/01/controlling-flirs-via-associations/comment-page-1/?unapproved=24900&moderation-hash=f64b22b07106e08d9a6d3af673985532#comment-24900](url) on the Dr ZWave bog about creating return routes for Frequently Listening Routing Slaves (FLiRS), which it appears the battery-powered shades are.  When I create an association using the Windows-based ZWave PC Controller, I can specify to assign a "return route" (apparently the equivalent of the SerialAPI command ZW_AssignReturnRoute( SourceNodeID, DestNodeID, callback) which is SerialAPI command 0x46 (see section 4.4.4 of [INS13954](https://www.silabs.com/documents/public/user-guides/INS13954-Instruction-Z-Wave-500-Series-Appl-Programmers-Guide-v6_81_0x.pdf)).)  Now my wall-mounted switch acts as a secondary controller for the shades using direct group association, and no problem with the device sleeping through the commands.

It would be great if this were a feature that could be implemented in to the ZWaveJS UI, so I can avoid the use of a Windows laptop and Zwave PC Controller to create this kind of association to a battery-powered FLiRS

AlCalzone, 2022-10-12:
1. Accepting PRS for the assign return route part, or I'll do it some time in 2023

2. About associations changed outside zwave-js: if we were to assume that everything can change, we'd have to do a lot more interviewing on startup, 99% of the time for no benefit.
We could think about re-querying associations before showing them, but this means you'd have to wake battery powered devices before and after editing associations.
I think a button to manually refresh them from the device would be a good middle ground.
This might need some work in the driver to "do the right thing"

I assume refreshing the groups currently just asks the driver again for the cached values, which essentially does nothing 🤷🏻‍♂️

darkbasic, 2022-12-11:
Being able to manually assign a return route will probably come in handy for https://github.com/zwave-js/node-zwave-js/issues/4477

closed 2023-03-09 by zwave-js-bot