zwave-js/zwave-js #5161: [Bug]: UZB Stick hangging

mw-nfi, 2022-10-14:


mw-nfi, 2022-10-14:
[zwavejs_2022-10-14.log](https://github.com/zwave-js/node-zwave-js/files/9787170/zwavejs_2022-10-14.log)
[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/9787172/zwavejs_current.log)
Here are the logs. This time it took a little longer. The thermostat now shows ES according to the UI but continues to send it merrily. But this happens with all devices.

jmgiaever, 2022-10-16:
Is this an UZB or an UZB 7?

mw-nfi, 2022-10-16:
UZB from zwave.me

https://z-wave.me/products/uzb/

freundliche Grüße
Niklas Fischbach

Niklas Fischbach | Software Engineer

Phone +49-2761-859858-43<tel:+49-2761-859858-43> | ***@***.******@***.***>

[cid:logoneu_bb270a0c-350e-4063-93f1-2037feddc936.png]          MEHRWERT SYSTEMHAUS GMBH
IN DER TRIFT 1
DE 57462 OLPE
https://mehrwert.net

Sitz der Gesellschaft: Olpe, AG Siegen HRB 7711 | Geschäftsführer: Marcus Schneider

jmgiaever, 2022-10-16:
500 series then.

hvorragend, 2022-10-21:
By the way: I am currently surprised that the blue LED on my ZME_UZB1 is permanently lit very often a day. I've had this for a few weeks. I don't know if that helps. But that came from some ZWaveJS version.

mw-nfi, 2022-11-08:
So I've a fresh installed docker instance. 
I have added 1 wall plug and get a network error. Here are the logs.
[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/9961418/zwave-js-ui-store.zip)

Is my stick broken?
Thanks!

mw-nfi, 2022-11-08:
2022-11-08T13:22:33.167Z DRIVER   version 10.3.0
2022-11-08T13:22:33.167Z DRIVER   
2022-11-08T13:22:33.167Z DRIVER   starting driver...
2022-11-08T13:22:33.167Z DRIVER   opening serial port /dev/zwave
2022-11-08T13:22:33.169Z DRIVER   serial port opened
2022-11-08T13:22:33.169Z SERIAL » [NAK]                                                                   (0x15)
2022-11-08T13:22:33.169Z DRIVER   loading configuration...
2022-11-08T13:22:33.170Z CONFIG   Using external configuration dir /usr/src/app/store/.config-db
2022-11-08T13:22:33.170Z CONFIG   version 10.3.0
2022-11-08T13:22:33.368Z SERIAL « [DISCARDED] invalid data 0xc01cd7000c0005000253efa7b75afcc0         (16 bytes)
2022-11-08T13:22:33.505Z DRIVER   beginning interview...
2022-11-08T13:22:33.505Z DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                                  1 registered
2022-11-08T13:22:33.505Z DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                                  1 registered
2022-11-08T13:22:33.505Z DRIVER   added request handler for ReplaceFailedNode (0x63)...
                                  1 registered
2022-11-08T13:22:33.505Z CNTRLR   querying controller IDs...
2022-11-08T13:22:33.507Z SERIAL » 0x01030020dc                                                         (5 bytes)
2022-11-08T13:22:33.508Z DRIVER » [REQ] [GetControllerId]
2022-11-08T13:22:34.509Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
2022-11-08T13:22:34.610Z SERIAL » 0x01030020dc                                                         (5 bytes)
2022-11-08T13:22:34.610Z DRIVER » [REQ] [GetControllerId]
2022-11-08T13:22:35.612Z CNTRLR   Failed to execute controller command after 2/3 attempts. Scheduling next try i
                                  n 1100 ms.
2022-11-08T13:22:36.713Z SERIAL » 0x01030020dc                                                         (5 bytes)
2022-11-08T13:22:36.713Z DRIVER » [REQ] [GetControllerId]
2022-11-08T13:22:37.715Z DRIVER   Failed to initialize the driver: ZWaveError: Timeout while waiting for an ACK 
                                  from the controller (ZW0200)
                                      at Driver.sendMessage (/usr/src/app/node_modules/zwave-js/src/lib/driver/D
                                  river.ts:3990:23)
                                      at ZWaveController.identify (/usr/src/app/node_modules/zwave-js/src/lib/co
                                  ntroller/Controller.ts:713:33)
                                      at Driver.initializeControllerAndNodes (/usr/src/app/node_modules/zwave-js
                                  /src/lib/driver/Driver.ts:1208:26)
                                      at Immediate.<anonymous> (/usr/src/app/node_modules/zwave-js/src/lib/drive
                                  r/Driver.ts:1015:16)
2022-11-08T13:22:37.715Z DRIVER   destroying driver instance...
2022-11-08T13:22:37.717Z DRIVER   driver instance destroyed
2022-11-08T13:22:37.720Z DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                                  ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                                    ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                                   ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                                  ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                                  ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
2022-11-08T13:22:37.720Z DRIVER   version 10.3.0
2022-11-08T13:22:37.720Z DRIVER   
2022-11-08T13:22:37.720Z DRIVER   starting driver...
2022-11-08T13:22:37.721Z DRIVER   opening serial port /dev/zwave
2022-11-08T13:22:37.722Z DRIVER   serial port opened
2022-11-08T13:22:37.723Z SERIAL » [NAK]                                                                   (0x15)
2022-11-08T13:22:37.723Z DRIVER   loading configuration...
2022-11-08T13:22:37.724Z CONFIG   Using external configuration dir /usr/src/app/store/.config-db
2022-11-08T13:22:37.724Z CONFIG   version 10.3.0
2022-11-08T13:22:38.065Z DRIVER   beginning interview...
2022-11-08T13:22:38.066Z DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                                  1 registered
2022-11-08T13:22:38.066Z DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                                  1 registered
2022-11-08T13:22:38.066Z DRIVER   added request handler for ReplaceFailedNode (0x63)...
                                  1 registered
2022-11-08T13:22:38.066Z CNTRLR   querying controller IDs...
2022-11-08T13:22:38.068Z SERIAL » 0x01030020dc                                                         (5 bytes)
2022-11-08T13:22:38.068Z DRIVER » [REQ] [GetControllerId]
2022-11-08T13:22:38.368Z SERIAL « [DISCARDED] invalid data 0xc01cd8000c0005000253efa7b759fcc0         (16 bytes)

robertsLando, 2022-11-08:
@AlCalzone any known bug?

AlCalzone, 2022-11-08:
No known issues. Unfortunately the UZB(1) is one of the sticks that cannot be automatically restarted. If it is hanging it needs to be done manually by re-plugging it.

mw-nfi, 2022-11-08:
Strangely enough, it works briefly on a reboot.
Why does the connection break? Do I then have to restart the container regularly?

mw-nfi, 2022-11-08:
Here is an other log entry from docker: 

2022-11-08 14:32:18.140 INFO Z-WAVE: Client closed
2022-11-08 14:32:18.142 INFO Z-WAVE: Connecting to /dev/zwave
Logging to file:
	/usr/src/app/store/logs/zwavejs_2022-11-08.log
2022-11-08 14:32:31.982 INFO Z-WAVE: Controller status: Driver: Failed to initialize the driver: ZWaveError: Timeout while waiting for a response from the controller (ZW0200)
    at Driver.sendMessage (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:3990:23)
    at ZWaveController.identify (/usr/src/app/node_modules/zwave-js/src/lib/controller/Controller.ts:713:33)
    at Driver.initializeControllerAndNodes (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:1208:26)
    at Immediate.<anonymous> (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:1015:16) (ZW0100)
2022-11-08 14:32:31.985 INFO Z-WAVE: Client closed
2022-11-08 14:32:31.987 INFO Z-WAVE: Connecting to /dev/zwave
Logging to file:
	/usr/src/app/store/logs/zwavejs_2022-11-08.log
2022-11-08 14:32:36.574 INFO Z-WAVE: Controller status: Driver: Failed to initialize the driver: ZWaveError: Timeout while waiting for an ACK from the controller (ZW0200)
    at Driver.sendMessage (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:3990:23)
    at ZWaveController.identify (/usr/src/app/node_modules/zwave-js/src/lib/controller/Controller.ts:713:33)
    at Driver.initializeControllerAndNodes (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:1208:26)
    at Immediate.<anonymous> (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:1015:16) (ZW0100)
2022-11-08 14:32:36.576 INFO Z-WAVE: Client closed
2022-11-08 14:32:36.578 INFO Z-WAVE: Connecting to /dev/zwave

mw-nfi, 2022-11-11:
What can I do now? It used to work without any problems. Should I try to downgrade the version? Which version would be best?

Thanks

AlCalzone, 2022-11-11:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) of the startup until it breaks, loglevel `debug` and **attach** it here as a file (drag & drop into the text field).

mw-nfi, 2022-11-11:
Here are the logs.... 
[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/9989080/zwave-js-ui-store.zip)

Thanks!

AlCalzone, 2022-11-11:
> 2022-11-11T09:45:19.970Z DRIVER   Serial port errored: The serial port closed unexpectedly! (ZW0100)

Everything looks Ok, then this happens (stick disconnects) and no further communication with the stick is possible.

This is either something on the hardware side (power supply, loose contact,...) or a problem on the host system. In any case, this is not an issue in zwave-js and you'll probably need to look at system logs to figure out why that happens.

mw-nfi, 2022-11-16:
Error detected.
Without Docker it works. Probably there are problems in the Docker stack.

robertsLando, 2022-11-17:
@AlCalzone Couldn't it be caused by soft reset?

AlCalzone, 2022-11-17:
No. Soft reset happens at startup (if not disabled/blacklisted), not in the middle of communication.

closed 2023-03-07 by AlCalzone