zwave-js/zwave-js #5163: Locks not updating entity during autolock they are not reporting back. Attention AlCalzone

thorrrr, 2022-10-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

YES, BUT a Z-Wave JS UI developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I have 2 brand new Yale Locks and a new Aeotec  7 Stick with 2 new Yale Z-Wave modules.
Both added to Z-Wave and looks ok.
But the locks have no groups binding and i noticed an error that when locks auto lock this is not being picked up.
It shows the unlock but when  either locks autolock it does not show in HA or update the entity.
It looks like there is no interview.

Also from the log 
he Get command (without Supervision) worked. The Set command with Supervision failed. !!!!

PLease note this was gained from help in Discord with a great user #freshcoast  
For the attention of AlCalzone

### Device information

Manufacturer: Aeotec 
Model name: Z-Wave Plus Aeotec Z-Stick 7
Node ID in your network: 

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Z-Wave JS UI
Current version: 1.1.0

Z-Wave Network
Diagnostics
Driver Version:
10.3.0
Server Version:
1.24.0
Home ID:
3714350544
Server URL:
ws://a0d7b954-zwavejs2mqtt:3000

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

file:///home/dale/Downloads/zwavejs_2022-10-18.log
https://logs.notifiarr.com/?cbc2e62e88fd7986#242X7THs4XibxCue8ufnCMsC2Ce8f3AcCwnJcD1nt8Fa

thorrrr, 2022-10-18:
[zwavejs_2022-10-18.log](https://github.com/zwave-js/node-zwave-js/files/9814344/zwavejs_2022-10-18.log)

thorrrr, 2022-10-18:
i only have 2 device ID 2 
Device ID
297-1-7 (0x0129-0x0007-0x0001)and ID 4 they are the identical locks
ID 4
Device ID
297-1-7 (0x0129-0x0007-0x0001) 

[image]

[image]

AlCalzone, 2023-03-06:
Apologies for the delay. I'm only slowly catching up with all the issues that were reported in Q4/2022.

I think your issue is caused by a missing lifeline association, which isn't configured because of a timeout.
That should be fixed by https://github.com/zwave-js/node-zwave-js/pull/5417, which was released as part of driver version 10.8.0.

Please update if you haven't already, then re-interview the device. If that doesn't fix your issue, please reopen and post a new interview log.

closed 2023-03-06 by AlCalzone

thorrrr, 2023-03-07:
Everything working fine for months cheers.
Please close