zwave-js/zwave-js #5164: yarn install fails on Raspberry Pi because turbo dropped arm support

darkbasic, 2022-10-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

turbo recently dropped its linux_arm target (https://github.com/vercel/turborepo/pull/1903/files) and thus yarn install fails on the Raspberry Pi:

```
pi@hass-home:~/devel/node-zwave-js$ yarn
➤ YN0000: ┌ Resolution step
➤ YN0002: │ @zwave-js/cc@workspace:packages/cc doesn't provide jest (p93dff), requested by jest-extended
➤ YN0002: │ @zwave-js/config@workspace:packages/config doesn't provide jest (pf36d7), requested by jest-extended
➤ YN0002: │ @zwave-js/host@workspace:packages/host doesn't provide jest (p78fa7), requested by jest-extended
➤ YN0002: │ @zwave-js/maintenance@workspace:packages/maintenance doesn't provide jest (p84488), requested by jest-extended
➤ YN0002: │ @zwave-js/serial@workspace:packages/serial doesn't provide jest (p3c88c), requested by jest-extended
➤ YN0002: │ @zwave-js/testing@workspace:packages/testing doesn't provide jest (pbf7f4), requested by jest-extended
➤ YN0002: │ zwave-js@workspace:packages/zwave-js doesn't provide jest (pd99e8), requested by jest-extended
➤ YN0000: │ Some peer dependencies are incorrectly met; run yarn explain peer-requirements <hash> for details, where <hash> is the six-letter p-prefixed code
➤ YN0000: └ Completed in 4s 957ms
➤ YN0000: ┌ Fetch step
➤ YN0013: │ typescript@npm:4.8.3 can't be found in the cache and will be fetched from the remote registry
➤ YN0013: │ typescript@patch:typescript@npm%3A4.7.4#~builtin<compat/typescript>::version=4.7.4&hash=a1c5e5 can't be found in the cache and will be fetched from the disk
➤ YN0013: │ typescript@patch:typescript@npm%3A4.8.3#~builtin<compat/typescript>::version=4.8.3&hash=a1c5e5 can't be found in the cache and will be fetched from the disk
➤ YN0000: └ Completed in 233m 52s
➤ YN0000: ┌ Link step
➤ YN0007: │ esbuild@npm:0.15.7 must be built because it never has been before or the last one failed
➤ YN0007: │ nice-napi@npm:1.0.2 must be built because it never has been before or the last one failed
➤ YN0007: │ @serialport/bindings-cpp@npm:10.7.0 must be built because it never has been before or the last one failed
➤ YN0007: │ turbo@npm:1.5.2 must be built because it never has been before or the last one failed
➤ YN0009: │ turbo@npm:1.5.2 couldn't be built successfully (exit code 1, logs can be found here: /tmp/xfs-3869fda1/build.log)
➤ YN0009: │ nice-napi@npm:1.0.2 couldn't be built successfully (exit code 1, logs can be found here: /tmp/xfs-c17ba418/build.log)
➤ YN0007: │ @zwave-js/repo@workspace:. must be built because it never has been before or the last one failed
➤ YN0000: └ Completed in 3m 49s
➤ YN0000: Failed with errors in 237m 49s
```

```
# This file contains the result of Yarn building a package (turbo@npm:1.5.2)
# Script name: postinstall

/home/pi/devel/node-zwave-js/node_modules/.store/turbo-npm-1.5.2-45c791897d/node_modules/turbo/node-platform.js:38
    throw new Error(`Unsupported platform: ${platformKey}`);
          ^

Error: Unsupported platform: linux arm LE
    at pkgAndSubpathForCurrentPlatform (/home/pi/devel/node-zwave-js/node_modules/.store/turbo-npm-1.5.2-45c791897d/node_modules/turbo/node-platform.js:38:11)
    at checkAndPreparePackage (/home/pi/devel/node-zwave-js/node_modules/.store/turbo-npm-1.5.2-45c791897d/node_modules/turbo/install.js:252:28)
    at Object.<anonymous> (/home/pi/devel/node-zwave-js/node_modules/.store/turbo-npm-1.5.2-45c791897d/node_modules/turbo/install.js:302:1)
    at Module._compile (node:internal/modules/cjs/loader:1155:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1209:10)
    at Module.load (node:internal/modules/cjs/loader:1033:32)
    at Function.Module._load (node:internal/modules/cjs/loader:868:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:81:12)
    at node:internal/main/run_main_module:22:47
    ```

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [X] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: git master (1989eceb5e0502f6c3171c78ab29b2764eadb59e)

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Older versions did work on the pi

### Attach Driver Logfile

Cannot attach driver log

AlCalzone, 2022-10-18:
Nice... Semver isn't a thing with them apparently.

I'm afraid I can't do anything about this for now, unless someone figures out a workaround.
Otherwise, I'll look at this in January.

darkbasic, 2022-10-18:
Can't we downgrade the version and pin it?

AlCalzone, 2022-10-19:
That version had other issues, so a downgrade will require a bunch of configuration changes I'm not able to do on mobile.

darkbasic, 2022-10-19:
Maybe we can fork it and revert that commit until you will find some time to migrate to a better library? I can look into it and make a PR if you want.

darkbasic, 2022-10-19:
Making things worse my main development workstation is ppc64le, deprecated as well.

AlCalzone, 2022-10-19:
If you want to look into it, I'd rather go back to 1.4.x. There should be a couple of commits to config changes (mainly relative paths outside of the package subdirectories) which are necessary for caching to work properly.

When I'm back from paternity leave, I'll check whether I'll be sticking with turborepo with a workaround or switch to something else.

darkbasic, 2022-10-19:
> When I'm back from paternity leave

Congrats! I'll have a look at what I can hack to make it work in the meantime and let you know if I need any help.

darkbasic, 2022-10-22:
I've tried a frankenstein turbo but I didn't manage to successfully build it, so I've just pinned 1.4.7 and I finally managed to build node-zwave-js on my Raspberry Pi 2:

```
pi@hass-home:~/devel/node-zwave-js$ git diff package.json 
diff --git a/package.json b/package.json
index 3bec04695..1563a5cb3 100644
--- a/package.json
+++ b/package.json
@@ -87,7 +87,7 @@
     "source-map-support": "^0.5.21",
     "ts-nameof": "^5.0.0",
     "ts-patch": "^2.0.2",
-    "turbo": "1.5.2",
+    "turbo": "1.4.7",
     "typescript": "4.8.3",
     "zwave-js": "workspace:*"
   },
```

```
pi@hass-home:~/devel/node-zwave-js$ yarn && yarn build
➤ YN0000: ┌ Resolution step
➤ YN0002: │ @zwave-js/cc@workspace:packages/cc doesn't provide jest (p93dff), requested by jest-extended
➤ YN0002: │ @zwave-js/config@workspace:packages/config doesn't provide jest (pf36d7), requested by jest-extended
➤ YN0002: │ @zwave-js/host@workspace:packages/host doesn't provide jest (p78fa7), requested by jest-extended
➤ YN0002: │ @zwave-js/maintenance@workspace:packages/maintenance doesn't provide jest (p84488), requested by jest-extended
➤ YN0002: │ @zwave-js/serial@workspace:packages/serial doesn't provide jest (p3c88c), requested by jest-extended
➤ YN0002: │ @zwave-js/testing@workspace:packages/testing doesn't provide jest (pbf7f4), requested by jest-extended
➤ YN0002: │ zwave-js@workspace:packages/zwave-js doesn't provide jest (pd99e8), requested by jest-extended
➤ YN0000: │ Some peer dependencies are incorrectly met; run yarn explain peer-requirements <hash> for details, where <hash> is the six-letter p-prefixed code
➤ YN0000: └ Completed in 5s 288ms
➤ YN0000: ┌ Fetch step
➤ YN0000: └ Completed in 11s 226ms
➤ YN0000: ┌ Link step
➤ YN0007: │ esbuild@npm:0.15.7 must be built because it never has been before or the last one failed
➤ YN0007: │ nice-napi@npm:1.0.2 must be built because it never has been before or the last one failed
➤ YN0007: │ @serialport/bindings-cpp@npm:10.7.0 must be built because it never has been before or the last one failed
➤ YN0007: │ turbo@npm:1.4.7 must be built because it never has been before or the last one failed
➤ YN0007: │ @zwave-js/repo@workspace:. must be built because it never has been before or the last one failed
➤ YN0000: └ Completed in 4m 46s
➤ YN0000: Done with warnings in 5m 6s
• Packages in scope: @zwave-js/cc, @zwave-js/config, @zwave-js/core, @zwave-js/host, @zwave-js/maintenance, @zwave-js/nvmedit, @zwave-js/serial, @zwave-js/shared, @zwave-js/testing, @zwave-js/transformers, zwave-js
• Running build in 11 packages
@zwave-js/transformers:build: cache miss, executing 425f9c969ebd9339
@zwave-js/shared:build: cache miss, executing ccd5b6ba7785e526
@zwave-js/core:build: cache miss, executing 8bd4d2ef97dcbf11
@zwave-js/nvmedit:build: cache miss, executing 764f3eae839e2db4
@zwave-js/maintenance:prebuild: cache miss, executing 6e45d3ce02645400
@zwave-js/maintenance:build: cache miss, executing 26399b1addbbecb9
@zwave-js/config:codegen: cache miss, executing bd77f644774d522a
@zwave-js/config:build: cache miss, executing 821a8519d60bb30c
@zwave-js/host:build: cache miss, executing 94fc90ad134d91cd
@zwave-js/serial:build: cache miss, executing c16c8361e572dc1b
@zwave-js/testing:build: cache miss, executing b6ee0b118871dfc1
@zwave-js/cc:codegen: cache miss, executing 148bff6c5fba3141
@zwave-js/cc:codegen: Found 49 CC value definitions...
@zwave-js/cc:codegen: Found 58 API classes...
@zwave-js/cc:build: cache miss, executing a2ccb8b98a9fb9c7
zwave-js:build: cache miss, executing 8f67b937034f88b0

 Tasks:    14 successful, 14 total
Cached:    0 cached, 14 total
  Time:    24m36.607s
```

After pinning `winston` and `semver` I also managed to `yarn link ../node-zwave-js --all` in `zwave-js-ui`:

```
pi@hass-home:~/devel/zwave-js-ui$ git diff
diff --git a/package.json b/package.json
[... 42 lines cut ...]
pi@hass-home:~/devel/zwave-js-ui$ yarn
➤ YN0000: ┌ Resolution step
➤ YN0002: │ ts-node-dev@npm:2.0.0 [b9e30] doesn't provide @types/node (p8a743), requested by ts-node
➤ YN0060: │ zwave-js-ui@workspace:. provides eslint (p560d0) with version 8.21.0, which doesn't satisfy what eslint-loader requests
➤ YN0000: │ Some peer dependencies are incorrectly met; run yarn explain peer-requirements <hash> for details, where <hash> is the six-letter p-prefixed code
➤ YN0000: └ Completed in 5s 261ms
➤ YN0000: ┌ Fetch step
➤ YN0000: └ Completed in 10s 890ms
➤ YN0000: ┌ Link step
➤ YN0072: │ The application uses portals and that's why --preserve-symlinks Node option is required for launching it
➤ YN0007: │ core-js@npm:2.6.12 must be built because it never has been before or the last one failed
➤ YN0007: │ docsify@npm:4.12.2 must be built because it never has been before or the last one failed
➤ YN0007: │ core-js@npm:3.24.1 must be built because it never has been before or the last one failed
➤ YN0007: │ vue-demi@npm:0.13.11 [0f984] must be built because it never has been before or the last one failed
➤ YN0007: │ leveldown@npm:5.6.0 must be built because it never has been before or the last one failed
➤ YN0007: │ @serialport/bindings-cpp@npm:10.7.0 must be built because it never has been before or the last one failed
➤ YN0007: │ level@npm:5.0.1 must be built because it never has been before or the last one failed
➤ YN0000: └ Completed in 3m 51s
➤ YN0000: Done with warnings in 4m 11s
```

Unfortunately running `yarn build:server` fails spectacularly:
[zwave-js-ui-build-error.txt](https://github.com/zwave-js/node-zwave-js/files/9844143/zwave-js-ui-build-error.txt)

If I don't link `node-zwave-js` then it builds correctly, but since I need to do development that is not an option. I don't know if there might be any kind of relationship with the pinned `turbo` 1.4.7. It used to work in the past (I've ported node-zwave-js to yarn 2 exactly because of this) but I didn't try it since a long time.

AlCalzone, 2022-10-22:
I haven't had a good experience with linking node-zwave-js into zwave-js-ui. What does work is packing and installing the packed tarballs, but this has two drawbacks:
1. no "live" code updates
2. The default `yarn pack` doesn't handle inter-package dependencies, so you need to do this: https://github.com/zwave-js/node-zwave-js/blob/c97dbf16d5aa727a6fe007a256c2cd67096377ac/.github/workflows/test-and-release.yml#L85

darkbasic, 2022-10-24:
I've made the tarballs, but how am I supposed to use them in zwave-js-ui?

```
pi@hass-home:~/devel/zwave-js-ui$ git diff
diff --git a/package.json b/package.json
index 872bb4e..72e4c0e 100644
--- a/package.json
+++ b/package.json
@@ -174,7 +174,7 @@
     "vuedraggable": "^2.24.3",
     "vuetify": "^2.6.8",
     "winston": "^3.3.3",
-    "zwave-js": "^10.3.0"
+    "zwave-js": "link:.tarballs/zwave-js.tgz"
   },
   "devDependencies": {
     "@babel/cli": "^7.18.10",
pi@hass-home:~/devel/zwave-js-ui$ yarn
➤ YN0000: ┌ Resolution step
➤ YN0001: │ Error: @zwave-js/cc@file:./zwave-js-cc.tgz::locator=zwave-js%40file%3A.tarballs%2Fzwave-js.tgz%3A%3Alocator%3Dzwave-js-ui%2540workspace%253A.: ENOENT: no such file or directory, open 'node_modules/zwave-js/zwave-js-cc.tgz'
    at ms (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:314:10129)
    at ro (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:314:10393)
    at Ai.readFileBuffer (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:319:30860)
    at Ai.readFilePromise (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:319:30494)
    at Vh.readFilePromise (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:319:37537)
    at Da.readFilePromise (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:319:37537)
    at BL.fetchFromDisk (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:604:6424)
    at async p (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:434:1414)
    at async m (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:434:1769)
    at async y (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:434:2778)
➤ YN0000: └ Completed in 6s 763ms
➤ YN0000: Failed with errors in 6s 856ms
```

AlCalzone, 2022-10-24:
Uhh shit I linked the wrong command.
You need to add the `--absolute` flag to the monopack command. Yarn has a bug where it resolves relative paths in tarballs incorrectly.

AlCalzone, 2022-10-24:
That is strange. I added that flag specifically for this error. Maybe `yarn remove zwave-js` first.

darkbasic, 2022-10-24:
```
pi@hass-home:~/devel/zwave-js-ui$ yarn add file:.tarballs/zwave-js.tgz 
Usage Error: The file:.tarballs/zwave-js.tgz string didn't match the required format (package-name@range). Did you perhaps forget to explicitly reference the package name?
```

`yarn add` fails so I've tried to manually add `"zwave-js": "file:.tarballs/zwave-js.tgz"` to `package.json` after `yarn remove zwave-js` but I still the same error:

```
pi@hass-home:~/devel/zwave-js-ui$ yarn
➤ YN0000: ┌ Resolution step
➤ YN0032: │ @serialport/bindings-cpp@npm:10.7.0: Implicit dependencies on node-gyp are discouraged
➤ YN0032: │ node-addon-api@npm:4.3.0: Implicit dependencies on node-gyp are discouraged
➤ YN0001: │ Error: @zwave-js/cc@file:./zwave-js-cc.tgz::locator=zwave-js%40file%3A.tarballs%2Fzwave-js.tgz%3A%3Alocator%3Dzwave-js-ui%2540workspace%253A.: ENOENT: no such file or directory, open 'node_modules/zwave-js/zwave-js-cc.tgz'
    at ms (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:314:10129)
    at ro (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:314:10393)
    at Ai.readFileBuffer (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:319:30860)
    at Ai.readFilePromise (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:319:30494)
    at Vh.readFilePromise (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:319:37537)
    at Da.readFilePromise (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:319:37537)
    at BL.fetchFromDisk (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:604:6424)
    at async p (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:434:1414)
    at async m (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:434:1769)
    at async y (/home/pi/devel/zwave-js-ui/.yarn/releases/yarn-3.2.1.cjs:434:2778)
➤ YN0000: └ Completed in 14s 183ms
➤ YN0000: Failed with errors in 14s 275ms
```

AlCalzone, 2022-10-24:
Can you take a look at the zwave-js.tgz contents? The package.json in there should reference the other tarballs using absolute paths.

darkbasic, 2022-10-26:
> but this has two drawbacks:
> no "live" code updates
> The default yarn pack doesn't handle inter-package dependencies, so you need to do this

There is a far bigger downside with this approach: whenever you will package new tarballs yarn will still use the old cached tarballs. To work it around you will have to `yarn cache clean --all` but that will take a looong time and it's not pleasant to develop this way.

yarn link used to work well, would you be interested in a bisect to find out when it stopped working? So we could hopefully fix it.

AlCalzone, 2022-10-26:
Sure. Last time I tried, it didn't work because node-zwave-js and zwave-js-ui used different versions of the same package.

darkbasic, 2022-10-26:
> Last time I tried, it didn't work because node-zwave-js and zwave-js-ui used different versions of the same package

That was always the case, but it's usually just two packages and it's a easy to fix as pinning them in zwave-js-ui.