zwave-js/zwave-js #5167: [Bug]:

catchthemonster, 2022-10-19:
### Checklist

- [ ]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

PKG executable

### Z-Wave JS UI version

8.2.1

### ZwaveJS version

10.3.0

### Describe the bug

Everything works!
But if I update firmware for inovelli switches ether through OTA updates or advanced firmware updates nothing is updated.
i still get in report old version of firmware...

### To Reproduce

same as above

### Expected behavior

to update firmware

kpine, 2022-10-19:
There's already an issue for Inovelli firmware issues, #5081.

catchthemonster, 2022-10-19:
Folks, kpine, I am not sure what is the error in firmware and which version...
I am trying to upgrade from 1.35 to 1.45, 1.57 to 1.61 (beta) with no success 
Unrelated is that I can't even update Aeon usb stick to 1.2 version of firmware but this has nothing to do with truly excellent package that Lando created.
so i have
zwave-js-ui: 8.2.1
zwave-js: 10.3.0
home id: 3605931879
home hex: 0xd6ee2767

Here is the log and screen shots of operation:
[image]
[image]
[image]
[image]

[zwavejs_2022-10-19.log](https://github.com/zwave-js/node-zwave-js/files/9821062/zwavejs_2022-10-19.log)
and after upload of firmware:
same things reported:
[image]

AlCalzone, 2022-10-19:
Looks like the first part (target 1 of the update) gets applied successfully, but the 2nd part doesn't start (yet?). Unfortunately I cannot investigate this until I am back from paternal leave.

The 100% is a UI issue though. The update isn't done, only the first target. @robertsLando I extended the firmware update events to indicate the overall progress and which part is being updated etc.
You should probably use the new parameter (see docs) to display progress.

catchthemonster, 2022-10-19:
ok, thanks for the update, we will wait for change when it happens...
For now i will try alternate methods ...

robertsLando, 2022-10-20:
> The 100% is a UI issue though. The update isn't done, only the first target. @robertsLando I extended the firmware update events to indicate the overall progress and which part is being updated etc.

AFAIK I'm already using that, let me do a check.

Just to be clear @catchthemonster the problem is the second part is not done so even if I fix UI this wouldn't work

AlCalzone, 2022-10-20:
Yes, seems so. I need to figure out why, but the ui should not indicate that the update is complete.

catchthemonster, 2022-10-20:
Agree with both...
For what is worth, zwave2mqtt  framework did this correctly  for me ...
Maybe you can look at that project and compare the code base ...

Ether way thank you both and app is excellent aside from firmware ...
Regards,

robertsLando, 2022-10-20:
> For what is worth, zwave2mqtt framework did this correctly for me

What you mean? zwave-js-ui and zwavejs2mqtt are the same application

catchthemonster, 2022-10-20:
zwavejs2mqtt if I remember used open-zwave cpp libs. It might be the same up I don't know. Last year i was able to update firmware with it just fine

AlCalzone, 2022-10-20:
zwave2mqtt used open-zwave which never supported firmware updates.

zwave**js**2mqtt and zwave-js-ui are the same and use node-zwave-js, which is (to my knowledge) the  only open source Z-Wave driver to support firmware updates.

Tyfosho, 2022-10-21:
Same issue here. I get a timeout error after it tries to start the second update.
`
[image]

`

AlCalzone, 2022-10-21:
@Tyfosho Seeing a driver log on loglevel debug of this would be helpful.
Please attach it here as a file using drag&drop

Tyfosho, 2022-10-21:
@AlCalzone - any instructions anywhere on how to grab that log? I can do it now if so.

AlCalzone, 2022-10-21:
Here you go: [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs)
Make it loglevel `debug` and attach it here as a file (drag & drop into the text field).

catchthemonster, 2022-10-21:
I honestly do not remember exactly, but I know I was able to upgrade
firmware just fine with the old implementation of the  node.js  service.
There were no issues and targets worked.
There was a bug in inovelli dimmers that would kill the light if the 80%
threshold was crossed.
Now I see that whatever I build today has nothing to upload firmware ...
Sorry if I confused everyone...

robertsLando, 2022-10-24:
https://github.com/zwave-js/zwave-js-ui/pull/2755

Tyfosho, 2022-10-24:
@AlCalzone driver log attached!

[z-ui_2022-10-24.log](https://github.com/zwave-js/node-zwave-js/files/9852998/z-ui_2022-10-24.log)

crawford, 2022-10-25:
@catchthemonster I also recall being able to do firmware updates to my LZW31-SN (Red Dimmers) switches back when the add-on was known as [zwavejs2mqtt](https://github.com/zwave-js/zwavejs2mqtt). I don't have any concrete evidence of it though, so maybe I'm mistaken.

I also want to add the following error messages to help make this issue easier to find for those who are troubleshooting. I ran into both of these while mindlessly pushing buttons after the firmware update failed:

> ERROR Z-WAVE-SERVER: Failed to start the update: A firmware upgrade is already in progress! (ZW1500)

> The firmware update was transmitted completely, cannot abort anymore. (ZW1504) abortFirmwareUpdate undefined

It looks like Mr. Calzone (congrats on the newborn - trying not to hail you) has a good understanding of the issue, but in case another logfile is needed:

<Details>

```
2022-10-25T21:25:37.386Z CNTRLR » [Node 022] Sending firmware fragment 384 / 389
2022-10-25T21:25:37.461Z CNTRLR » [Node 022] Sending firmware fragment 385 / 389
2022-10-25T21:25:37.530Z CNTRLR » [Node 022] Sending firmware fragment 386 / 389
2022-10-25T21:25:37.598Z CNTRLR » [Node 022] Sending firmware fragment 387 / 389
2022-10-25T21:25:37.666Z CNTRLR » [Node 022] Sending firmware fragment 388 / 389
2022-10-25T21:25:37.734Z CNTRLR » [Node 022] Sending firmware fragment 389 / 389
2022-10-25T21:25:38.171Z CNTRLR « [Node 022] Firmware update (part 1 / 2) succeeded with status OK_RestartPending
2022-10-25T21:25:38.173Z CNTRLR [Node 022] Continuing with next part in 5 seconds...
2022-10-25T21:25:38.173Z CNTRLR [Node 022] Updating firmware (part 2 / 2)...
2022-10-25T21:25:38.173Z CNTRLR » [Node 022] Starting firmware update...
2022-10-25 14:25:44.571 INFO Z-WAVE: Node 22: value updated: 38-0-currentValue 0 => 0
2022-10-25 14:25:46.794 INFO Z-WAVE: Node 22: value updated: 50-0-value-66049 0 => 0
2022-10-25 14:26:38.239 ERROR Z-WAVE-SERVER: Received no matching command within the provided timeout! (ZW0201)
ZWaveError: Received no matching command within the provided timeout! (ZW0201)
at Timeout.<anonymous> (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:4379:6)
at listOnTimeout (node:internal/timers:559:17)
at processTimers (node:internal/timers:502:7)
```

catchthemonster, 2022-10-26:
@crawford , agree! I know I am not senile yet, -)...
Yes congrats to Mr. Calzone, that is a wonderful thing...
Hopefully this will be fixed soon.
BTW, I am not super happy with Inovelly. They do have a cool dimmers, but buggy firmware, and for some reason when I loose power in house with generator, these switches have tendency to freak out and my light start acting up...
so firmware is critical to be up to date and easy to upload and install.
it would be also nice to include in selection beta drivers, so that people upload latest to test, instead of going number of clicks more to actually get a downloaded package from disk...
Regards,

AlCalzone, 2022-10-26:
@catchthemonster Beta drivers will be available, but AFAIK none of the UIs have this implemented yet.

robertsLando, 2022-10-28:
> Beta firmwares will be available, but AFAIK none of the UIs have this implemented yet.

What?

catchthemonster, 2022-10-28:
For what is worth that is correct. However, if zwave-js firmware driver is
capable via REST api calls to find beta drivers, then those drivers should
show as a choice in gui interface inline with stable drivers...
Just me...

robertsLando, 2022-10-28:
Opened a feat request: https://github.com/zwave-js/zwave-js-ui/issues/2768

robertsLando, 2023-02-09:
It's already available

InSearchOfTh1ngs, 2023-02-10:
@robertsLando, Ok I didn't know as this issue wasn't closed nor is the update server showing any firmware updates for my devices as I know they're not on the latest.  I'll try to manually update my devices this weekend and see if that works.

AlCalzone, 2023-02-10:
No progress yet

robertsLando, 2023-02-10:
Sorry closed it because I thought the issue was about the possibility to select a beta release from ota

AlCalzone, 2023-03-07:
Duplicate of #5081

(closing for housekeeping purposes, one issue about this is enough)

closed 2023-03-07 by AlCalzone