zwave-js/zwave-js #5170: Tmerature sensor information is lost when restarting

umtauscher, 2022-10-21:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

YES, BUT a Z-Wave JS UI developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
restarting homeassistant
What do you observe?

What did you expect to happen?
that temprature information for my sensors be kept during retsrat

Steps to reproduce the behavior:
  1. Go to restart
  2. Click on restart
  3. Scroll down to '...'
  4. Tmeratures are lost for all my Fibaro window/door sensors

### Device information

Manufacturer:  Fibaro
Model name: FGK101
Node ID in your network: <77>

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 2022.10.5
`node-zwave-js` branch: 8.1.0
`zwave-js-ui` branch: 10.3.0

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

I'm sorry, but I cannot produce any significant logs.

umtauscher, 2022-10-21:
This is the z-wave debug info after a restart
[Keller Nord.txt](https://github.com/zwave-js/node-zwave-js/files/9839846/Keller.Nord.txt)

and thsi ist the info of an idetical sensor after it has aquired the temperature again
[Keller Sued.txt](https://github.com/zwave-js/node-zwave-js/files/9839858/Keller.Sued.txt)

I suppose the temerature sensor info is not stored, because it is an optional sensor.
I have also tried enabling MQTT but it doesn#t change anything.
Cheers

umtauscher, 2022-10-21:
btw the Fibaro sensors2 with built in temperature sensor don't have this problem. They immediately show the temp.

AlCalzone, 2023-02-01:
Sorry for the delay, I'm only now getting to some of these issues. 
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).

While the log is running, re-interview the device and wait for the interview to finish.

umtauscher, 2023-02-01:
Thanks, for responding.
I hope I did everything right.
I was re-interviewing Node 101 which is a Fibaro door/windows sensor FGK101 with attached temperture sensor

[zwavejs_2023-02-01.log](https://github.com/zwave-js/node-zwave-js/files/10558592/zwavejs_2023-02-01.log)

Cheers

AlCalzone, 2023-02-07:
I think I found the issue. When the fix is released, you'll need to re-interview the device so the temperature sensor is detected correctly and stays.

closed 2023-02-07 by AlCalzone