zwave-js/zwave-js #5171: Yale YRD450-ZW2/YRMZW2-US Communication Issues

mbaran5, 2022-10-21:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

**What causes the bug?**
Attempting to control the lock state (locked/unlocked) via RF

**What do you observe?**
Most often, the lock physically actuates but fails to update the new status (locked > unlocked, or unlocked > locked) within a timely manner - typically takes 30-60 seconds. 

If I physically actuate the lock bolt myself, I typically see the status reflected in a very timely manner, within a second or so. The issue seems to be limited to attempting to control the lock via RF. 

**What did you expect to happen?**
I expect both the actuation of the bolt and the updated status to happen within a timely windows rather than the observed 30-60 seconds. 

**Steps to reproduce the behavior:**
  1. Attempt to toggle the lock state via HA
  2. Usually, see the lock bolt actuate
  3. Wait 30-60 seconds
  4. Have HA reflect the new lock state 

### Device information

**Manufacturer:** Yale/Assa Abloy
**Model name:** YRD450-ZW2 w/ YRMZW2-US module (ZW+ v1, S0 security) 
**Node ID in your network:** 22

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

`node-zwave-js`version: 10.3.0
`zwave-js-ui` version: 8.1.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

I have attempted to use the lock both as undefined in current FW (lock is too new and has no device json) as as well as using lock by adding the code from this PR https://github.com/zwave-js/node-zwave-js/pull/5154 to the config folder. 

I have attempted to (temporarily) move the stick about 5 feet closer to the lock. The total distance between stick and lock is less than 50 feet direct line of sight no walls. This did not make an impactful difference. 

### Did this work before?

Don't know, this is a new device

### If yes, where did it work?

This is a new setup, I've just migrated to z-wave js. This lock previously worked reliably on Hubitat running the latest Hubitat code. 

### Attach Driver Logfile

[zwavejs_2022-10-21.log](https://github.com/zwave-js/node-zwave-js/files/9842585/zwavejs_2022-10-21.log)

The event occurred around 2022-10-21T19:34 in the log file. I have emailed you my S0 code with reference to my GH user name.

mbaran5, 2022-10-22:
Update - toggling the lock via the keypad externally also seems to induce these communication issues where toggling the bolt from the inside does not. I’ll pull a new log shortly with me performing a lock from the keypad.

AlCalzone, 2023-03-07:
Sorry for taking so long to respond, I'm still catching up with some issues after my leave.
Is this still happening? If so, can you provide a new log, preferably on loglevel "debug", not "silly"?

mbaran5, 2023-03-07:
I've avoided the issue by running my ZW stick up in the attic as close to the front door lock as possible. The issue really seemed to be routing with S0 security. It would always force it's way back directly to the master node and then timeout trying to nonce handshake.

I am happy to move the stick back and run a few repairs though to capture a log if you feel there's value in doing that and possibly making the situation/timing of it better. I'd love to not have my stick running over a tcp bridge. Appreciate the info/help.

AlCalzone, 2023-03-07:
I can take a look, but this sounds like an issue in the physical world then.

closed 2023-03-18 by zwave-js-assistant[bot]