zwave-js/zwave-js #5172: All Of Sudden: Lots of Dead Nodes and VERY slow to scan on startup

RJ-Make, 2022-10-21:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
I came home today with a bunch of dead nodes

What do you observe?
Dead nodes, and pings generally wont fix them. 

### Device information

Manufacturer: Zooz
Model name: ZST10 700
Node ID in your network:  1

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: Latest
`zwave-js-ui` branch: Latest

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Had no problem with mass amounts of dead nodes

### Attach Driver Logfile

[zwave-js-ui-store(1).zip](https://github.com/zwave-js/node-zwave-js/files/9842976/zwave-js-ui-store.1.zip)

blhoward2, 2022-10-23:
What fw is your stick on? There is a nasty bug with fw < 7.17.2 on all 700 series sticks. Please check that, and that your stick is on a USB extension cable, then report back.

RJ-Make, 2022-10-23:
Thanks.. I'm on 7.18.1. hopefully the logs will provide some insight because today everything is back working normally. 

I have a feeling it's a device failing... bringing the network down.

RJ-Make, 2022-10-25:
At this point I'm going to close this, as I'm now 90% sure a failing ZEN20 Version 3 power strip brought the entire ~80 device network down.

closed 2022-10-25 by RJ-Make