zwave-js/zwave-js #5173: Dropping message with invalid payload: 0x010900a80001050000bfe5

LordMike, 2022-10-22:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
N/A

What do you observe?
The logs occasionally log some packets that zwave-js was unable to decode.

What did you expect to happen?
N/A

Steps to reproduce the behavior:
N/A

### Device information

Unsure which device. If any.

The controller is an Aeotec Z-Stick 7, firmware `v7.18.1`

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

zwave-js-ui via Snap

zwave-js-ui: 8.2.1.30d0208
zwave-js: 10.3.0
home id: 3815907651
home hex: 0xe3722143

### Did you change anything?

no

### If yes, what did you change?

I have not looked at the logs for a long time, so I'm not sure when this started appearing.

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

```
[... 113 lines cut ...]
2022-10-22T19:54:22.128Z SERIAL » [ACK]                                                                   (0x06)
2022-10-22T19:54:22.352Z SERIAL « 0x011d00a98600025702b3c4bb7f7f00000503160000030e03027f7f7f7f7f35    (31 bytes)
2022-10-22T19:54:22.353Z SERIAL » [ACK]                                                                   (0x06)
2022-10-22T19:54:22.355Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            134
                                    transmit status:        OK, took 5990 ms
                                    repeater node IDs:      3, 22
                                    routing attempts:       14
                                    protocol & route speed: Z-Wave, 100 kbit/s
                                    ACK RSSI:               -77 dBm
                                    ACK RSSI on repeaters:  -60 dBm, -69 dBm
                                    ACK channel no.:        0
                                    TX channel no.:         0
                                    route failed here:      3 -> 2
2022-10-22T19:54:22.366Z CNTRLR » [Node 005] Sending firmware fragment 54 / 2487
2022-10-22T19:54:22.382Z SERIAL » 0x013a00a901052e7a0600368420d21cfeadf080290eed33fd4007985006d5f0f22 (60 bytes)
                                  2c398fd202814ea22c2d5ec30e709b2d5e4c301072500000000878e
2022-10-22T19:54:22.383Z DRIVER » [Node 005] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      135
                                  └─[FirmwareUpdateMetaDataCCReport]
                                      report #: 54
                                      is last:  false
```

blhoward2, 2022-10-30:
What's the problem, here? If we're receiving an invalid payload that's what we're receiving so we can't do anything but drop it. This could be due to the packet being malformed over the air by interference. That's why we check and then drop them...

LordMike, 2022-10-30:
Ah - I was under the impression it might be a yet-to-be-supported packet.. I'm seeing these regularly, apparently, so I figured some device was sending something out.

If this is entirely normal (interference and whatnot), this is a non-issue.

AlCalzone, 2022-10-30:
In addition: Sometimes devices send invalid stuff, very rarely our parsing code has an incorrect check. I'll look at this after my parental leave, to know which it is

AlCalzone, 2022-11-13:
Thats an empty command from Node 5. I have no clue why it sends it, if it actually sends it, or if this is generated by the stick for some reason. In any case this should be discarded.

closed 2022-11-16 by zwave-js-assistant[bot]