zwave-js/zwave-js #5182: Incorrect configuration Minoston MS13ZS (0x0312-0xee00-0xee04)

gi-man, 2022-10-27:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Turn ON the light, then OFF then ON then OFF via JS leads to dead node. 

What do you observe?
The configuration for this device does not seem correct. The database has a different manual attached than the one provided by the vendor in the box. The database manual says MS13ZS in one area, but then in list the parameters as MS11ZS. 

Further one parameter = 7 ([25-112-0-7-2] Send Status Change Report: 3-Way seems to be what causes the problem of broken node. Keeping this disabled seems to avoid the problem, until an update or something changes it back, I am not sure. The current configuration in JS has 7-1, 7-2, 7-4 and 7-8, but the MS11ZS doesnt list Parameter 7 and the MS13ZS manual only looks to have a single parameter 7 with a single value to it (no 7-2, 7-4...)

What did you expect to happen?
Parameter match the manual and no broken nodes.

Steps to reproduce the behavior:
  1. Go to toggle light on, then off, then on, then off.

### Device information

Manufacturer: Minoston 
Model name: MS13ZS
Node ID in your network: 25

This pdf matched the printed manual in the box for this device:
[MS13Z-Manual.pdf](https://github.com/zwave-js/node-zwave-js/files/9881180/MS13Z-Manual.pdf)

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js-ui: 8.2.1
zwave-js: 10.3.0

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Months ago it used to work. I had the automation off so I'm not sure when this started. 

### Attach Driver Logfile

[zwavejs_2022-10-27.log](https://github.com/zwave-js/node-zwave-js/files/9881244/zwavejs_2022-10-27.log)

I cannot get it recreate during this log. I will post an updated one if it helps, but the configuration mismatch with the manual is still an issue.

blhoward2, 2022-10-30:
We never clobber the parameters so if it's changing that's a device bug. The default in our file is just for the UIs to display. It's not automatically set.

blhoward2, 2022-10-30:
I can't look at your log on mobile, but what stick are you on?

gi-man, 2022-10-30:
I created a new log showing the node (25) going from alive to dead. https://pastebin.com/L0bcVvyH

The stick I'm using is: 

Sigma Designs (Former Zensys) | QuickStick Combo | HUSBZB-1

https://www.amazon.com/GoControl-CECOMINOD016164-HUSBZB-1-USB-Hub/dp/B01GJ826F8

AlCalzone, 2022-10-30:
> transmit status: NoAck

On three consecutive attempts. Either there is a flaky connection or the device really stops responding in this scenario.

gi-man, 2022-10-30:
I can leave the device on for days without any issues (connections is still alive). I get the dead node only when I trigger a Turn Off via HA.

AlCalzone, 2022-10-30:
Other operations continue working? Like turn on, change config params, etc?

gi-man, 2022-10-30:
As far as I could tell, yes. I could do an automation to change the brightness or so every hour to test. It is just an outside porch light, so I mostly use Turn on as sunset and off at 22:00.

AlCalzone, 2022-10-30:
If it's really only the "off" command, then this smells like a device bug. The only difference between off and any other brightness is a zero for the brightness.

closed 2023-01-03 by AlCalzone

gi-man, 2023-01-03:
For reference in case anyone else stumbles on this in the future. I kept the settings from the manual MS13SZ instead of the default ones in JS. No issues in the last 2 months.