zwave-js/zwave-js #5183: More interview timeouts with Schalge Connect Lock (BE469)

mew1033, 2022-10-30:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I'm having problems with my Schalge Connect (BE469) Zwave (non-plus) lock. It's worked well for the most part for multiple years and I've always been able to get it back up and going when things go south. This time is different, I can't seem to get it to interview with the correct data. For example, there's almost nothing there when I look at the User Codes section.

I just tried setting the timeout parameter higher as per https://github.com/zwave-js/node-zwave-js/issues/4746, but that didn't help. In fact, it seemed to act _exactly_ the same, which means either I did it wrong or it isn't a timeout issue. I've attached the driver logs to this issue.

### Device information

Schalge BE469
Node 17.

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwave-js-ui` branch:

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Worked just fine until I started trying to reinterview it tonight... Not sure when things stopped working exactly.

### Attach Driver Logfile

[zwavejs_2022-10-30.log](https://github.com/zwave-js/node-zwave-js/files/9895394/zwavejs_2022-10-30.log)

AlCalzone, 2022-10-30:
Not much that I can investigate  on mobile, but the increased timeout (10s) is applied here.

mew1033, 2022-10-30:
I just spoke with someone I work with that uses Schlage Connects with Zwavejs, sounds like they recently started having problems as well. I'll see if they can get any logs.

AlCalzone, 2022-10-30:
We have several issues for this problem. Each time the lock randomly stops responding 🤷🏻‍♂️

mew1033, 2022-11-01:
Weird... I know these locks have been incredibly annoying. When they're working, they're awesome, I just wish they worked consistently. 
It seems like it started with a recent version of zwavejs. Did something change recently that could have messed with things?

One thing I've seen in the logs is that after I attempt to reinterview the lock, I see a few discarded packets. Is there any way to see what those are to see if maybe the lock is just responding outside of spec or something?

mew1033, 2022-11-01:
Sorry for the comment spam. I just downgraded to 6.8.0 and I can get the interview process to work reliably.
[image]
I've attached the zwave log from that version.
[zwavejs_2022-11-01.log](https://github.com/zwave-js/node-zwave-js/files/9913474/zwavejs_2022-11-01.log)

Next I'm going to try to find the exact version where the problem appears.

idcmp, 2022-11-06:
+1 with some extra details:

* this happens with both 8.2.2 and master zwavejs/zwave-js-ui docker images
* this happens on both my Zooz ZST10 700 S2 Stick (with the latest firmware)
* this happens on my HUSBZB-1 (ZigBee/ZWave) stick
* this happens on both my laptop and the server running HomeAssistant
* this happens regardless of distance to lock when enrolling
* this happens on both Schlage 469ZP locks I have

If it does enroll, the lock gets reported dead within a few minutes.

babrase, 2023-02-16:
I'd like to report that issue seems to be resolved, at least it is for me.  An attempt to add my lock, approximately 1 month ago failed much like as described in this issue. I don't have a record of what version I was running at that time.  A few days ago I upgraded my zwave js ui  installation to latest.  This put me at  z-wave js ui 8.8.5 and z-wave js 10.10.0.  This afternoon I again attempted to include my lock and was successful.  The interview completed successfully and 6 hours later the lock node still active and functional.

AlCalzone, 2023-02-16:
@babrase this might be the effect of a change I did in `node-zwave-js` v10.8.0. After detecting this Lock, all communication with it now waits 10s instead of the default 1s for a response (essentially how it behaved in 9.x for all devices). You could previously change the timeout globally in the Z-Wave JS UI settings, but this could increase unnecessary delays.

AlCalzone, 2023-02-16:
@mew1033 how's it working for you with the latest version?

mew1033, 2023-02-16:
Still not working correctly I think. I had the product and manufacturer info correctly listed here (not sure how), and I tried to do a re-interview. Ended up looking like this:
[image]

The major problem is that the entire User Code Values section is missing.

AlCalzone, 2023-02-16:
Try this please 

https://github.com/zwave-js/node-zwave-js/issues/5413#issuecomment-1421264904

mew1033, 2023-02-20:
No, looks like it's still missing some options. 
[image]

Want me to do a reinterview with all the debugging on?

AlCalzone, 2023-02-21:
Yep

closed 2023-08-28 by zwave-js-assistant[bot]