zwave-js/zwave-js #5184: Random spikes in power consumption [kWh] for all Z-Wave modules

Alfamonk, 2022-10-31:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

YES, BUT a Z-Wave JS UI developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Hello,

Since I have moved all my Z-Wave devices from Fibaro Home Center 2 to Home Assistant (using Aeotec Z-Stick Gen5) i see random spikes in power consumption, which makes the Energy panel and every consumption statistic unusable. I have tried to understand if the spikes happen at a specific time, or maybe just some specific devices have the issue (in hope those devices are broken and need to be replaced). But no. This happen to almost all my Z-Wave plugs and dimmers, at completely random times.

What happens exactly? Well, let's take for example "sensor.birou_plug_internet_electric_consumption_kwh" (It's a Fibaro Wall Plug). It's usually under 1kWh per day. But today 8:35 AM, there is a recording for 3546.82 kWh, which is impossible!

Another example is one with negative values, which should also be impossible for a cumulative attribute, like kWh (unless you reset it to 0). In this case, it's "sensor.smart_meter_electric_consumed_kwh" from a Qubino Smart Meter - ZMNHTD1. The last value in consumption was above 34,531.5 kWh. But today at 8:36 AM, there is a recording for only 517.2 kWh, which messes all the statistics in Energy.

There must be something related to Z-Wave, as my other modules that use Zigbee don't have this issue.

What have i tried? Well, I have healed the network, I have updated Aeotec Z-Stick Gen5 to the latest firmware, I have updated Home Assistant and all addons to the latest versions. Restarted Home Assistant, restarted VMWare, changed the stick location (using an USB extender to not interfere with other devices)... I don't know what else to do.

The only fix I found was to write down the entity with issues, go into Developer Tools - Statistics, search for the entity, go at that exact time where the spike happened, and replace it with 0 values. It was OK for a while (when I had like 3-4 items), but now that I moved completely to Home Assistant, it's a real pain! 20-30 devices per day going to random values... it's just killing me.

Can you please let me know what can I do to fix this?
Thank you in advance!

P.S. I must admit I have also posted this on github.com/hassio-addons/addon-zwave-js-ui (already closed) and github.com/home-assistant/core, but as I haven't received any answers, I hoped I will get some solution in here. Thanks!

### Device information

Manufacturer: AEON Labs
Model name: Z‐Stick Gen5 USB Controller
Node ID in your network: 134-90-1 (0x0086-0x0001-0x005a)

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 

zwave-js-ui: 8.2.1
zwave-js: 10.3.0
home id: 3574832442
home hex: 0xd5139d3a

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Yes, all my Zwave devices worked flawlessly for years on my old Fibaro Home Center 2. I probably have seen a spike like that once a year, on a single device. But now I have dozensof these spikes, daily. 

### Attach Driver Logfile

[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/9899643/zwave-js-ui-store.zip)

Alfamonk, 2022-10-31:
Sorry @zwave-js-bot - i'm re-attaching the right files now.
Thank you!
[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/9899765/zwave-js-ui-store.zip)

blhoward2, 2022-11-02:
Sometimes packets get garbled over the air. There is no way to control for this. You could probably make a template sensor that disregards values over a certain threshold, I’d think.

Alfamonk, 2022-11-02:
> Sometimes packets get garbled over the air. There is no way to control for this.
I'm sorry, but this is not the right answer. Let's see why:
- First, most of these Z-Wave devices have been working flawlessly for more than 5 years on my old Fibaro smart home. Never had these issues, so nothing was "garbled over the air"
- Second, these over the air devices have checksum for packets... They just don't "throw away" data, and hope for the best. Some of them are crucial devices (like security locks, fire alarms, co2 sensors, and so on"). Nobody in the right mind would ever use a device that would send "garbled" packets... 
- Third, let's take the case of spiking down (minus values). Even if such a power consumption device would send such a crazy value, HA or the controller, or maybe the Z-wave addon should not even register such a thing. A kWh consumption can only go up (unless we talk about kWh produced, but that's usually another entity). So, it should be discarded by default. Or at least have an option to do so.

I have seen this issue with many other people (even on the HA oficial group on Facebook), but none of them found a way, and most of them just gave up using the energy usage section. A thing I don't want to lose. 

I don't know how to do those templates (and I am not sure I will be able to do that for 50-100 devices), but I also think there should be some kind of way to fix these.

P.S. If this helps, I have also noticed a strange things in the timestamps these spikes happen (I was checking to see, maybe there is something wrong in my house at that time, like power issues). So, for example:

1:36:31 p.m.	29.oct
5:36:36 p.m.	29.oct
8:36:42 a.m.	29.oct
3:36:35 a.m.	30.oct
1:36:33 a.m.	30.oct

Those are 4 different entities, sending wrong infos into the kWh values. Different devices, different hours, different seconds but THE SAME MINUTES (36)? After a while, the pattern has changed... and the "36" was replaced by "52".

blhoward2, 2022-11-02:
The same minute is expected if you’re using time-based reporting. Once you lose power once they’ll all be synced up.

AlCalzone, 2022-11-03:
> First, most of these Z-Wave devices have been working flawlessly for more than 5 years on my old Fibaro smart home. Never had these issues, so nothing was "garbled over the air"

You changed the radio receiver and most likely the location of it, which definitely has an influence on RF performance.

> Second, these over the air devices have checksum for packets... They just don't "throw away" data

Yes, a simple XOR "checksum". This just requires two bits to flip at the same position in different bytes and you can no longer detect the corruption. 
Including devices with encryption will prevent this FWIW.

> Third, let's take the case of spiking down (minus values). Even if such a power consumption device would send such a crazy value, HA or the controller, or maybe the Z-wave addon should not even [...]

I half-agree but this isn't the job of the driver IMO. For many devices we can't even know if they support distinguishing between produced and consumed energy and some actually can go up and down in total.

> Those are 4 different entities, sending wrong infos into the kWh values. Different devices, different hours, different seconds but THE SAME MINUTES (36)? After a while, the pattern has changed... and the "36" was replaced by "52".

I'd look into this. There might actually be something in your house (or the neighbor?) causing RF noise at certain times.
Also

alvinchen1, 2022-12-01:
@Alfamonk, I get your frustration on this, I had a very similar problem with a lot of errant spikes and unrealistic numbers like apparently my Xbox Series X was consuming or generating enough energy to power all of Tokyo, haha.  Here are some suggestions that I found tuning my Z-Wave network which helped A LOT. 

1) Adjust some of the default energy reporting intervals.  Instead of every 10 seconds (if that's the default of your device) try every 30 seconds.  I have over 30 devices reporting power (mostly Zooz) and this adjustment alone significantly improved the reliability of my numbers in Energy Dashboard.  Even if you can adjust the reporting on 75% of your devices, maybe leaving 25% critical devices that you need the default reporting intervals, this will still likely help a lot.

2) Check your device in the Z-Wave-JS Config DB to see if it has known reporting issues.  I believe a recent update actually surfaces this information in the device page, but you might want to check as well.  For example, I had a bunch of these devices https://devices.zwave-js.io/?jumpTo=0x0258:0x0100:0x1027:0.0 and you can see in the Database it says

[image]

3) Of those two don't help, try and disable energy reporting on a few sample devices to see if perhaps you have certain devices in your network that are problematic.  For example, I was able to determine that two of my furthest devices were constantly sending "mangled" packets sequentially and when I disabled energy reporting on them, I saw a lot of those go away.  I had to make some tough decisions as to whether or not those two devices' energy metering was worth the instability of my Z-Wave network.  Spending time in the debug logs helped me find some of my problem nodes.

Hope that helps!

Alfamonk, 2022-12-02:
> Yes, a simple XOR "checksum". This just requires two bits to flip at the same position in different bytes and you can no longer detect the corruption. Including devices with encryption will prevent this FWIW.
I have noticed that 99% of my Zwave modules are connected on my coordinator without security enabled (I just let them connect as default, I thought that should be enough). So, should i just remove and re-connect all my Zwave modules with security enabled? I am guessing that will be encrypted and could solve the wrong values issue?

P.S. Isn't there any other option that I could just enable the security, without having to remove and re-add the modules? Some of them are really hard to access (physically)... I was hoping to do everything from inside the Z-Wave JS panel.

alvinchen1, 2022-12-10:
Hmmm, I am not sure what changing the TX power would do for you.  I really can't say on that one.  I definitely did also put my Z-Wave Stick on an extension cable and put it approx 4 feet away from my Raspberry Pi 4 which was highly recommended to prevent radio interference from the Pi.  And finally, yes, I have a 700 series too, so maybe it's time for that.  I would maybe try and introduce individual changes one at a time and let your Z-Wave network bake for a week or so before introducing another one and see if that helps your network a bit.  Frustrating, I know to wait, but too many changes at once and you might mix issues/results/causes.

closed 2023-03-25 by zwave-js-assistant[bot]