zwave-js/zwave-js #5185: Z-wave JS no longer scans parameters list from Aeotec 700 series devices

iCSpotRun, 2022-10-31:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Unknown.

What do you observe?
Parameter values are missing when interrogating the MultiSensor 7 from Aeotec. Support engineer has confirmed that this affects their other 700 series devices as well. A user can, however, make an adjustment to a value and that value will stick. Re-interviewing does not fix issue.

What did you expect to happen?
Values should be present

Steps to reproduce the behavior:
  1. In Home Assistant, go to 'Z-wave JS integration -> Devices -> Aeotec 700 device' (mine is the MultiSensor 7)
  2. Click on 'Configure'
  3. 'Notice that all parameter fields are empty'
[image]

### Device information

Manufacturer: Aeotec Ltd.
Model name: ZWA024
Node ID in your network: 12

Though Aeotec support has confirmed to me that this affects all their 700 series devices.

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
JS Server:  1.24.0
Z-Wave JS: 10.3.0

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Cannot recall what version of HA/Z-wave JS, but this worked earlier in the year.

### Attach Driver Logfile
[... 76 lines cut ...]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 144
                                    └─[SupervisionCCGet]
                                      │ session id:      8
                                      │ request updates: false
                                      └─[BinarySensorCCReport]
                                          type:  Motion
                                          value: false
2022-10-31T15:42:02.633Z SERIAL » 0x011d00a9010c119f03db005c5dc758781996870c325b24b324000000000028    (31 bytes)
2022-10-31T15:42:02.633Z DRIVER » [Node 012] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x24
                                  │ callback id:      0
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 219
                                    └─[SupervisionCCReport]
                                        session id:          8
                                        more updates follow: false
                                        status:              Success
                                        duration:            0s
2022-10-31T15:42:02.640Z SERIAL « [ACK]                                                                   (0x06)
2022-10-31T15:42:02.641Z SERIAL « 0x010401a90152                                                       (6 bytes)
2022-10-31T15:42:02.641Z SERIAL » [ACK]                                                                   (0x06)
2022-10-31T15:42:02.642Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true

AlCalzone, 2022-11-01:
What the bot said. This log snippet is incomplete. And please attach the logfile through drag and drop.

iCSpotRun, 2022-11-01:
[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/9912241/zwave_js.log)

Log attached. In this log, I triggered Motion and manually adjusted Parameter 4 values from: Blank -> 11 -> 10 -> 11

cadwizzard, 2022-11-04:
I also saw this behaviour. I solved it by moving the sensor within good direct connection range of 1m and re-interviewing. 
trying to re-interview at distance/over repeaters always left the fields empty. The sensor worked otherwise as expected though. I tried several times with the same results. On 2 sensors.

I also saw that trying to Smartstart include this sensor never worked, resulting in 3 new z wave nodes being created, but none configuring as the sensor, this was within 1m of the controller. I tested this several times, but the nodes keep incrementing x3, so tried manual inclusion, then it was fine,
The sensors were on the latest 1.4.1 firmware

iCSpotRun, 2022-11-04:
I tried re-interviewing numerous times with the sensor only 1m from the radio and still no luck with the config parameters. I even excluded and included at this distance. I am using the Home Assistant Z-wave JS integration. Not sure how to do any other type of inclusion. I tried with Z-wave JS UI, but getting some strange serial port error when setting up the Z-wave controller.

iCSpotRun, 2022-11-04:
Ok, got past the Z-wave JS UI serial port issue. I started completely from scratch and followed the UI inclusion process with the sensor no more than 1m away from the radio. Still no parameters :(

iCSpotRun, 2022-11-07:
[zwavejs_2022-11-07.log](https://github.com/zwave-js/node-zwave-js/files/9955996/zwavejs_2022-11-07.log)

Ask requested. Inclusion via Z-wave JS UI with sensor only 1m away from radio. I did this numerous times to be sure. Configuration parameters still missing.

I think this may have something to do with security. Forced "no encryption" during inclusion and the Parameters are present in this scenario.

iCSpotRun, 2022-11-07:
[zwavejs_2022-11-07-1.log](https://github.com/zwave-js/node-zwave-js/files/9956032/zwavejs_2022-11-07-1.log)

Here's the log with Inclusion on "No Encryption". Parameters getting pulled in successfully.

iCSpotRun, 2022-11-07:
Additional info. I tried this with Zooz Stick and Aeotec Z-stick 7. Latest firmware: 7.18.1
Same results :(

AlCalzone, 2022-11-13:
> [zwavejs_2022-11-07.log](https://github.com/zwave-js/node-zwave-js/files/9955996/zwavejs_2022-11-07.log)
> 
> Ask requested. Inclusion via Z-wave JS UI with sensor only 1m away from radio. I did this numerous times to be sure. Configuration parameters still missing.

That log still contains no inclusion at all.

> I think this may have something to do with security. Forced "no encryption" during inclusion and the Parameters are now present.

It may, but I can't tell why without a proper logfile.

iCSpotRun, 2022-11-13:
@AlCalzone I think I enabled the wrong log file. I did it under Settings - Z-wave rather then Settings - General.

I will redo when I am back home later this week. Thanks.

iCSpotRun, 2022-11-17:
Ok, let's try this again. This time with proper log:

[z-ui_2022-11-17.log](https://github.com/zwave-js/node-zwave-js/files/10034255/z-ui_2022-11-17.log)

iCSpotRun, 2022-11-17:
How about this? Thanks for your patience.

[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/10035214/zwave-js-ui-store.zip)

MartinHjelmare, 2022-12-16:
I'm also seeing weird behavior from this device (firmware version 1.1.1, driver version 10.3.0) after S2 inclusion. This device is at level 2 in my house and my controller is in the basement. Maybe it's just a bad connection?

The configuration parameter values are missing from the config panel in Home Assistant after secure inclusion. After one inclusion with S2 the device didn't update measurements or trigger on motion. After an exclusion and second S2 inclusion it now reports on motion but configuration parameter values are still missing. There were also unexpected values (entities) reported after the second inclusion. These values were gone after a re-interview.

Inclusion attempt 2 with S2:
[zwave_js_inclusion_s2_multisensor_7_20221216.log](https://github.com/zwave-js/node-zwave-js/files/10246143/zwave_js_inclusion_s2_multisensor_7_20221216.log)

Including without security makes the device behave as expected.

Inclusion attempt 3 without security:
[zwave_js_inclusion_no_security_multisensor_7_20221216.log](https://github.com/zwave-js/node-zwave-js/files/10246220/zwave_js_inclusion_no_security_multisensor_7_20221216.log)

PR #5811 (2023-05-16): fix(cc): optimize config parameter scan during interview
With this PR, we reduce the traffic during the Configuration CC (v3) interview by a lot. If a config parameter is documented in a config file, we now no longer query the name/label and info/description, which makes up the majority of queries.

Also, we no longer abort the entire interview if querying a parameter or its info fails.

fixes: #5185
fixes: #5810