zwave-js/zwave-js #5186: Inclusion fails: DRIVER no handlers registered!

devanl, 2022-11-02:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

SmartStart provisioned device failing to be included in network

What do you observe?

Partially added node is failed and lights repeatedly flicker until I disable provisioning.

What did you expect to happen?

- node gets added

Steps to reproduce the behavior:
  1. Go to 'watch debug console after provisioning device'

### Device information

Controller:
Manufacturer: Zooz  
Model name: ZST10-700 
Node ID in your network: 1
FW: v7.17.2

Device:
Manufacturer: Zooz  
Model name: ZEN77 V3.0

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

HA standalone addon 1.2.0
version:
zwave-js-ui: 8.2.1
zwave-js: 10.3.0

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

[... 89 lines cut ...]
2022-11-02T00:46:31.432Z DRIVER     invoking handler #0
2022-11-02T00:46:31.432Z CNTRLR   handling add node request (status = Failed)
2022-11-02T00:46:31.433Z CNTRLR   Adding the node failed
2022-11-02T00:46:31.440Z CNTRLR   stopping inclusion process...
2022-11-02T00:46:31.454Z SERIAL » 0x0105004ac59aef                                                     (7 bytes)
2022-11-02T00:46:31.455Z DRIVER » [REQ] [AddNodeToNetwork]
                                    action:       Stop
                                    high power:   true
                                    network wide: true
                                    callback id:  154
2022-11-02T00:46:31.462Z SERIAL « [ACK]                                                                   (0x06)
2022-11-02T00:46:31.466Z SERIAL « 0x0107004a9a0642006c                                                 (9 bytes)
2022-11-02T00:46:31.468Z SERIAL » [ACK]                                                                   (0x06)
2022-11-02T00:46:31.469Z DRIVER « [REQ] [AddNodeToNetwork]
                                    status:      Done
                                    callback id: 154
2022-11-02T00:46:31.473Z CNTRLR   The inclusion process was stopped
2022-11-02T00:46:31.479Z CNTRLR   Enabling Smart Start listening mode...
2022-11-02T00:46:31.487Z DRIVER       the message was handled
2022-11-02T00:46:31.491Z SERIAL » 0x0105004a4900f9                                                     (7 bytes)
2022-11-02T00:46:31.492Z DRIVER » [REQ] [AddNodeToNetwork]
                                    action: Enable Smart Start listening mode
2022-11-02T00:46:31.496Z SERIAL « [ACK]                                                                   (0x06)
2022-11-02T00:46:31.501Z CNTRLR   Smart Start listening mode enabled
```

AlCalzone, 2022-11-02:
The problem is this
```

2022-11-02T00:46:31.424Z DRIVER « [REQ] [AddNodeToNetwork]
                                    status:      Failed
                                    callback id: 153

```

Which means the stick failed to communicate and exchange all necessary inclusion commands with the node.
Make sure to follow the instructions for controller placement here: https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting
This is usually the issue when inclusion fails.

devanl, 2022-11-02:
I removed the failed node (exclusion did not work for this) and then retried after adding a new device in-between.  The inclusion succeeded.  A problem I would point out here is that before this happened I was at node 22 and now I appear to be at node 74.  

Is there something getting cleaned up too early that results in us thrashing (ie "no handlers registered!")?

Thank you for reviewing this

AlCalzone, 2022-11-02:
Maybe but I won't be able to investigate until January.