zwave-js/zwave-js #5189: Random additional Node IDs popping up that aren't legitimate devices

yetdog, 2022-11-02:
I currently have a ~30-node network, backed by a Zooz 700 USB stick running 7.17.2. All nodes are Zooz switches or dimmers, all included with no security and all 700-series z-wave+ devices. I've been slowly building out the mesh and things seem to be working well. However, yesterday, and again today, I went to check the network graph and noticed many nodes were displaying as disconnected. Looking in the logs, the GetRoutingInfo command was returning a TON of additional nodes IDs?? 

I saw this yesterday, but first noticed it when trying to add a new device, it popped up with a Node ID of 95 - which was WAY past where I was (my last node ID was 25), so I went looking and found that for some reason something was incrementing that counter.

Here's an example of where I first saw these node IDs popping up - based on the timestamp, it was just when I loaded the network graph this morning. The node IDs don't pop up in a log search before then. 

GOOD / TRUE / YESTERDAY log entry:
```
15:23:36.115 DRIVER « [RES] [GetRoutingInfo]
                        node ids: 3, 6, 7, 8, 9, 23, 24, 25, 27, 28
15:23:36.116 CNTRLR « [Node 029] node neighbors received: 3, 6, 7, 8, 9, 23, 24, 25, 27, 28
```

BAD / FALSE / TODAY log entry:
```
10:10:18.371 DRIVER « [RES] [GetRoutingInfo]
                        node ids: 25, 26, 28, 32, 33, 35, 40, 41, 89, 91, 93, 94, 95, 97, 100, 101, 
                      102, 103, 106, 125, 126, 127, 128, 133, 134, 150, 153, 154, 156, 160, 161, 163
                      , 164, 165, 166, 167, 185, 186, 187, 190, 191, 192, 193, 194, 222, 228, 231
10:10:18.374 CNTRLR « [Node 001] node neighbors received: 25, 26, 28, 32, 33, 35, 40, 41, 89, 91, 93
                      , 94, 95, 97, 100, 101, 102, 103, 106, 125, 126, 127, 128, 133, 134, 150, 153,
                       154, 156, 160, 161, 163, 164, 165, 166, 167, 185, 186, 187, 190, 191, 192, 19
                      3, 194, 222, 228, 231
```

The nodes tab in the UI still only shows my legitimate devices and IDs up to 29, which was where I left things yesterday after adding a handful of devices.

Yesterday after I saw the node ID issue, I essentially unplugged my USB stick, and replugged it back in. That seemed to clear things up and it "forgot" about all those other node IDs. 

Attempts to heal fail - so something seems to be awry with the controller or driver overnight? I do perform nightly backups of NVM and the store.

Does anyone have any thoughts on this?

robertsLando, 2022-11-02:
Did you try to healing your network?

robertsLando, 2022-11-02:
@kpine @blhoward2

yetdog, 2022-11-02:
So here's something interesting. I noticed that the logs tended to "drop out" during the NVM backup. Like I said I've disabled the scheduled ones going forward (so tomorrow morning will be telling), but I figured I'd poke the bear and try to force the issue. So I manually ran an NVM backup, and sure enough, after I ran the backup (which supposedly completed successfully and quickly per logs), i checked the neighbors via the graph again, and those odd nodeIDs showed up again. So it seems to be related to the NVM backup function in this case.

I've disabled scheduled backups so I'll keep an eye on things tomorrow morning to confirm,  but perhaps this still bears running down.

AlCalzone, 2022-11-02:
That's a bug in the 700 series firmware, triggered by NVM backups, scheduled to be fixed in 7.19.
For now the only workaround is to do a soft reset after backup. The driver is supposed to do that automatically, but only if soft reset isn't disabled.
Alternatively the stick can be re-plugged as a manual soft reset.

AlCalzone, 2023-01-03:
Closing since there's nothing additional we can do here except wait for the firmware update.

closed 2023-01-03 by AlCalzone