zwave-js/zwave-js #5192: ZRC-90 not recognized on (re-)interview

noci2012, 2022-11-04:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
????

What do you observe?
Device id is (0xXXXX-0xXXXX-0xXXXX) after interview.
Deleting the device, adding the device again, reinterview all result in the same. Only battery level/islow is reported. 

What did you expect to happen?
Full recognision of remote with ID 0x5254-0x0001-0x8510

Steps to reproduce the behavior:
  1. Go to '...' Device press wake button to keep it awake for 30 seconds
  2. Click on '...'  Reinterview
  3. Scroll down to '...' 
  4. See error
Device ID
undefined-undefined-undefined (0xXXXX-0xXXXX-0xXXXX) 

No buttons etc. are seen

### Device information

Manufacturer:  Remotec
Model name:  ZRC-90EU
Node ID in your network: 2

Device ID
undefined-undefined-undefined (0xXXXX-0xXXXX-0xXXXX) 

Expected device id from docs: 0x5254-0x0001-0x8510
(if manualy added this device to the XXXX.values.jsonl, and restart of docker, the device name shows up correctly , just without all buttons & events.  When reinterviewed the device ID is wiped).

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwave-js-ui: 8.2.2.4dd1414
zwave-js: 10.3.0

### Did you change anything?

yes (please describe)

[... 170 lines cut ...]
2022-11-04T11:55:56.131Z SERIAL » [ACK]                                                                   (0x06)
2022-11-04T11:55:56.136Z DRIVER « [Node 002] [REQ] [ApplicationCommand]
                                  └─[WakeUpCCWakeUpNotification]
2022-11-04T11:55:56.141Z CNTRLR « [Node 002] received wakeup notification
2022-11-04T11:55:59.242Z SERIAL « 0x010b000404020380035cca00e0                                        (13 bytes)
2022-11-04T11:55:59.250Z SERIAL » [ACK]                                                                   (0x06)
2022-11-04T11:55:59.257Z CNTRLR   [Node 002] [Battery] level: metadata updated                      [Endpoint 0]
2022-11-04T11:55:59.291Z CNTRLR   [Node 002] [+] [Battery] level: 92                                [Endpoint 0]
2022-11-04T11:55:59.334Z CNTRLR   [Node 002] [Battery] isLow: metadata updated                      [Endpoint 0]
2022-11-04T11:55:59.355Z CNTRLR   [Node 002] [+] [Battery] isLow: false                             [Endpoint 0]
2022-11-04T11:55:59.374Z CNTRLR   [Node 002] [Battery] chargingStatus: metadata updated             [Endpoint 0]
2022-11-04T11:55:59.388Z CNTRLR   [Node 002] [Battery] rechargeable: metadata updated               [Endpoint 0]
2022-11-04T11:55:59.406Z CNTRLR   [Node 002] [Battery] backup: metadata updated                     [Endpoint 0]
2022-11-04T11:55:59.450Z CNTRLR   [Node 002] [Battery] overheating: metadata updated                [Endpoint 0]
2022-11-04T11:55:59.482Z CNTRLR   [Node 002] [Battery] lowFluid: metadata updated                   [Endpoint 0]
2022-11-04T11:55:59.515Z CNTRLR   [Node 002] [Battery] rechargeOrReplace: metadata updated          [Endpoint 0]
2022-11-04T11:55:59.534Z CNTRLR   [Node 002] [Battery] disconnected: metadata updated               [Endpoint 0]
2022-11-04T11:55:59.564Z CNTRLR   [Node 002] [Battery] lowTemperatureStatus: metadata updated       [Endpoint 0]
2022-11-04T11:55:59.585Z DRIVER « [Node 002] [REQ] [ApplicationCommand]
                                  │ type: broadcast
                                  └─[BatteryCCReport]
                                      level:  92
                                      is low: false

```

noci2012, 2022-11-04:
Well it was a copy from a screendump.. But i can save it as well and upload it.
And when dropping the file into a box i get:  "Something went really wrong, and we can't process that file. try again"
either with extension .txt or .log.

noci2012, 2022-11-04:
[zrc90.log](https://github.com/zwave-js/node-zwave-js/files/9939689/zrc90.log)

noci2012, 2022-11-04:
It seems the next messages are missed during a discovery:
```
2022-11-04T22:03:36.235Z SERIAL « 0x010a00040402028407c700b1                                          (12 bytes)
2022-11-04T22:03:36.241Z SERIAL » [ACK]                                                                   (0x06)
2022-11-04T22:03:36.245Z DRIVER « [Node 002] [REQ] [ApplicationCommand]
                                  │ type: broadcast
                                  └─[WakeUpCCWakeUpNotification]
2022-11-04T22:03:36.248Z CNTRLR « [Node 002] received wakeup notification
2022-11-04T22:03:37.236Z SERIAL « 0x010a00040402028407c300b5                                          (12 bytes)
2022-11-04T22:03:37.243Z SERIAL » [ACK]                                                                   (0x06)
2022-11-04T22:03:37.249Z DRIVER « [Node 002] [REQ] [ApplicationCommand]
                                  │ type: broadcast
                                  └─[WakeUpCCWakeUpNotification]
2022-11-04T22:03:37.255Z CNTRLR « [Node 002] received wakeup notification
2022-11-04T22:03:38.228Z SERIAL « 0x010a00040402028407c600b0                                          (12 bytes)
2022-11-04T22:03:38.234Z SERIAL » [ACK]                                                                   (0x06)
2022-11-04T22:03:38.239Z DRIVER « [Node 002] [REQ] [ApplicationCommand]
                                  │ type: broadcast
                                  └─[WakeUpCCWakeUpNotification]
2022-11-04T22:03:38.247Z CNTRLR « [Node 002] received wakeup notification
2022-11-04T22:03:39.239Z SERIAL « 0x010a00040402028407bf00c9                                          (12 bytes)
2022-11-04T22:03:39.246Z SERIAL » [ACK]                                                                   (0x06)
2022-11-04T22:03:39.251Z CNTRLR » [Node 002] Sending node back to sleep...

```

AlCalzone, 2022-11-13:
The device doesn't report any CCs at all to be supported:
```
2022-11-04T11:55:44.953Z DRIVER « [Node 002] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x0203010106
2022-11-04T11:55:44.965Z CNTRLR « [Node 002] node info received
                                  supported CCs:
2022-11-04T11:55:44.996Z CNTRLR   [Node 002] Interview stage completed: NodeInfo
```
(notice there's nothing after "supported CCs").

Without any information what the device supports, there's no way to query anything.

closed 2022-11-16 by zwave-js-assistant[bot]

noci2012, 2022-11-19:
This is a device that is supposed to be supported. 
So where does it go wrong.

AlCalzone, 2022-11-19:
I explained it in my last post. If a factory reset doesn't help, you'll have to contact manufacturer support.
Can't support a device that's not behaving correctly.

noci2012, 2022-11-21:
@AlCalzone , problem solved
Reset according to documentation is non functional  when pressing R, after nearly 5 seconds the lamp lights up, after around 8 seconds it turns off, after 10 seconds it flashes.
according to Doc;s after 10 seconds the R should be pressed 3 times.  This appearantly fails.
The next reset did work:
pres R, wait until the solid blue light turns off, Release R, press R 3 times..., the device answer with a solid blue light of a few seconds. After this the device does register properly.