zwave-js/zwave-js #5193: Missing device configuration: Yale Keyless Connected Smart lock (P-KFCON-MOD-YALE)

cadwizzard, 2022-11-04:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Yale Keyless Connected Smart lock (P-KFCON-MOD-YALE)

### Manufacturer ID

0x129

### Product Type

0x07

### Product ID

0x00

### Firmware Version

12.32

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/2576?selectedFrequencyId=1

### Do you have a manual?

[Keyless-manual-2020-1.pdf](https://github.com/zwave-js/node-zwave-js/files/9938326/Keyless-manual-2020-1.pdf)

[image]

The device will include, but the lock controls are not present. After multiple exclusions/inclusions very close to a z stick 7 controller, the highest security I could get was S0.

Re-interviewing always results in the same behaviour and exact timeouts:
`2022-11-04T12:36:46.478Z CNTRLR « [Node 112] received response for protocol info:
                                  basic device class:    Routing Slave
                                  generic device class:  Entry Control
                                  specific device class: Secure Keypad Door Lock
                                  node type:             End Node
                                  is always listening:   false
                                  is frequent listening: 1000ms
                                  can route messages:    true
                                  supports security:     false
                                  supports beaming:      true
                                  maximum data rate:     40000 kbps
                                  protocol version:      3
2022-11-04T12:36:46.479Z CNTRLR   [Node 112] Interview stage completed: ProtocolInfo
2022-11-04T12:36:46.480Z CNTRLR » [Node 112] querying node info...
2022-11-04T12:36:47.709Z CNTRLR « [Node 112] node info received
                                  supported CCs:
                                  · Manufacturer Specific
                                  · Version
                                  · Security
2022-11-04T12:36:47.711Z CNTRLR   [Node 112] Interview stage completed: NodeInfo
2022-11-04T12:36:47.712Z CNTRLR » [Node 112] Querying securely supported commands (S0)...
2022-11-04T12:36:48.849Z CNTRLR   [Node 112] Timed out while waiting for a response from the node (ZW0201)
2022-11-04T12:36:48.850Z CNTRLR   [Node 112] Querying securely supported commands (S0), attempt 1/3 failed. Retr
                                  ying in 500ms...
2022-11-04T12:36:50.415Z CNTRLR   [Node 112] Timed out while waiting for a response from the node (ZW0201)
2022-11-04T12:36:50.416Z CNTRLR   [Node 112] Querying securely supported commands (S0), attempt 2/3 failed. Retr
                                  ying in 500ms...
2022-11-04T12:36:51.987Z CNTRLR   [Node 112] Timed out while waiting for a response from the node (ZW0201)
2022-11-04T12:36:51.988Z CNTRLR   [Node 112] Querying securely supported commands (S0) failed
2022-11-04T12:36:51.991Z CNTRLR   [Node 112] Interviewing Manufacturer Specific...
2022-11-04T12:36:51.992Z CNTRLR » [Node 112] querying manufacturer information...
2022-11-04T12:36:52.050Z CNTRLR « [Node 112] received response for manufacturer information:
                                    manufacturer: ASSA ABLOY (0x0129)
                                    product type: 0x07
                                    product id:   0x00
2022-11-04T12:36:52.052Z CNTRLR   [Node 112] Interviewing Version...
2022-11-04T12:36:52.053Z CNTRLR » [Node 112]   querying the CC version for Version...
2022-11-04T12:36:52.122Z CNTRLR   [Node 112]   supports CC Version (0x86) in version 1
2022-11-04T12:36:52.123Z CNTRLR » [Node 112] querying node versions...
2022-11-04T12:36:52.167Z CNTRLR « [Node 112] received response for node versions:
                                    library type:      Enhanced Slave (0x03)
                                    protocol version:  3.34
                                    firmware versions: 12.32
2022-11-04T12:36:52.168Z CNTRLR » [Node 112] querying CC versions...
2022-11-04T12:36:52.169Z CNTRLR » [Node 112]   querying the CC version for Basic...
2022-11-04T12:36:52.208Z CNTRLR   [Node 112]   supports CC Basic (0x20) in version 1
2022-11-04T12:36:52.208Z CNTRLR » [Node 112]   querying the CC version for Door Lock...
2022-11-04T12:36:52.245Z CNTRLR   [Node 112]   supports CC Door Lock (0x62) in version 1
2022-11-04T12:36:52.245Z CNTRLR » [Node 112]   querying the CC version for User Code...
2022-11-04T12:36:52.282Z CNTRLR   [Node 112]   supports CC User Code (0x63) in version 1
2022-11-04T12:36:52.282Z CNTRLR » [Node 112]   querying the CC version for Manufacturer Specific...
2022-11-04T12:36:52.331Z CNTRLR   [Node 112]   supports CC Manufacturer Specific (0x72) in version 1
2022-11-04T12:36:52.331Z CNTRLR   [Node 112]   skipping query for Security (0x98) because max implemented versio
                                  n is 1
2022-11-04T12:36:52.348Z CNTRLR   [Node 112] Embedded device config loaded
2022-11-04T12:36:52.350Z CNTRLR   [Node 112] Interviewing Door Lock...
2022-11-04T12:36:52.355Z CNTRLR » [Node 112] requesting lock configuration...
2022-11-04T12:36:53.436Z CNTRLR   [Node 112] Timed out while waiting for a response from the node (ZW0201)
2022-11-04T12:36:53.437Z CNTRLR » [Node 112] requesting current lock status...
2022-11-04T12:36:54.507Z CNTRLR   [Node 112] Timed out while waiting for a response from the node (ZW0201)
2022-11-04T12:36:54.510Z CNTRLR   [Node 112] Interviewing User Code...
2022-11-04T12:36:54.510Z CNTRLR » [Node 112] querying number of user codes...
2022-11-04T12:36:55.571Z CNTRLR   [Node 112] Timed out while waiting for a response from the node (ZW0201)
2022-11-04T12:36:55.573Z CNTRLR   [Node 112] Querying number of user codes timed out, skipping interview...
2022-11-04T12:36:55.574Z CNTRLR   [Node 112] Interview stage completed: CommandClasses
2022-11-04T12:36:55.576Z CNTRLR   [Node 112] Interview stage completed: OverwriteConfig
2022-11-04T12:36:55.577Z CNTRLR   [Node 112] Interview completed`

AlCalzone, 2022-11-04:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) of the inclusion, loglevel `debug` and attach it here as a file (drag & drop into the text field).
Inline logs on a different loglevel are not useful.

cadwizzard, 2022-11-17:
Since my report of this, i've not been able to get further logs. The lock is fitted to a gate and takes around an hour to remove the security from the gate to dismantle the lock and remove it, to get it close enough to a controller for any hope of secure inclusion. This leaves the gate insecure. As such i've not removed it again yet in order to be able to re-exclude, include and capture more logs

AlCalzone, 2022-11-17:
> sensor/alarm that is not in Z-Wave JS

AFAIK they are disabled/hidden in HA by default.

andrewng88, 2022-11-17:
So based on deprecated JS , that alarm/sensor as indicated in image 1 is an action I can retrieve so initiate the automation.

Do you mean the -JS doesn't use that and uses something else and does it mean if I can re-interview with no errors. It's working or.. ? 

Thanks

ADDZ2108, 2022-11-19:
I’ve just started my ha journey I’ve posted on ha forum and thought I’d look here and I’m having similar issues with this lock not showing in ha when triggered manually.

The lock includes fine with s2 security and can be locked and unlocked through ha but when used manually by codes or rfid tags nothing shows in ha.

closed 2023-03-07 by AlCalzone