zwave-js/zwave-js #5194: August Pro inclusion doesn't set up S2 security, interview times out

trainman419, 2022-11-06:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

### What causes the bug?
When including an August Pro smart lock on my Zwave network, using the Z-Wave JS UI, the inclusion shows the S2 DSK and asks for the S2 PIN (most of the time), but when the pairing is complete, the August app warns that the lock was paired insecurely, and the Z-Wave JS UI shows that the lock does not support security.

The interview process appears to fail, as the lock manufacturer, product, product code and firmware are not detected. The log contains errors about interview steps timing out.

I'm unable to control the lock.

I've found that if I do not select the security mode quickly enough during pairing, I am not prompted for the S2 PIN.

I'm not certain if this is a Z-Wave JS or Z-Wave JS UI bug, but since this seems to affect the implementation of S2 inclusion more than the UI, I'm reporting it here.

I'm using Z-Wave JS with HomeAssistant, but my issue seems to be with the Z-Wave JS inclusion process.

I'm fairly comfortable with docker and python, but not very familiar with the Z-Wave protocol internals. I'm happy to build my own docker image and try patches or experiments to help troubleshoot.

### What do you observe?

Lock not detected correctly:
[image]

August app warning:
[image]

Z-Wave JS log from pairing:
[august_pairing.log](https://github.com/zwave-js/node-zwave-js/files/9944907/august_pairing.log)

### What did you expect to happen?
I would expect S2 keys to be correctly set up during pairing, and that the interview would complete successfully.

Steps to reproduce the behavior:
  1. Ensure that your August Pro lock is not associated with a Zwave network.
  2. From Z-Wave JS UI, go to Actions -> Manage Nodes
  3. Select Inclusion and click Next
  4. Enter the device name and location, and click Next. I'm using Name "Front Door Lock" and Location "Front Entry" but I don't think these matter.
  5. Open the August app (I'm using the iPhone app) and navigate to Lock Settings.
  6. In the August app lock settings, select "Z-Wave Setup"
  7. In the August app, select "Add to a Z-Wave Network"
  8. In the Z-Wave JS UI, select Default and click Next
  9. In the August app, select "My Hub is in inclusion mode"
  10.  In Z-Wave JS, Inclusion should proceed to "Security Classes"
  11. In Z-Wave JS, select only S2 and click Next
  12. In Z-Wave JS, enter the device's S2 PIN and click Next
  13. Inclusion should complete.
  14. The August app should show the Security Alert
  15. Observe that the August Lock Pro is not enumerated properly in the Z-Wave JS UI.

I've found that if I don't click through the "Security Classes" prompt quickly enough, it does not prompt me for a S2 PIN, and instead shows that the node was added with security "None"

### Device information

Manufacturer: August
Model name: Smart Lock Pro
Node ID in your network: 82 (changes each time I try inclusion, so some of the screenshots may be inconsistent)

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 10.3.0
`zwave-js-ui` branch: 8.3.0.d050453

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

Log snippet from one pairing attempt:
[august_pairing.log](https://github.com/zwave-js/node-zwave-js/files/9944907/august_pairing.log)

Complete log file from today.
WARNING: I've been setting up a number of other devices, so there's a lot of other irrelevant noise in here. You asked for it, so have at it!
[zwavejs_2022-11-06.log](https://github.com/zwave-js/node-zwave-js/files/9944931/zwavejs_2022-11-06.log)

AlCalzone, 2022-11-12:
The problem is that the lock doesn't follow the S2 specs correctly. The issue is here:
[image]

specifically this part:
[image]

The logs show that the frame is sent exactly once, before we can decrypt it (no PIN entered yet) and after 240s Z-Wave JS times out.

AlCalzone, 2022-11-12:
The interview timeout happens, because no encryption is used, yet the lock requires encryption to communicate.

trainman419, 2022-11-13:
What do you think I should do?

Should I open a support ticket with August and try to get them to fix their firmware? It seems like their support team has not been super responsive, but I also haven't seen a bug report this specific, either.

Is it possible to modify Z-Wave JS or the Z-Wave JS UI so that I can enter the PIN code or the key before the pairing process starts, so that Z-Wave JS is able to decrypt the first KEX Set message?

I used to be able to use this lock without security enabled, but that doesn't seem to work any more; I suspect a recent firmware update disabled that functionality.

AlCalzone, 2022-11-13:
> Should I open a support ticket with August and try to get them to fix their firmware?

IMO that's the way to go. Or return it if they won't help. From what I've heard, August aren't the most reliable.

> Is it possible to modify Z-Wave JS or the Z-Wave JS UI so that I can enter the PIN code or the key before the pairing process starts, so that Z-Wave JS is able to decrypt the first KEX Set message?

No but if the lock (and your stick) supports SmartStart, you can scan a QR code and have the inclusion automated, side-stepping the issue.

trainman419, 2022-11-13:
Sadly, my lock doesn't support SmartStart, so that option is a non-starter for me.

I've submitted a support request to August; I'm hoping that they'll be able to make progress with the info here.

wystewart, 2022-12-17:
I managed to get a August smart lock pro (3rd gen) to include with S2.  I was having the same problem, it would not include with S2 only S0.  What I did was to use the inclusion function of the Zwave-JS GUI / Control Panel, then enter the DSK number from the lock during inclusion.  It included as S2 and correctly noted the manufacturer.

trainman419, 2022-12-18:
@wystewart interesting. Did you use the full DSK or the 5-digit DSK prefix included on the lock?
Which version of Zwave-JS and the Zwave-JS GUI are you using?
Which firmware version do you have on your August lock?

wystewart, 2022-12-18:
latest version of Z-Wave JS UI (1.4.0), just the 5 digit, that is all the add-on asked for.    the lock app says 1.59.0-2.0.4.  Bit confusing the fw version, I believe the latest is 2.0.4, but the 1.59.0 is confusing.
In the Z-wave JS UI, click on the right, ACTIONS->MANAGE NODES then select INCLUSION.
But sure to have the lock close to the zwave hub so there are no errors in transmission,  In my experience zwave is not reliable when nodes are far until it has built the network mesh.

AlCalzone, 2022-12-18:
How did you skip entering the DSK?

wystewart, 2022-12-18:
don't recall, but any case it seems to be needed to have S2

AlCalzone, 2022-12-18:
Yeah, the only exception is S2 Unauthenticated, which is not to be used by locks.

closed 2023-03-26 by zwave-js-assistant[bot]