zwave-js/zwave-js #5196: Zooz ZEN25 Assigned Lifeline to Non-Existant NODE_1

kaijk, 2022-11-07:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Include the ZEN25 into a Network whose Master Controller is NOT Node_1

What do you observe?
Debug Log reports failure to assign Lifeline and Lifeline ends up on NODE_1

What did you expect to happen?
Lifeline to be assigned to the Node hosting the Master Controller (in my case, NODE_46)

### Device information

Manufacturer:  ZOOZ
Model name:  ZEN25
Node ID in your network:  ZEN25 on Node 85, with Master Controller on Node ID 46

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
10.3.0
`zwave-js-ui` branch:
8.2.2

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwave-js_2022-11-07.log](https://github.com/zwave-js/node-zwave-js/files/9953677/zwave-js_2022-11-07.log)

kaijk, 2022-11-07:
Fixed the log file name. I couldn't get it to download from zwave-js-ui so I copied the log data into a new file.

kaijk, 2022-11-07:
> Have you tried removing the existing lifeline associations, and then re-interviewing? 

Yes, I  have tried removing the Assn. However, the Association will not remove, either by selecting the Remove Action for the Assn itself, or by Selecting Remove All.  I've also tried adding a second one hoping that it could then delete the first, but as you note, only one is allowed.

Here is the log for the attempted removal.

[zwavejs_2022-11-07a.log](https://github.com/zwave-js/node-zwave-js/files/9954889/zwavejs_2022-11-07a.log)

> later it looks as if the device was preset with a node 1 association

Interesting that the Device may be "Preset" with Node 1.  I have Reset the device any number of times and this is how it Includes every time.

blhoward2, 2022-11-12:
All Zooz did was email us to ask us...

kaijk, 2022-11-12:
I'm told that their engineers are looking into the issue.

AlCalzone, 2022-11-12:
I'm pretty sure this is Zooz's borked Supervision CC implementation again...
```
2022-11-07T19:26:23.199Z DRIVER » [Node 085] [REQ] [SendDataBridge]
                                  │ source node id:   46
                                  │ transmit options: 0x25
                                  │ callback id:      50
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 199
                                    │ extensions:
                                    │ · type: SPAN
                                    │   sender EI: 0x24eb60b24ced8d06f63dc479f2de21c4
                                    └─[SupervisionCCGet]
                                      │ session id:      29
                                      │ request updates: true
                                      └─[MultiChannelAssociationCCRemove]
                                          group id:  1
                                          node ids:
                                          endpoints: 1:0

success:
2022-11-07T19:26:23.337Z DRIVER « [Node 085] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -65 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 232
                                    └─[SupervisionCCReport]
                                        session id:          29
                                        more updates follow: false
                                        status:              Success
                                        duration:            0s

but nothing actually happens when removing that association
2022-11-07T19:26:23.346Z DRIVER » [Node 085] [REQ] [SendDataBridge]
                                  │ source node id:   46
                                  │ transmit options: 0x25
                                  │ callback id:      51
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 200
                                    └─[MultiChannelAssociationCCGet]
                                        group id: 1

2022-11-07T19:26:23.398Z DRIVER « [Node 085] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -65 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 233
                                    └─[MultiChannelAssociationCCReport]
                                        group id:           1
                                        maximum # of nodes: 1
                                        node ids:
                                        endpoints:          1:0
```

AlCalzone, 2022-11-12:
The ones where it does work probably have Supervision CC force-disabled. We can do the same here, but this isn't good, since Supervision CC is "necessary" to make S2 communication more reliable.

blhoward2, 2022-11-13:
> I'm told that their engineers are looking into the issue.
> 
> 
> > All Zooz did was email us to ask us...

They are not. They email @AlCalzone and me.

kaijk, 2022-11-20:
The ZEN25 has now been discontinued. Firmware updates? Not without their looking into the issue.  Maybe its easier to  kill a product than to fix it. Interesting.

alvinchen1, 2022-12-01:
That sucks! I have a bunch of ZEN25s and I'm generally pretty happy with them.  Sucks that they are discontinued.

closed 2023-03-07 by AlCalzone