zwave-js/zwave-js #5200: External temp. sensor readings not received by the thermostatic valve

shadowking001, 2022-11-08:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

**What causes the bug?**
I don't know

**What do you observe?**
External temperature sensor readings (Aeotec Multisensor 7) are not received properly by the Eurotronics Spirit Z-wave+ thermostatic valve.

What did you expect to happen?
The TRV should use my multisensor temperature to drive the valve opening instead of its own measurements.

Steps to reproduce the behavior:
  1. Add a group association on the Multisensor 7 => Number 11 for [Air temperature (Multivel)]
  
[image]

  2. Set parameter 8 of the Eurotronics TRV to -128 to use external temperature sensor reports
  
[image]

  3. Valve opening percentage will not vary anymore, if it is positive, it can reach 0 but it won't increase anymore. I did some monitoring for a month, as you can see it sticks to 0 (recent peaks corresponding to internal temperature measurement) :
  
[image]

  4. If i set the paramteter 8 to a valid integer (offset), TRV opens up and down again.
  5. I tried with two different Multisensor 7, and 3 TRV, same behaviour all the time.

My guess is that it should work out of the box, in the TRV documentation we can see what's expected to be reported to the valve for parameter 8 :

[image]

### Device information

**Sensor**
Manufacturer: Aeotec
Model name: ZWA024 (Multisensor 7), fw 1.4.1
Node ID in your network:  40

**TRV**
Manufacturer: Eurotronics
Model name: Eurotronic Spirit Z-Wave Plus, fw 0.16
Node ID in your network:  26

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 8.4.1
`node-zwave-js` branch: 18.10.0
`zwave-js-ui` branch: latest

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

Logs start when I'm doing the group association / setting parameter 8 to 128

[z-ui_2022-11-08.log](https://github.com/zwave-js/node-zwave-js/files/9965201/z-ui_2022-11-08.log)

shadowking001, 2022-11-08:
My bad, I added the zwave logs to the OP.

AlCalzone, 2022-11-13:
Are both devices included using the same S2 key/security class (Unauthenticated/Authenticated)? Everything else looks good IMO.
Theoretically they should work if they both share a common S2 key as long as the sending node has the same or higher highest security class. But in practice many devices can only inter-operate if both have the same highest class.

shadowking001, 2022-11-14:
Hello, thx for your input.
Both devices have been included with S2_Unauthenticated security class.
I ended up using the ZwaveJS API through MQTT with my domotic software to send the sensor temperature to the valve.

So I can tell now that the valve correctly receives the payload from the controller. Then afaik there are two possibilities left :

- The Multisensor 7 doesn't send anything to the valve through the group
- The format is not readable by the valve software

Maybe I could put the controller in the association and see what happens in the logs ?

AlCalzone, 2022-11-14:
Worth a try. Then you can compare if the format is the same.

closed 2023-09-04 by zwave-js-assistant[bot]