zwave-js/zwave-js #5201: Intermittent flickering on Fibaro FGD212

MichaelBitard, 2022-11-11:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
I'm not really sure, it's intermittent :(

What do you observe?
The light switch on and off every 4-5 seconds during ~1 minute, then switch off

What did you expect to happen?
The ligth stays on

Steps to reproduce the behavior:
Light on the lights using the wall switch

### Device information

Manufacturer: Fibargroup
Model name: FGD212
Firmware: 3.3 
Node ID in your network: 4

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

home assistant hassio addon version:  0.1.74

### Did you change anything?

no

### If yes, what did you change?

---

### Did this work before?

Yes (please describe)

### If yes, where did it work?

I have 2 other modules, and I don't have this behavior on them.

### Attach Driver Logfile

When this appears, I got a bunch of 

```
2022-11-11T17:08:51.492Z CNTRLR   [Node 004] [Notification]
                                    type:  Power Management
                                    event: Voltage drop/drift
2022-11-11T17:08:59.467Z CNTRLR   [Node 004] [Notification]
                                    type:  Power Management
                                    event: Voltage drop/drift
2022-11-11T17:09:12.473Z CNTRLR   [Node 004] [Notification]
                                    type:  Power Management
                                    event: Voltage drop/drift
2022-11-11T17:09:21.445Z CNTRLR   [Node 004] [Notification]
                                    type:  Power Management
                                    event: Voltage drop/drift
```

Also, on home assistant, I have a diagnostic panel (see screenshot) that display an "Unsafe" message on `Power Management - Load error`.
I have another FGD212 which is in the same state (unsafe) but the light works well.

My next step is to switch 2 of my FGD212 to see if this one is not faulty.

AlCalzone, 2022-11-11:
This looks like an issue with the switch's power supply. It even tells you that there is a voltage drop.

MichaelBitard, 2022-11-23:
I tried to replace a bulb with a non-flickering one, but the problem persist.

I'll try to switch with another FGD212 to see if it's the module that is dying

closed 2023-03-07 by AlCalzone