zwave-js/zwave-js #5204: Missing device configuration:  DONEXON varmo TZ eco

FaSchoeGo, 2022-11-13:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

 DONEXON varmo TZ eco

### Manufacturer ID

0x0148.0

### Product Type

0x0004

### Product ID

0x0148.0x0003.0x0004

### Firmware Version

0.16

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/3328

### Do you have a manual?

[image]

FaSchoeGo, 2022-11-13:
need help

Dommoe, 2022-12-04:
Hi,

i have Change the File:
https://github.com/zwave-js/node-zwave-js/commit/229fee4b35efead21bc698b09501a866a9551fe0

But I don't know what to do next

AlCalzone, 2023-01-31:
fixed in https://github.com/zwave-js/node-zwave-js/pull/5274

closed 2023-01-31 by AlCalzone