zwave-js/zwave-js #5211: Minoston MP21Z Fails after Z-Wave JS 1.3.0 Update in Home Assistant

jdggit, 2022-11-17:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

v8.4.1

### ZwaveJS version

10.3.0

### Describe the bug

Minoston MP21Z Wall Plug had been working with Home Assistant Z-Wave JS UI until latest update. With the update, it's dead. Roll back, it's live again.

Home Assistant
Prior - working version: Z-Wave JS UI to v8.2.2
HA Z-Wave JS UI v1.2.1

Updated - failing version: Z-Wave JS UI to v8.4.1 
HA Z-Wave JS UI v1.3.0

### To Reproduce

With latest update tried rebooting RPI4, reloading, healing, re-interviewing, pinging, etc. Node was declared dead. All other zwave devices are working including locks, wall plugs (other brands), wall switches, etc. with no intervention needed, everything came alive on its own.

Rolled back to v8.2.2, wall plug was live again. Did the update and roll back again to verify the only change is the latest update.

### Expected behavior

A seamless update...

### Additional context

Noted:  [bug] "Last seen" is updated on dropped message zwave-js/zwave-js-ui#2800 
- I'm seeing this same behavior on the dead node after update. Not sure it's related to the node being dead in the first place

jdggit, 2022-11-17:
[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/10036018/zwave-js-ui-store.zip)

After turning on logs, went to the Control Panel with the intent to ping the dead node. It was alive before I could ping it. Didn't do anything else and Logbook shows it turned on while I was figuring out which settings to change to generate and download logs. Log attached, but I don't know why it started working

robertsLando, 2022-11-18:
No clue, let's see if @AlCalzone can tell us more by checking logs. What's the node id?

jdggit, 2022-11-18:
[zwave-js-ui-store Update and reboots.zip](https://github.com/zwave-js/node-zwave-js/files/10042092/zwave-js-ui-store.Update.and.reboots.zip)

Node 7 was the problem. In the attached I had rolled back, then updated and restarted. - now the node won't fail. I'll have to add turn on log to file as part of my zwave debug process. Unless you can see anything in the file, I'm OK marking this as Closed.
Thanks

closed 2023-01-31 by AlCalzone