zwave-js/zwave-js #5219: Aeon Labs Garage Door Controller Gen5

TorbjornZachrisson, 2022-11-23:
I´m running Zwave2mqtt in a docker container, The latest working version for my barrier operator is container with zwavejs2mqtt 6.8.0 and Z-wave JS 9.0.6. All later versions do not have a working barrier operator for the Aeon Labs Garage Door Controller Gen5. Unfortunately I do not know if it´s the zwavejs2mqtt or Z-wave JS  that has the problem.

Kind regards
Z

TorbjornZachrisson, 2022-11-24:
Tried reinterview but it goes on forever. Goes fast with old version. 
[Node 010] Error during node interview: subsystemType is not a SubsystemType (
                                  ZW0322)
[zwavejs_2022-11-24.log](https://github.com/zwave-js/node-zwave-js/files/10087145/zwavejs_2022-11-24.log)

robertsLando, 2022-11-25:
cc @AlCalzone

laguinch, 2023-01-05:
Hy, I have also a problem with the module, the interview don't finish.
Thanks for your help

AlCalzone, 2023-03-08:
This should be fixed by https://github.com/zwave-js/node-zwave-js/pull/5465

closed 2023-03-08 by AlCalzone

TorbjornZachrisson, 2023-03-09:
Yes this works now. Many thanks!