zwave-js/zwave-js #5237: Driver: options is not a MultilevelSwitchCCStartLevelChangeOptions (ZW0322) / "options":{"preserveUnknownValues":true}

arnog23, 2022-12-03:
Following discusison here : [Missing commands for NICE Spa - IBT4ZWAVE - BusT4-Z-Wave interface #4804](https://github.com/zwave-js/node-zwave-js/discussions/4804#discussioncomment-4298074)

With `"options":{"preserveUnknownValues":true}`, when using Zwave commands of Nice Bidi, I have the following issue:

open or close, then stop. When opening or closing again, I have the following error message `Driver: options is not a MultilevelSwitchCCStartLevelChangeOptions (ZW0322)` and I am not able to control the Bidi anymore. I have to restart ZwaveJS.

This error do not appear when the option `preserveUnknownValues` is set to default (False).

arnog23, 2022-12-15:
Do you need more logs or tests from my side ?

AlCalzone, 2022-12-15:
No I think I have everything. Except time => https://github.com/zwave-js/node-zwave-js/issues/5129

arnog23, 2022-12-15:
> Except time => #5129

Thanks for the reminder. I had not seen it.

Enjoy it :-)

arnog23, 2023-03-02:
> No I think I have everything. Except time => https://github.com/zwave-js/node-zwave-js/issues/5129

Did you had time to look at it ?

AlCalzone, 2023-03-08:
now I did

PR #5542 (2023-03-08): fix: do not start level change with unknown `startLevel`
fixes: #5237