zwave-js/zwave-js #5245: Vision Security ZM1602: Siren Strobe Mode doesn't work

EchoPouet, 2022-12-08:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
I can't change the mode.

What do you observe?
Interface in HA indicate that change but not, I'm blocked in strobe mode.

What did you expect to happen?
The mode change.

Steps to reproduce the behavior:
  1. Go to Vision Security ZM1602 configuration page
  2. Click on Siren Strobe Mode combo box
  3. Change `Strobe` to `Siren`
  4. Mode change OK
  5. Turn on siren
  6. No sound !

### Device information

Manufacturer: Vision Security
Model name:  ZM1602
Firmware: 15.10
Node ID in your network: 13

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js-server-python = 0.43.1

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

Todo

EchoPouet, 2023-01-14:
Hello, How I can help you to debug this issue ?

AlCalzone, 2023-01-21:
Do you have a manual for that device? It may be that our config parameters are defined incorrectly.

AlCalzone, 2023-01-21:
zwave.eu listing:
[image]

vs. our file:
[image]

AlCalzone, 2023-01-21:
Ok, it looks like there are two versions:
300 series, FW 1.x, label ZM1602
[image]

and 500 series, FW 15.x, label ZM1602-5
[image]

We will need to separate the definitions by FW version again.

EchoPouet, 2023-01-22:
This this my manual -> https://doc.eedomus.com/files/ZM%201601-5%20Siren_20150602.pdf

Thanks a lot

PR #5539 (2023-03-08): fix(config): separate config for 300 and 500 series of ZM1602
fixes: #5245
closed 2023-03-08 by AlCalzone