zwave-js/zwave-js #5247: Zooz ZSE42 water leak sensor shows no sensor in HA when included with S2 security

madbrain76, 2022-12-10:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Unknown. Could be a Z-Wave JS bug or device bug. Zooz support thinks it's the former.

What do you observe?
If I include the device with S0 security, the "Water leak detected" sensor shows up in HA, and the device works.

If I can include my ZSE42 with S2 security (either authenticated or not), the "Water leak detected" sensor does not appear in HA, making the device unusable.
If I then try to re-interview the device, the process for re-interview is broken, but the sensor eventually does show up.

What did you expect to happen?
1) "Water leak detected" sensor should have shown up with S2 security in the first place, without a need for further re-interview.
2) Interview process should not be broken (see steps).

Steps to reproduce the behavior:
  1. Go to Settings / Devices / Z-Wave JS / Add device
  2. Press the Z-wave button 3 times on the ZSE42
  3. When prompted, type in the 5 digit PIN for S2 auth
  4. After inclusion, look click "View device". See that that "Water leak detected" sensor is missing
  5. I have included 3 log files for this inclusion scenario . 
  6. Click the vertical | menu under "Device info" and "Re-interview"
  7. Press the Z-wave button 4 times on the ZSE42 to wake it up
  8. Wait about 20 seconds. See that the re-interview process does not complete.
  9. Press the Z-wave button 4 times on the ZSE42 again
  10.  Repeat steps 6-9 , since these are not very reliable. Eventually the "Water leak detected" sensor will show up.
  11.  I have attached 2 logs of this re-interview process.
  12. Go to Settings / Devices / Z-Wave JS / Remove device
  13. Click "start exclusion"
  14. Press the Z-wave button 3 times on the ZSE42
  15. Go to Settings / Devices / Z-Wave JS / Add device
  16. Click "How do you want to add your device"
  17. Click "Legacy secure"
  18. Click "search device"
  19. Press the Z-wave button 3 times on the ZSE42
  20. Click "view device"
  21. Notice how the "Water leak detected" sensor shows up, unlike in step 4
  22. I have attached a log for the S0 inclusion in steps 15-19

### Device information

Manufacturer: Zooz
Model name: ZSE42
Node ID in your network: changing, see logs for inclusion/interview

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Not sure. Here are the versions info I see
Driver Version: 10.3.0
Server Version: 1.24.0

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwave_js_ZSE42_S2_inclusion_3.log](https://github.com/zwave-js/node-zwave-js/files/10201194/zwave_js_ZSE42_S2_inclusion_3.log)
[zwave_js_ZSE42_S2_inclusion_2.log](https://github.com/zwave-js/node-zwave-js/files/10201195/zwave_js_ZSE42_S2_inclusion_2.log)
[zwave_js_ZSE42_S0_inclusion.log](https://github.com/zwave-js/node-zwave-js/files/10201196/zwave_js_ZSE42_S0_inclusion.log)
[zwave_js_ZSE42_S2_re_interview.log](https://github.com/zwave-js/node-zwave-js/files/10201197/zwave_js_ZSE42_S2_re_interview.log)
[zwave_js_ZSE42_S2_re_interview_3.log](https://github.com/zwave-js/node-zwave-js/files/10201198/zwave_js_ZSE42_S2_re_interview_3.log)

madbrain76, 2022-12-13:
zwave-js-bot,
I renamed the log files to make it clear which test cases they were for.

Ikcelaks, 2022-12-14:
Firmware version 1.40 fixes this issue.

I was able to reproduce this issue by including an old ZSE42 with S2 security (firmware 1.10).
I then performed a firmware update to version 1.40. This worked fine with the S2 inclusion.
After the update completed, I performed a re-interview, and the leak sensor appeared.
I tested the sensor by setting it on a thin layer of water, and the leak was detected (there is a delay of a few seconds to minimize false positives, so it might not seem like it's working right away).

You will need to use four rapid presses on the internal button to wake the device immediately after initiating the firmware update and again immediately after initiating the re-interview. The blue light will flash to verify the device woke up.

The firmware is available from Zooz's support site. You need to register your device to gain access.

madbrain76, 2022-12-14:
Ikcelaks,
FYI, my two ZSE42 are both on firmware 1.40.1 .

> I then performed a firmware update to version 1.40. This worked fine with the S2 inclusion.

No, it does not work "fine", even on firmware 1.40.1 . The "Water leak" sensor should appear in step 4 of my test case, after S2 inclusion. It does not appear for me, and your comment indicates that it does not appear for you either, even with the latest firmware. That's undeniably a bug. The subsequent re-interview (steps 6-10) is a workaround for this bug. Steps 15-21 show that this bug does not appear when doing inclusion with S0 security, also. Ie. the bug is specific to S2 inclusion.

I filed this issue because I want this bug fixed. I don't know if it's a bug in the device firmware, or a bug in Z-Wave JS. I was hoping someone could determine that, so that the bug could be fixed in the right place. I have been corresponding with Zooz developers and pointed them to this defect and logs as well.

Ikcelaks, 2022-12-14:
@kpine Do your Zniffer logs show what's happening when zwave-js is showing the device as asleep between the "protocol info" and "node info" interview stages? My guess was that the driver was assuming the device was asleep due to missing ACKs. But, missing ACKs don't seem to be the issue in these logs... the device is sending the wrong info with correct communication.

@madbrain76 My apologies for coming off as dismissive in my previous reply. The first three lines of my comment were simply a log of exactly what I did and what I observed. My "this worked fine with S2 inclusion" was simply a note that I was able to perform the firmware update while in S2, so communication was working even though the sensor didn't show up.

I didn't realize that you were on 1.40.1 at the time, so the one improvement I thought I experienced was that my re-interview after the update completed normally, and the leak sensor appeared immediately after it completed. My reading of the initial issue made me think that your re-interview initially failed, and that the sensor only showed up sometime later (presumably after the node re-awoke and completed the interview). Since I always expect to need to re-interview a Zooz 700 series device after performing a firmware update, I made the bad assumption that things were now normal. Sorry about that.

madbrain76, 2022-12-14:
@Ikcelaks , no worry.
As far as the re-interview sometimes failing, that is correct. It didn't always work on the first try. It isn't reliable for me, even on firmware 1.40.1 . It never completes unless I press the Z-Wave button to wake up the device again in the middle of the interview, and the success or failure seems very dependent on timing. This is not a good user experience to be sure. and it's probably not something an average user would be able to figure out, IMO.

madbrain76, 2022-12-14:
@kpine, thank you for doing that !
This still leaves the question as to why the initial inclusion fails to find the sensor - the subject of this issue. Is it conclusively a firmware bug ?

madbrain76, 2022-12-14:
@kpine , thanks ! I have passed this on to Zooz support.

AlCalzone, 2023-01-30:
Anything new from Zooz? I'm inclined to close this in favor of the wakeup issue, which is something we can fix unlike temporarily incorrect device reports.

madbrain76, 2023-01-30:
No news from Zooz.

AlCalzone, 2023-01-31:
Ok. Let's close this issue then until there's news that points towards a driver issue.

closed 2023-01-31 by AlCalzone