zwave-js/zwave-js #5250: Device stuck at Protocolinfo

boscorelly, 2022-12-13:


boscorelly, 2022-12-13:
Hi,

this one ?
[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/10222878/zwavejs_current.log)

boscorelly, 2022-12-13:
mostly this one
[zwavejs_2022-12-11.log.gz](https://github.com/zwave-js/node-zwave-js/files/10222890/zwavejs_2022-12-11.log.gz)

boscorelly, 2022-12-28:
any update on this ?

robertsLando, 2022-12-29:
Sorry @boscorelly Alcalzone will be back soon and will take a look at this. Based on the logs you sent I think you have a 'zombie' node here. What I suggest is to exclude the node first, factory reset it and try to run replace failed node against node id 8 to re-include it

boscorelly, 2022-12-29:
Hi,I think you are right about the zombie node. The problem is that everything I tried to revive it never worked. Funny fact, the node has already been reset and is properly working since the next day I posted the issue 😂Message ID: ***@***.***>

robertsLando, 2023-01-02:
Closing for now so, feel free to re open it in case the bug comes back

closed 2023-01-02 by robertsLando

boscorelly, 2023-01-02:
The bug is still present. I was able to te-add the node but the ghost node is still present !

AlCalzone, 2023-01-02:
You'll have to remove it using the "remove failed node" feature.

boscorelly, 2023-01-04:
I already know. But it’s not working. If it was, I was not needing to create an issue !

AlCalzone, 2023-01-04:
Then please share driver logs of your attempts. On github.com, attaching them to an email doesn't work.

boscorelly, 2023-01-09:
they are here : https://github.com/zwave-js/node-zwave-js/issues/5250#issuecomment-1350045358

AlCalzone, 2023-01-09:
Ok, I honestly have no idea why that is failing. The response returned by the stick when asked to remove the node (0x10) isn't even a valid one according to both the 500 (allows 0x00...0x02) and 700 (allows 0x00...0x05) series documentation.
Maybe you'll have some luck after physically re-plugging the stick. Other than that, I don't know.