zwave-js/zwave-js #5254: Missing device configuration: Donexon varmo TZ eco

bushvin, 2022-12-15:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Donexon varmo TZ eco

### Manufacturer ID

0x0148

### Product Type

0x0003

### Product ID

0x0004

### Firmware Version

0.16

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/3328

### Do you have a manual?

[Manual](https://products.z-wavealliance.org/ProductManual/File?folder=&filename=MarketCertificationFiles/3328/varmo_TZ_eco_BAL_web_EN_0917.pdf)

IMO, the device is very similar, if not identical, to [this device](https://devices.zwave-js.io/?jumpTo=0x0371:0x0002:0x0015:0.0)

FaSchoeGo, 2022-12-16:
I did a PullRequest for this Problem some days ago https://github.com/zwave-js/node-zwave-js/pull/5242

bushvin, 2022-12-17:
let's hope it will be looked at soon...

FaSchoeGo, 2022-12-17:
👍

bushvin, 2023-01-20:
@robertsLando Are these requests looked at?

robertsLando, 2023-01-20:
Sorry @bushvin, @AlCalzone is back now and he may have missed this

bushvin, 2023-01-20:
No worries, usually you guys are very reactive. 
Maybe the bots configured to monitor these requests failed to warn you guys.

robertsLando, 2023-01-20:
Giving that there is an open PR this should be a fast fix, let's see if everything is ok 🙏🏼

AlCalzone, 2023-01-30:
fixed in #5274

AlCalzone, 2023-01-30:
@bushvin lots of work accumulated over the 3 months of leave and now I have less time to work 😬

closed 2023-01-30 by AlCalzone

robertsLando, 2023-01-31:
Making a release on my side too 🚀