zwave-js/zwave-js #5260: Schlage Connect locks started losing functionality.  Re-pairing doesn't bring the functionality back.

jds11111, 2022-12-16:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

YES, BUT a Z-Wave JS UI developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Not sure.  Using the UI withe the Schlage Connect lock?  It did work for quite awhile, but then Values started disappearing. Below is the description of starting from scratch to try and fix it. 

What do you observe?
User codes on the Schlage Connect lock are changed only temporarily.  The lock deletes them again, like so:
2022-12-16 10:52:08.653 INFO Z-WAVE: Writing '12345678' to 2-99-0-userCode-1
2022-12-16 10:52:11.479 DEBUG MQTT: Publishing to zwave/Back_Stairway/Backdoor_Lock/99/0/userCode/1: { time: 1671209531479, value: '12345678' } with options { qos: 1, retain: true }
2022-12-16 10:52:11.481 INFO Z-WAVE: Node 2: value updated: 99-0-userCode-1 00000000 => 12345678
2022-12-16 10:52:11.483 INFO Z-WAVE: Success zwave api call writeValue true
2022-12-16 10:52:16.633 INFO Z-WAVE: ValueRemoved: 99-0-userIdStatus-1 from node 2
2022-12-16 10:52:16.634 INFO Z-WAVE: Node 2: value removed: 99-0-userIdStatus-1
2022-12-16 10:52:16.636 INFO Z-WAVE: ValueRemoved: 99-0-userCode-1 from node 2
2022-12-16 10:52:16.637 INFO Z-WAVE: Node 2: value removed: 99-0-userCode-1
2022-12-16 10:52:16.640 INFO Z-WAVE: Node 2: metadata updated: 99-0-userIdStatus-1
2022-12-16 10:52:16.642 INFO Z-WAVE: Node 2: metadata updated: 99-0-userCode-1

What did you expect to happen?
I expected the values to stay on the lock.

Steps to reproduce the behavior:

1. Install the docker container for ZwaveJS-UI on a RPi4B. 
2. Attach the SiLabs ZST10-700  stick on one of the USB 2.0 slots. Find its location by device to be /dev/serial/by-id/usb-Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller_146e66591959ea11a40763035af1e32e-if00-port0
3. Update the firmware on the stick to 7.18.3
4. Replace the batteries in the lock to be brand new.
5. Place the stick touching the lock. Start pairing, using the DSK code and S2 Access control. 
6. See that it does not have all the available values, so unpair the lock, factory reset it, factory reset the stick, and start over. Repeat this process a couple dozen times until finally all the values are there. Wait 20 minutes or so, to make sure that all communication is complete.  Check the logs that there is nothing happening between the stick and the lock.
7.  Increase the code size to 8 digits.  Check the logs and the UI that it has taken hold, and everything looks OK.
8. Go to User Codes in the UI and see the slots for all 23 codes are available.  Replace one of the codes with something above like 12345678, watch the logs to see that it get written to the lock, and observe that it does temporarily, but then gets deleted again. 
9. Come here to report a bug, and go through the recommended steps, including refreshing the values. No change, except now that User Code slot has disappeared entirely.
10. Desperately try re-interviewing the node, but that just makes the situation worse, and Values disappear.  The whole time, the stick is next to the lock.

### Device information

Manufacturer: Schlage/Allegion
Model name: Connect
Node ID in your network:  2

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Same setup, I think, though I used to use it on HAOS successfully, too.

I came here because your documentation told me to, not because a developer did.

### Attach Driver Logfile

[zwavejs_2022-12-15.log.gz](https://github.com/zwave-js/node-zwave-js/files/10247813/zwavejs_2022-12-15.log.gz)

ornative, 2023-01-07:
I am seeing some weird behavoir as well with the BE469ZP (Schlage Connect locks).  In my case I tried to set user code (4) under user code v1.  It threw a driver error saying it must be string, I stopped and restarted Zwavejs-ui and the value fields no longer show up.  I did notice after this that more user codes showed up with 000000 in the code field- all of them through 16. I have user codes (2) and (3) configured, so I tried user code (1), which indicated success, but then the value went back to zero and the code and status started showing up (still with 000000) as the very last ones in the list.  

Not sure what is going on, but clearly there is something not quite working with these locks.

AlCalzone, 2023-01-09:
If you search for issues with Schlage locks, you'll see that there is often erratic behavior, up to the point where they randomly stop responding during the interview or suddenly take long breaks:
https://github.com/zwave-js/node-zwave-js/issues?q=is%3Aissue+Schlage

Some of those issues include workarounds (e.g. upping the response timeout in zwave-js), but it's not guaranteed to work.

Unfortunately I've wasted so much time helping with troubleshooting those pieces of ****, so I won't do that anymore.

philh30, 2023-01-23:
I've been struggling with this same issue of lock codes consistently reverting back to their prior state on a BE469ZP that I just moved over from ST where it was working properly. After zniffing and a lot of trial and error, I've found that it rejects any User Code commands that are encapsulated with Supervision. The lock accepts those commands after disabling Supervision.

AlCalzone, 2023-01-23:
That's some helpful information. Just to be sure so we don't apply the fix on a different device, can you show me the fingerprint of the device you tested this on?

philh30, 2023-01-23:
0x003b-0x0001-0x0469
Firmware 3.3

jds11111, 2023-01-24:
Is there anything that we mere mortals can do to fix this, other than wait for a patch?  Or can we help somehow?

AlCalzone, 2023-01-25:
If you use Z-Wave JS UI, you can put this file https://raw.githubusercontent.com/zwave-js/node-zwave-js/4a0e6f3ee8493d5eaaf2cd35852cb1c6e904c746/packages/config/config/devices/0x003b/be469zp.json into the custom device files directory: https://zwave-js.github.io/zwave-js-ui/#/device-support/custom-device-files

closed 2023-01-25 by AlCalzone

jds11111, 2023-01-25:
Fantastic!  I can confirm that this works.  However, the already-paired locks were not fixed. I had to exclude them, factory reset them, and re-pair them.  As always, this pairing is a bit finicky: the lock next to the controller, use the QRcode data before pairing for the Smart Start data, and it took two attempts.  In fact, the time that it paired was when I did not use the QR data while pairing, but instead went and re-enabled the existing entry in the table from my previous attempt. I wonder if that is the issue that I always have on first attempt.  I will try to check that theory when I fix the second lock. There was also some funkiness with refreshing the data for the User Codes. 

Many thanks @AlCalzone !

AlCalzone, 2023-01-25:
Actually the change should have been picked up with re-interviewing the device. But glad it works.

jds11111, 2023-01-25:
Yeah, I tried re-interviewing, but that actually made the situation worse.

AlCalzone, 2023-01-25:
Good old Schlage locks 💩

PR #5359 (2023-01-25): fix(config): disable Supervision for Schlage BE469ZP
fixes: #5260