zwave-js/zwave-js #5264: Kamstrup meter z-wave module not recognized

Thedexter1979, 2022-12-19:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
My best guess is that there’s no “driver” for the device?

What do you observe?
The device is discovered and linked, but not recognised 

What did you expect to happen?
I expect the device to Be recognized

Steps to reproduce the behavior:
  1. Go to: add the z-wave device
  2. Click on the added device
  4. See error

### Device information

Manufacturer: Kamstrup
Model name: Туре [68-50-061]
Node ID in your network: 

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwave-js-ui` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwave-js-ui-store.xlsx.zip](https://github.com/zwave-js/node-zwave-js/files/10256873/zwave-js-ui-store.xlsx.zip)

Thedexter1979, 2022-12-19:
> 👋 Hey @Thedexter1979!
> 
> It looks like you attached a logfile, but its filename doesn't look like it a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) that came from Z-Wave JS. Please make sure you upload the correct one.

[zwavejs_2022-12-19.log](https://github.com/zwave-js/node-zwave-js/files/10256911/zwavejs_2022-12-19.log)

Thedexter1979, 2022-12-19:
[image]
[image]
[image]

AlCalzone, 2023-01-30:
The interview is failing because the device reports support for Multi Channel CC v2, and then does not respond when that command is used. Also the ID on the Z-Wave Alliance website seems to be wrong.

I've got an updated config file [here](https://raw.githubusercontent.com/zwave-js/node-zwave-js/kamstrup-meter/packages/config/config/devices/0x0091/electricity_meter.json) for you to test using [these instructions](https://zwave-js.github.io/zwave-js-ui/#/device-support/custom-device-files).
We may have to do another iteration after you've applied that.

AlCalzone, 2023-03-06:
ping @Thedexter1979

closed 2023-07-20 by AlCalzone