zwave-js/zwave-js #5266: Multiple nodes showing up when adding device

tamorgen, 2022-12-20:
Good morning,
I’m trying to add a Zooz outdoor plug (Zen14) to my HA using Zwave-JS-UI. I scanned the QR code, plugged it in, and turned it on. All of a sudden I keep getting multiple nodes trying to add (up to 6 nodes now), and none of them complete. I’ve let it go for like 20 minutes now, and it’s just spinning at me being dumb. I’ve had this happen a few times, but just 1 or 2 extra inclusions, and they usually end up adding a final secure node.

I tried soft resetting Z-Wave-JS-UI, and I’ve tried restarting Home Assistant, but it just keeps spinning.

What is going on here?

I’m running HA 2022.10.5 (not upgrading to 11.1 until the bugs are worked out)
OS 9.3
RaspPi 4.
[image]

AlCalzone, 2022-12-20:
When I've seen this before, it was the device crashing and seemingly factory resetting in the middle of the interview, but only when included via SmartStart. That then triggers another inclusion and so on...

As a workaround just do a normal manual inclusion. I guess also report this to Zooz but I'm not sure is this is something they can fix themselves. since the same thing seems to happen across manufacturers, it might also be an SDK bug.

makopov, 2023-01-25:
> Also make sure your dongle is on an USB cord, away from an USB input
Where is this documented? I have mine connected directly to the box and am having this same issue, however mine appears to be paired successfully but then it keeps adding the same one ever 20s.

AlCalzone, 2023-01-25:
There: https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting

AlCalzone, 2023-08-18:
The original issue was fixed in v11.0.0: https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md#1100-2023-06-19--im-on-the-highway-to-hell-certification

This happens when including a device via SmartStart fails (e.g. due to communication issues or timeouts in the key exchange). The device then resets and tries again.
We now remove the remains of the failed inclusion automatically.

closed 2023-08-18 by AlCalzone