zwave-js/zwave-js #5270: Support MultiCommandCCCommandEncapsulation

mundschenk-at, 2022-12-21:
**Is your feature request related to a problem? Please describe.**
Currently commands using `MultiCommandCCCommandEncapsulation` are not handled.

**Describe the solution you'd like**
Multiple commands combined via `MultiCommandCCCommandEncapsulation` should be supported.

**Additional context**
```text
2022-12-21T07:17:40.048Z CNTRLR   [Node 104] [translateValueEvent: value updated]
                                    is root endpoint:        true
                                    is application CC:       true
                                    should hide root values: false
2022-12-21T07:17:40.051Z DRIVER « [Node 104] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 175
                                    │ extensions:
                                    │ · type: SPAN
                                    │   sender EI: 0x7a0df133621bad62beb810939cc886a8
                                    └─[MultiCommandCCCommandEncapsulation]
2022-12-21T07:17:40.054Z DRIVER   Received a command that contains multiple CommandClasses. This is not supporte
                                  d yet! Discarding the message...
2022-12-21T07:17:40.055Z CNTRLR « [Node 104] TODO: no handler for application command
```

AlCalzone, 2023-01-02:
Any chance you know what's supposed to be in that command?
"simple" encapsulated commands which just report a value are already supported. CCs that need special handling like `Notification CC` aren't at the moment.

mundschenk-at, 2023-01-02:
I'll ask Sensative engineers. In theory, this should only be several reported sensor values combined (temp, humidity, lux, the device has an option to switch MultiCommandCC off and on), but I don't know how they are encoding it exactly.

AlCalzone, 2023-01-02:
I see. It might very well be that the log output is a red herring and everything is already working as expected.

mundschenk-at, 2023-01-12:
I got a reply from Sensative:

- MultiCmdEncapsulation (MLSRT + MLSRH): Used when Temperature + Humidity Reporting enabled
- MultiCmdEncapsulation (CSN + MLSRT + NR): Used for Temperature Alarms (High\Low)
- MultiCmdEncapsulation (MLSRT + NR): Temperature Alarms Clear (High\Low)

Abbreviations:

- MLSR - MultiLevel Sensor Report
- T - Temperature
- H - Humidity
- NR - Notification (Alarm) Report
- CSN - Central Scene Notification

AlCalzone, 2023-01-12:
ok, Notification/Alarm and Central Scene must be handled specifically.

PR #5423 (2023-02-06): fix: handle all commands contained in a Multi Command CC
fixes: #5270

also adds logging for the Multi Command CC, so we can see what's in it.
closed 2023-02-06 by AlCalzone