zwave-js/zwave-js #5272: Parameter 16 (and several others although I only need 16) are not shown by the Z-wave driver for GE 14295 switch

u144175, 2022-12-21:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

What do you observe? Installed the subject smart dimmer switch and added to HA. I want this to work as a switch only, not a dimmer. In order to do so, I must set parameter 16 to 1. Parameter 16 is not listed by the z-wave driver in the UI. I did update the device's firmware to version 5.26 (original out-of-box firmware version was 5.21) as the change notes said this would provide parameter 16 however parameter 16 still is not shown. Was told by an HA developer:

"The parameters shown in the UI are defined by the Z-Wave driver, node-zwave-js: https://devices.zwave-js.io/?jumpTo=0x0063:0x4944:0x3130:0.0

You’ll need to create a [bug report](https://github.com/zwave-js/node-zwave-js/issues/new/choose) at that project for the new parameters to be added."

What did you expect to happen? Please add missing parameters for this device to the UI.

Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer: GE / Jasco
Model name: 14295 / ZW3004
Node ID in your network: 28

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: Driver Version: 10.3.0
Server Version: 1.24.0 

`node-zwave-js` branch:
`zwave-js-ui` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/10278470/zwave_js.log)

u144175, 2022-12-21:
[zwave_js(1).log](https://github.com/zwave-js/node-zwave-js/files/10278538/zwave_js.1.log)

blhoward2, 2023-04-08:
How about linking us to the GE documentation for this device and the change, please?