zwave-js/zwave-js #5276: NVMEDIT - ZWaveError: Unsupported protocol file format: 5 (ZW0280)   Firmware 7.19

dwaleke, 2022-12-27:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Trying to convert firmware version 7.19 from UZB7 stick NVM backup from bin to json.

What do you observe?

ZWaveError: Unsupported protocol file format: 5 (ZW0280)
    at nvmObjectsToJSON (/root/.npm/_npx/98595b58b0c35f8f/node_modules/@zwave-js/nvmedit/build/convert.js:117:23)
    at nvmToJSON (/root/.npm/_npx/98595b58b0c35f8f/node_modules/@zwave-js/nvmedit/build/convert.js:554:17)
    at Object.handler (/root/.npm/_npx/98595b58b0c35f8f/node_modules/@zwave-js/nvmedit/build/cli.js:41:40) {
  code: 280,
  context: undefined,
  transactionSource: undefined
}

What did you expect to happen?
I expected nvmedit to read the format and do the conversion.

Steps to reproduce the behavior:
npx @zwave-js/nvmedit nvm2json --in 700NVM-NVM_20221227161600.bin --out 700NVM-NVM_20221227161600.json

### Device information

Manufacturer: SILabs
Model name: UZB-7
[700NVM-NVM_20221227161600.bin.txt](https://github.com/zwave-js/node-zwave-js/files/10309603/700NVM-NVM_20221227161600.bin.txt)

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 10.3.1
`zwave-js-ui` branch: 8.6.1

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

npx @zwave-js/nvmedit nvm2json --in 700NVM-NVM_20221227161600.bin --out 700NVM-NVM_20221227161600.json

ZWaveError: Unsupported protocol file format: 5 (ZW0280)
    at nvmObjectsToJSON (/root/.npm/_npx/98595b58b0c35f8f/node_modules/@zwave-js/nvmedit/build/convert.js:117:23)
    at nvmToJSON (/root/.npm/_npx/98595b58b0c35f8f/node_modules/@zwave-js/nvmedit/build/convert.js:554:17)
    at Object.handler (/root/.npm/_npx/98595b58b0c35f8f/node_modules/@zwave-js/nvmedit/build/cli.js:41:40) {
  code: 280,
  context: undefined,
  transactionSource: undefined
}

AlCalzone, 2022-12-27:
Thanks for the report. I'll have to figure out what changed in 7.19

AlCalzone, 2023-02-16:
fixed in https://github.com/zwave-js/node-zwave-js/releases/tag/v10.10.0

closed 2023-02-16 by AlCalzone