zwave-js/zwave-js #5279: Problems with 6th Allegion BE469, BE469Z

ornative, 2022-12-29:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

8.6.1.7833cd8

### ZwaveJS version

10.3.1

### Describe the bug

I have added 4 - Allegion BE469Z, and 1 - BE469 locks with ZwaveJs-UI directly. But the 5th will not add correctly.  Some background:
   
   The 5th lock when doing inclusion ends without being included, according to ZwaveJS, however when you then check, it has been added with no security and most of the parameters missing to configure the lock and it indicates it does not recognize the device.  After fighting with it, I called Schlage and they sent a replacement after receiving the logs.  However, the replacement behaves the same... I have a hard time believing two locks would malfunction this way given the firmware is the same as the rest.

I will include the email text and files sent to Schlage (Allegion).  Let me know if you need additional information.

### To Reproduce

How to reproduce is in the 'describe the bug' section.  Install the 4 locks + the one older model and then try to add another BE469Z.

### Expected behavior

The 6th device should add correctly when included.  

### Additional context

Just to be clear I have more of these locks that I don’t have issues with.

1.	The lock excludes correctly; it flashes orange a couple times then green and the UI states exclusion complete.  The exclusion log is included. I’ve done exclusion several times now after each inclusion. After exclusion I’ve also done factory resets in some cases. See exclusion.log file.
2.	The lock is being added through inclusion from the Zwavejs instance, the same as the other locks. But won’t do S2 and fails to get full interview parameters.  See inclusion.log file.
3.	Home Assistant is the Hub or should I say UI that is controlling the lock, with Zwavejs as the primary component that interacts directly with the lock. I do not add my Zwave devices through Home Assistant, as it seems inconsistent.

To summarize, I have 5 BE469ZP on exterior doors.  4 work fine and show all parameters in ZwaveJs and use S2 security (they all work reliably) and the 5th refuses to include properly.  I also have a BE469 (earlier model) on a interior door that uses Legacy S0 find and it also works fine.  I’ve included debug files from ZwaveJS for two locks, one that works (Patio Door) and one that does not (Front Door).

This lock does not report all parameters to the Zwave interview and refuses to do S0 or S2.  

Current Environment:

-	Rpi 4B – 4 GB memory
-	Docker Container Community Edition
o	Homeassistant 2022.11.2
o	ZwaveJS: 10.3.0
-	Aeotec 700 Series ZST10-700  Firmware v7.17.2 SDK v7.17.2 installed on extender cable to prevent interference from USB-3
-	Front Door Lock BE469ZP (failing lock) located within 10 ft of ZST10-700 ; although I have had it sitting next to the ZST10-700 during troubleshooting process.
-	Patio Door Lock BE469ZP (working lock) located 20 ft of ZST10-700
[debug BE469ZP - Patio.json.txt](https://github.com/zwave-js/zwave-js-ui/files/10316311/debug.BE469ZP.-.Patio.json.txt)
[exclusion.log.txt](https://github.com/zwave-js/zwave-js-ui/files/10316312/exclusion.log.txt)
[inclusion.log.txt](https://github.com/zwave-js/zwave-js-ui/files/10316313/inclusion.log.txt)
[debug BE469ZP - Front Door.json.txt](https://github.com/zwave-js/zwave-js-ui/files/10316314/debug.BE469ZP.-.Front.Door.json.txt)

ornative, 2023-01-05:
I also tried to add through provisioning.  I imported the QR code (my pi camera just can't focus on some of these) and it provisioned with the device manufacturer, device type, description, etc.  When I powered up the lock it included and was s2 access controlled.  However, the manufacturer, device type, description, as well as many of the control parameters are still missing.

If you ne
[image]
[image]
ed any additional information, let me know.

AlCalzone, 2023-01-06:
I'm going to need a driver log of the complete inclusion process, yours cuts off after a second.

ornative, 2023-01-06:
log started from saving debug settings till after the node adds incorrectly.
[zwavejs_2023-01-06.log](https://github.com/zwave-js/node-zwave-js/files/10362498/zwavejs_2023-01-06.log)

ornative, 2023-01-06:
Added logging information in comment after this request.

ornative, 2023-01-08:
I had an older Smartthings Connect Home Hub and added the lock to it to test.  While I can't see all the parameters, I was able to add it with S2 both with QR Code and with the DSK pin and it operated normally.  I wanted to make sure it wasn't a lock problem and it does not appear to be.

ornative, 2023-01-08:
I setup a spare pi 4b with same os version, same Aeotec gen 7 Zwave-controller and the same version of zwavejs-ui.  It included the BE469Z I've been having problems with, immediately and correctly.  The only difference is that this instance has no other Zwave devices on it.  My production pi has 65 other devices.

AlCalzone, 2023-01-09:
The log you posted seems to happen after there was already a problem during the first interview, which lead to communicating with the device without encryption. You'll have to re-interview again, this time check the checkbox for "reset security classes". Oh and please lower the loglevel from "silly" to "debug".

ornative, 2023-01-09:
Actually, that log was done from a clean inclusion (I have tried a re-interview doing the reset in the past and it did not help at all).  Cleared failed node after unplugging and pinging, factory reset lock.  Start log, run inclusion, stop after zwavejsui said inclusion failed.  Then still adding the node.  One log is done with devpi (added to name of file), which is the one with no nodes on it, start inclusion, lock flashes green, and is included with S2 relatively quickly.  The second one is done on homepi (production node) the same way, start logging to file, run inclusion, lock flashes red, zwavejsui continues the countdown until it is zero, then shows the stop inclusion banner, then eventually stops with   "timed out! no device has been found to complete inclusion" yet, it did include the device as the 213th node.  no manufacturer, no device name, no S2 (it never asked for the DSK, as where the devpi asks almost immediately).

On the homepi I had to be very careful to wait for the browser display to finish updating after turning on logging, the first time I didn't wait and it timed out almost immediately with the log flashing red and zwavejsui saying it failed immediatly.  This leads me to start thinking there is an asynchronous  timing problem (pi too busy?  race condition?).  The log here does not contain that interaction, as I send /dev/null to the log file and restarted logging, waited until the display sync'd and then started inclusion.

Clearly, given the lack of consistent results in these scenarios, there is something going on that is causing these issues.  

[zwavejs_2023-01-09_homepi.log](https://github.com/zwave-js/node-zwave-js/files/10375406/zwavejs_2023-01-09_homepi.log)
[zwavejs_2023-01-09_devpi.log](https://github.com/zwave-js/node-zwave-js/files/10375408/zwavejs_2023-01-09_devpi.log)
ufacturer, no device name, no S2 (it never asked for the DSK, as where the devpi asks almost immediately).

AlCalzone, 2023-01-09:
Just judging from your log, the homepi network seems very busy. Looks like there are at least 5-10 interviews going on and some commands take a long time to finish:
```
transmit status:        NoAck, took 12030 ms
```
Even the low-level inclusion process which normally takes less than 5s took 1m40s.
With that amount of traffic, inclusions (and the following interview) will have a hard time.

I suggest you wait until those interviews are done, then include the lock again.

AlCalzone, 2023-01-09:
Not sure why they are happening.

> So how will I know when they complete?

[image]

;)

ornative, 2023-01-10:
I have installled a second Aeotec (had a spare) gen 7 zwave stick, setup a second docker instance and added the lock successfully.  I've not dug into the specs on that zwave stick, but it could be that the processor is too slow for this many devices or the cache is too small to handle all the traffic.  I know the gen 8 hardware is supposed to be 20% faster, but that may not be enough to go much further than I already have reliably.  I've had problems with devices showing dead as well and having to ping them back to life. 

So it doesn't like the combination of devices I have (65 total), I noticed some running at 100 kbps and some at 40 kbps and very few of them appear to tolerate slowness on the receiver  - particularly locks and covers.  Clearly the protocol is closer to UDP than TCP!.

I am going to move some things to the new stick and see how things go, that is whether or not this is a fix.  Also hoping some changes come in the newer version (9?).  Will have to look at matter, but given I was a Network Engineer for years I know the pitfalls of IP as well.  As it is I have 5 networks on separate VLANs with extensive rules in a Opnsense firewall.  Having additional IP nodes does not install me with faith...Argh.

AlCalzone, 2023-01-10:
The main bottleneck in Z-Wave is the physical world (RF, interference, noise, and duty cycles). Even the "slow" 500 series sticks are able to handle 232 nodes.

ornative, 2023-01-10:
Two things.  First thanks for taking a look and replying.

1.  I don't base my requirements on theoretical numbers, which that appears to be (I know the standard but have not seen it work in a real-world environment).
2.  Most of what I've installed are Zwave+ and sure, I get the interference thing, but none the less my practical experience tells me (along with a lot of posts) that getting anywhere near 232 devices you have to be in an 800 sq foot house with nothing else going on.

I can tell you that if our house has that much interference, etc. particularly where devices are showing dead, nobody can get this to work at that level (just review forums and you will find a lot of people giving up on Smartthings or Zwave just because of this issue).   Most of the time I see the death problem is when I restart zwavejsui; several devices show up dead.  I'm not saying interference, et. al. isn't a factor but given my topology it should not be the case (if it is, then zwave needs a complete re-factor). 

Either way I'm not going any further with zwave at this point until I can come up with a configuration that is more stable.

closed 2023-01-21 by zwave-js-assistant[bot]