zwave-js/zwave-js #5281: aerQ temperature and humidity sensor v2 (ZWA039) config parameters incorrect ranges

yycmikeb, 2022-12-31:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug? Not a bug. Ranges in the config file don't match the manufacturer's stated ranges.

What do you observe? Incorrect ranges for parameters 5 and 6

Steps to reproduce the behavior:
  1. View "Configuration v4" section for the device node

Parameter 6 ("Low Temperature Basic Set Threshold") is listed as having a range of 200 - 1000
--> range should be -200 - 1000 (lowest possible minimum low temp alarm setting is -20.0 C/F)

Parameter 5 ("High Temperature Basic Set Threshold") is listed as having a range of 0 - 1000
--> range should be 1 - 1000

For reference, see the "Configuration Parameters" section of Aeotec user guide for this device at:
https://aeotec.freshdesk.com/support/solutions/articles/6000227918-a%C3%ABrq-temperature-and-humidity-sensor-user-guide

### Device information

Manufacturer: Aeotec
Model name: aerQ Temperature and Humidity Sensor V2.0, model ZWA039-A
Firmware version: 2.1.1
Config file: https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0371/zwa039.json

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 10.3.1

From "info" icon in Z-Wave JS UI:
zwave-js: 10.3.1
zwave-js-ui: 8.6.2.faa2543

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

Not provided as this is a config file issue. Device is working and communicating properly.

AlCalzone, 2023-01-02:
> Parameter 6 ("Low Temperature Basic Set Threshold") is listed as having a range of 200 - 1000
> --> range should be -200 - 1000 (lowest possible minimum low temp alarm setting is -20.0 C/F)

Yeah, that's likely an oversight on our part.

> Parameter 5 ("High Temperature Basic Set Threshold") is listed as having a range of 0 - 1000
> --> range should be 1 - 1000

That's not true though, 0 is a valid value:
[image]

yycmikeb, 2023-01-02:
Fair enough - I wasn't sure how range and default values interacted. Thanks for addressing!

closed 2023-01-02 by yycmikeb