zwave-js/zwave-js #5291: Shenzhen Neo Electronics Co (0x0258)

esszopo, 2023-01-02:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Shenzhen Neo Electronics Co Ltd / Nas-PD03

### Manufacturer ID

0x0258

### Product Type

0x0200

### Product ID

0x1036

### Firmware Version

3.3

### Do you have a manual?

[Gescand document.pdf](https://github.com/zwave-js/node-zwave-js/files/10332359/Gescand.document.pdf)

esszopo, 2023-01-04:
Hi,
I've made a nas-pd05z.json file that more or less works. The problem I'm having with it is:
it has 16 parameters of these I can't get the 1th five working. When browsing online in zwave-js-ui config params these 5 are not filled in with default/queried values and are listed at the end off all params. All 5 params are one byte in size the 6th that's filled in is 2 bytes. Any help would be nice.

esszopo, 2023-01-04:
[test.zip](https://github.com/zwave-js/node-zwave-js/files/10344226/test.zip)

hope this helps

AlCalzone, 2023-01-04:
The device responds with invalid commands when querying the parameters 1 to 5:
[image]

Either the parameters don't exist and it tries to communicate that fact this way, or this is a firmware bug.

esszopo, 2023-01-04:
Ok,
I'll contact the seller, at least my config file was coded ok.

AlCalzone, 2023-01-04:
There are a few things that don't match our style guide but at least no obvious errors.

esszopo, 2023-01-04:
I was trying to make it working without syntactic errors, that was hard enough. My plan was to clean it up later, but then this happened. I've seen an other xml file and it starts at param 6 I think that's a quick bug fix because you want to have the option of tuning the sensitivity.

esszopo, 2023-01-07:
I've left a problem at the site of Neo but it's a chinese site so I'm not very optimistic I'll get a new firmware. Do you want me to clean up the config json if so you'll have to tell me what's wrong with it, I don't have any knowledge of you standards. I suppose we have 2 options go the openhab way and delete the first 5 options or leave them in for when they happens to fix the firmware. If you want to cleanup the file yourself I think you should rename it to a nas-zp03z?.json file because I think it's a zp03 and not a zp05.

AlCalzone, 2023-01-09:
I have no hopes either, if you consider the glaring bugs in some of their devices. We can remove params 1-5 as they are not working.
As for correcting the file: Please check if https://zwave-js.github.io/node-zwave-js/#/config-files/style-guide helps you understand some of our rules, especially the [part about labels](https://zwave-js.github.io/node-zwave-js/#/config-files/style-guide?id=labels). If not, I can give you a detailed review.
Also if you `$import` something from a template and continue to override most of it, either the import is unnecessary, or some of the overwritten values.

esszopo, 2023-01-27:
hi,
I've finally made the config file. I've left the first 5 options in it so that should the unlikely happen that they fix the firmware you have a config file that can be used, it's up to you if you rip them out or not.

Side note I've duwi remote zwave controller that is configured as a slave zwave controller to my main zwave controller, zwave-js keep trying to interview this controller while it cannot be interviewed because it's a controller. I've tried what would happen if I do a reconnect but then zwave-js does a hiccup and gives it the same values as the next entry in the configuration. If you want to check out the logging I send you it's the device with id: 3.  

[device.zip](https://github.com/zwave-js/node-zwave-js/files/10519394/device.zip)

closed 2023-02-06 by esszopo