zwave-js/zwave-js #5292: Missing updates from Sunricher Knob Smart Dimmer

Martijn02, 2023-01-02:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I have 3 Sunricher Knob Smart Dimmers, All 3 are bought in one go, the have exactly the same firmware, same settings, same associations, etc.
One (node 20) has been connected to my zwave network for a slightly longer time, and works perfectly. If I turn the dimmer on and off, it's state changes are reflected in both zwave-js-ui and home-assistant.

The two other nodes don't seem to be sending updates, or updates are not being processed correctly. If I switch using home-assistant, or the zwave-js-ui to toggle the device, all is fine, but if I use the physical knob, the light comes on or goes off, but the updated state does not come through. If I trigger the refreshCCvalues option in zwave-js-ui manually, I see the status update.

I have tried a lot. Re-interviewing, excluding and re-including the device, factory resetting the device, and then including it again, all using different versions of the zwave-js-ui docker package. I also tried both with and without security enabled.

I tried following the steps in https://zwave-js.github.io/node-zwave-js/#/troubleshooting/no-updates
- The device does support Multi Channel Association CC
- Lifeline association is set up
[image]
- The logs do not show anything when tuning the lamp on or off using the button

### Device information

Manufacturer: Sunricher
Model name: Knob Smart Dimmer (ZV2835RAC-NF)
Node ID in your network: 33

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

`zwave-js-ui`: 8.6.2.faa2543
`zwave-js`: 10.3.1

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

I have one node that still works perfectly, and 2 others that no longer work. I think they used to work fine, and stopped working after I excluded and re-included them without security as I attempted to set up associations with an unsecured device. 

I don't know which exact version did work (and I'm unable to find out) but I made a clean start of my HA in my new home at the beginning of Nov 2022. I think it broke beginning of December 2022.

### Attach Driver Logfile

[zwavejs_re-interview-plus-switch-on-off.log](https://github.com/zwave-js/node-zwave-js/files/10333294/zwavejs_re-interview-plus-switch-on-off.log)
[zwavejs_full.log](https://github.com/zwave-js/node-zwave-js/files/10333295/zwavejs_full.log)
[node_33.json.txt](https://github.com/zwave-js/node-zwave-js/files/10333313/node_33.json.txt)

AlCalzone, 2023-01-04:
If the associations and configuration is identical, I see 2 options:
- Perform a heal on Node 33, maybe it has no route to the controller, so it cannot report. I see no dropped packets in the log.
- Re-include it without encryption and see if that changes things. Node 20 is included insecurely, and it does report.

Martijn02, 2023-01-04:
Thank you very much for looking into this. I did multiple heals on node 33, that didn't change anything. I also tried without encryption before, but I have just excluded it, and re-included the device without security. It's now node 43. Unfortunately the issue remains.

If I turn the device on of off, or change the dimming level using the knob on the device, I do not see any traffic in the log, and the state in zwave-js-ui and home assistant is not updated until I poll the device by doing a refreshCCvalues in zwave-js-ui.

```
2023-01-04T22:20:29.237Z CNTRLR   [Node 034] Healing node...
2023-01-04T22:20:29.238Z CNTRLR » [Node 034] refreshing neighbor list (attempt 1)...
2023-01-04T22:20:29.255Z SERIAL » 0x01050048221888                                                     (7 bytes)
2023-01-04T22:20:29.257Z DRIVER » [Node 034] [REQ] [RequestNodeNeighborUpdate]
                                    callback id: 24
2023-01-04T22:20:29.259Z SERIAL « [ACK]                                                                   (0x06)
2023-01-04T22:20:29.376Z SERIAL « 0x0105004818218b                                                     (7 bytes)
2023-01-04T22:20:29.380Z SERIAL » [ACK]                                                                   (0x06)
2023-01-04T22:20:29.382Z DRIVER « [REQ] [RequestNodeNeighborUpdate]
                                    callback id:   24
                                    update status: UpdateStarted
2023-01-04T22:20:30.042Z SERIAL « 0x0105004818218b                                                     (7 bytes)
2023-01-04T22:20:30.043Z SERIAL » [ACK]                                                                   (0x06)
2023-01-04T22:20:30.047Z DRIVER « [REQ] [RequestNodeNeighborUpdate]
                                    callback id:   24
                                    update status: UpdateStarted
2023-01-04T22:20:33.639Z SERIAL « 0x01050048182288                                                     (7 bytes)
2023-01-04T22:20:33.641Z SERIAL » [ACK]                                                                   (0x06)
2023-01-04T22:20:33.644Z DRIVER « [REQ] [RequestNodeNeighborUpdate]
                                    callback id:   24
                                    update status: UpdateDone
2023-01-04T22:20:33.658Z CNTRLR « [Node 034] neighbor list refreshed...
2023-01-04T22:20:33.659Z CNTRLR » [Node 034] Deleting SUC return route...
2023-01-04T22:20:33.679Z SERIAL » 0x01050055221994                                                     (7 bytes)
2023-01-04T22:20:33.681Z DRIVER » [Node 034] [REQ] [DeleteSUCReturnRoute]
                                    payload: 0x2219
2023-01-04T22:20:33.684Z SERIAL « [ACK]                                                                   (0x06)
2023-01-04T22:20:33.691Z SERIAL « 0x0104015501ae                                                       (6 bytes)
2023-01-04T22:20:33.692Z SERIAL » [ACK]                                                                   (0x06)
2023-01-04T22:20:33.694Z DRIVER « [RES] [DeleteSUCReturnRoute]
                                    was executed: true
2023-01-04T22:20:33.784Z SERIAL « 0x010500551900b6                                                     (7 bytes)
2023-01-04T22:20:33.785Z SERIAL » [ACK]                                                                   (0x06)
2023-01-04T22:20:33.785Z DRIVER « [REQ] [DeleteSUCReturnRoute]
                                    callback id:     25
                                    transmit status: OK
2023-01-04T22:20:33.792Z CNTRLR » [Node 034] Assigning SUC return route...
2023-01-04T22:20:33.799Z SERIAL » 0x01050051221a93                                                     (7 bytes)
2023-01-04T22:20:33.799Z DRIVER » [Node 034] [REQ] [AssignSUCReturnRoute]
                                    payload: 0x221a
2023-01-04T22:20:33.801Z SERIAL « [ACK]                                                                   (0x06)
2023-01-04T22:20:33.811Z SERIAL « 0x0104015101aa                                                       (6 bytes)
2023-01-04T22:20:33.812Z SERIAL » [ACK]                                                                   (0x06)
2023-01-04T22:20:33.813Z DRIVER « [RES] [AssignSUCReturnRoute]
                                    was executed: true
2023-01-04T22:20:36.121Z SERIAL « 0x010500511a00b1                                                     (7 bytes)
2023-01-04T22:20:36.125Z SERIAL » [ACK]                                                                   (0x06)
2023-01-04T22:20:36.127Z DRIVER « [REQ] [AssignSUCReturnRoute]
                                    callback id:     26
                                    transmit status: OK
2023-01-04T22:20:36.137Z CNTRLR » [Node 034] deleting return routes (attempt 1)...
2023-01-04T22:20:36.148Z SERIAL » 0x01050047221b84                                                     (7 bytes)
2023-01-04T22:20:36.149Z DRIVER » [Node 034] [REQ] [DeleteReturnRoute]
                                    payload: 0x221b
2023-01-04T22:20:36.152Z SERIAL « [ACK]                                                                   (0x06)
2023-01-04T22:20:36.161Z SERIAL « 0x0104014701bc                                                       (6 bytes)
2023-01-04T22:20:36.162Z SERIAL » [ACK]                                                                   (0x06)
2023-01-04T22:20:36.163Z DRIVER « [RES] [DeleteReturnRoute]
                                    has started: true
[... cut ...]
2023-01-04T22:20:38.820Z SERIAL » [ACK]                                                                   (0x06)
2023-01-04T22:20:38.821Z DRIVER « [REQ] [AssignReturnRoute]
                                    callback id:     28
                                    transmit status: OK
2023-01-04T22:20:38.827Z CNTRLR   [Node 034] healed successfully
```
[node_34.json.txt](https://github.com/zwave-js/node-zwave-js/files/10347883/node_34.json.txt)

Is there anything else I can do/try in order to get the issue resolved?

AlCalzone, 2023-01-05:
Please double-check the associations that they are truly identical on both devices. I guess there's not much to get wrong here though - node 1 in group 1...

The manual refers to this parameter, but I also think this only affects the state after a power loss:
[image]

Last but not least, you could try excluding, factory resetting and then including again:
> Press and hold down the reset button for over 10 seconds, the dimmer will be reset to factory defaults, the load
brightness will be restored to 50% and will flash slowly.

Other than that I don't see anything wrong. If nothing here works, I'm afraid this is a case for the manufacturer support.

Martijn02, 2023-01-19:
I still have no clue what was causing the issue, but I eventually just bought another dimmer, and that one works fine. After that I put the Sunricher dimmer on my test bench (read: my desk) and for some reason it works fine now. Without resetting, without even re-joining the zwave network, it just started to work. Maybe removing it from power, and re-powering it did the trick? I don't know, and I'm afraid there is no way to find out.

Anyway: thank you for your efforts into looking in to this.

closed 2023-01-19 by Martijn02