zwave-js/zwave-js #5298: Ring Retrofit Alarm Kit

webmogul1, 2023-01-03:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Ring Retrofit Alarm Kit

### Manufacturer ID

0x0346

### Product Type

0x0b01

### Product ID

0x0101

### Firmware Version

1.6

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/3696?selectedFrequencyId=-1

### Do you have a manual?

[Retrofit Alarm Kit Installation Manual.pdf](https://github.com/zwave-js/node-zwave-js/files/10338371/Retrofit.Alarm.Kit.Installation.Manual.pdf)
[Ring Retrofit Alarm Kit Zwave Technical Manual.pdf](https://github.com/zwave-js/node-zwave-js/files/10338372/Ring.Retrofit.Alarm.Kit.Zwave.Technical.Manual.pdf)

AlCalzone, 2023-01-03:
@zwave-js-bot import config 3696 from zwa

webmogul1, 2023-01-03:
Unfortunately, I have no idea how to do it.

AlCalzone, 2023-01-04:
Don't worry, I just did it

closed 2023-01-04 by AlCalzone

webmogul1, 2023-01-04:
> Don't worry, I just did it

Wonderful! And how do I add the new config?

AlCalzone, 2023-01-04:
You wait for the next update

webmogul1, 2023-01-04:
Thank you.

The info below is regarding the version without the new configuration. I don't know if the new one will solve the issues.
Maybe you can help having the information below.

It looks like there is a problem with the multi association thingy. (I am totally new to this)

I can get notifications when cover is removed. When zone 1 changes state I get the "Intrusion Notification" and that's it.
Changes in state of of zones 2 through 8 do not give me notifications. 

This is my notification table.
[image]

This is my group table
[image]

Another fellow reports this:
https://community.home-assistant.io/t/ring-retrofit-hardwire-kit/430619
His group table looks like this:
[image]

But I can't add other endpoints.
[image]

PR #5299 (2023-01-04): feat(config): add Ring Retrofit Alarm Kit
fixes: #5298

**TODO:**
- [x]	Change PR title to be more specific