zwave-js/zwave-js #5302: Unable to get Ring Retrofit device working properly

webmogul1, 2023-01-05:
Hi 
Is this something you can help me with?

Can't get it to work.

https://products.z-wavealliance.org/products/3696?selectedFrequencyId=-1

AlCalzone, 2023-01-05:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).
While that log is running, re-interview the device. Based on what you posted in the other issue, I suspect we may either not need the association definitions in the config file or we'll have to extend them.

webmogul1, 2023-01-06:
I will get the logs soon. In the mean time:
It looks like there is a problem with the multi association thingy. (I am totally new to this)

I can get notifications when cover is removed. When zone 1 changes state I get the "Intrusion Notification" and that's it.
Changes in state of of zones 2 through 8 do not give me notifications.
This is my notification table.
[image]
This is my group table
[image]

Another fellow reports this:
https://community.home-assistant.io/t/ring-retrofit-hardwire-kit/430619
His group table looks like this:
[image]

But I can't add other endpoints.
[image]

webmogul1, 2023-01-06:
Is this one that you need?
[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/10357622/zwavejs_current.log)

AlCalzone, 2023-01-06:
Ok I think I know what the problem is. The driver attempts to communicate with the endpoints without S2-encryption and gets no response for its queries. I'll have to figure out why it does that.

webmogul1, 2023-01-23:
Unfortunately, It didn't work. 
I still cannot change endpoints 1-8 associations.

[image]

zwave-js-ui: 8.8.0
zwave-js: 10.4.0

webmogul1, 2023-01-23:
Device in question is Node 35

[zwavejs_current (2).log](https://github.com/zwave-js/node-zwave-js/files/10484220/zwavejs_current.2.log)

webmogul1, 2023-01-23:
https://support.ring.com/hc/en-us/articles/360038043371-Retrofit-Alarm-Kit-Z-Wave-Technical-Manual-

Manual says:
Association
This sensor has two Association groups with only one node for each group. Group one is a lifeline group who will receive unsolicited messages relating to case tamper notifications, battery level notifications, Wake-Up notifications, and Device Reset Notifications. Group two is intended to be used with multi-channel association for a controller to subscribe to contact switch open/close notifications sent encapsulated from each of the eight endpoints one for each contact switch. Each endpoint must be enabled by putting the controller’s node ID into the device’s Association Group 2 with either an Association Set encapsulated with Multi-Channel Encapsulation OR a Multi-Channel Association Set encapsulated with Multi-Channel Encapsulation.

AlCalzone, 2023-01-23:
hmm doesn't look like the fix did what it was supposed to.

webmogul1, 2023-01-24:
Does it matter that the device is showing
Only S2 Authenticated
instead of S2 Authenticated, S2 Unauthenticated
as Security classes?

webmogul1, 2023-01-25:
Hope that helps
Here is the thing. I added the device using the "Z-Wave PC Controller" from SiliconLabs.
It seems that I got the correct associations showing in the controller. It shows a lot more items than when I added with ZwaveJS.

[image]

[image]

This is when it is added with zwaveJS

[image]

webmogul1, 2023-01-25:
But then if I connect the stick to zwavejs, the node is interviewed again and there are no options for the other endpoints.

I have no idea how that works but isn't just a matter of reading the node info already included?

AlCalzone, 2023-01-26:
I know what the issue is, I don't yet know why it happens. Unfortunately, my working hours are pretty limited atm. Give me a bit of time to reproduce and solve it.

webmogul1, 2023-01-26:
Ok. Just wanted you to have as much info as possible.

PR #5310 (2023-01-06): fix: always use S2 for endpoint communication if node uses S2
fixes: #5302
closed 2023-01-31 by AlCalzone