zwave-js/zwave-js #5303: Status changes done by primary controller (Fibaro HC3) do not get picked up by secondary controller (Zwave JS UI)

JLindhout, 2023-01-05:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

8.6.2

### ZwaveJS version

10.3.1

### Describe the bug

Dear all,
I am using 2 zwave controllers: 
1) Fibaro HC 3 
2) Zwave JS UI using Aeotec Gen 5+ stick 

Zwave JS UI communicates to domoticz using MQTT (mosquitto on a dedicated raspberry pi). 
All works fine, just that changes made on the Fibare HC3, do not get cascaded to domoticz. It looks like changes in the network, not done by the 2nd-ary controller are not picked-up. 

### To Reproduce

Tested by changing multiple devices and different brands.

### Expected behavior

A status change to cascade through from primary, to secondary controller.

JLindhout, 2023-01-05:
In my case: 
- Fibaro HC3 is the primary controller 
- Z-Wave JS is the secondary controller. 

Changes made by Fibaro appear not to be picked up. But if a secondary controller is not yet supported, that would explain it of course. 
k.rgds. Jeroen

AlCalzone, 2023-01-06:
Right, that's part of the issue. The other part is that many devices only support a single lifeline association, which is where they report updates to. This is usually the primary controller, meaning the secondary controller has no way of getting these updates.
If you control devices from Z-Wave JS and it receives no update in response, it will query the status. It cannot do that if it doesn't know there might have been a change.