zwave-js/zwave-js #5313: SES 303 misbehaving (with ses002 pipe sensor attached)

GSV3MiaC, 2023-01-08:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug? I have attached a SES303 temp/humidity sensor to the system, with a pipe sensor attached. The pipe sensor data is being updated (but only every 30 mins, despite me setting the parameter to ask for 5 mins), but the initial temp/humidity are not showing up except at include, or re-interview, time

What do you observe?

What did you expect to happen?

Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer: SES
Model name:  303 with pipe sensor
Node ID in your network:  50

{
  "id": 50,
  "name": "Water Tank",
  "loc": "Airing Cupboard",
  "values": [
    {
      "id": "50-112-0-2",
      "nodeId": 50,
      "commandClass": 112,
      "commandClassName": "Configuration",
      "endpoint": 0,
      "property": 2,
      "propertyName": "Temp Reporting Interval",
      "type": "number",
      "readable": true,
      "writeable": true,
      "description": "Reporting interval in minutes",
      "label": "Temp Reporting Interval",
      "default": 5,
      "stateless": false,
      "commandClassVersion": 1,
      "min": 0,
      "max": 255,
      "unit": "Mins",
      "list": false,
      "value": 5,
      "newValue": 5
    },
    {
      "id": "50-112-0-1",
      "nodeId": 50,
      "commandClass": 112,
      "commandClassName": "Configuration",
      "endpoint": 0,
      "property": 1,
      "propertyName": "Delta Temp",
[... 1940 lines cut ...]
### Which branches or versions?

version:
`node-zwave-js` branch:
`zwave-js-ui` branch:

zwave-js-ui: 8.0.1
zwave-js: 10.2.0
home id: 3619703272
home hex: 0xd7c049e8

### Did you change anything?

yes (please describe)

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwave-js-ui-store (1).zip](https://github.com/zwave-js/node-zwave-js/files/10368218/zwave-js-ui-store.1.zip)
[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/10368224/zwave-js-ui-store.zip)

Looks to me from the second (.1) log that the endpoint for the humidity/temp are showing as 0, should be 1?

GSV3MiaC, 2023-01-08:
I uploaded the zipped versions, the unpacked ones are huge!

GSV3MiaC, 2023-01-08:
[zwavejs_2023-01-07.log](https://github.com/zwave-js/node-zwave-js/files/10368263/zwavejs_2023-01-07.log)

GSV3MiaC, 2023-01-08:
I'm not the expert, but looking at the log the temp/humidity ARE being received every 5 mins now, but as endpoint 0, not 1. the pipe sensor (endpoint 2) only comes every 30 mins. Maybe there is separate config params to send for that endpoint?

robertsLando, 2023-01-09:
@AlCalzone @blhoward2 by checking the manual seems there are some configuration parameters missing, each endpoint should have it's configuration timeout and there only one is listed

robertsLando, 2023-01-09:
@GSV3MiaC Please could you tell us the device id? Once you click and expand a node in control panel table it's the first title you see (also click on the link if you want and share the link to device db)

AlCalzone, 2023-01-09:
@GSV3MiaC I've never seen a device where individual endpoints support Configuration CC separately. Can you get confirmation that parameters 1 and 2 actually need to be configured on each endpoint?

This won't be an easy fix though, since a lot in zwave-js assumes Configuration CC only exists on the node, not on endpoints.

robertsLando, 2023-01-09:
@AlCalzone I don't think each endpoint supports Configuration CC separately, I just think the actual device configuration is missing some configuration for those values

robertsLando, 2023-01-09:
Oh in that case it could be, yes

GSV3MiaC, 2023-01-09:
Device is

89-3-13 (0x0059-0x000d-0x0003)

Link goes to

https://devices.zwave-js.io/?jumpTo=0x0059:0x000d:0x0003:0.0

⁣Rgds, Dave Storey​

GSV3MiaC, 2023-01-09:
Well that's what the docs say, and the support guy from vesternet (seller) told me same thing. I wonder if just sending the params to all endpoints would fix it. I would settle for them all the same (given I am only using 1 and 2). 

I do have a support contact at ses that I have been talking to, let me know if you want their email address, though i am not sure how zwave clued they are. 

⁣Rgds, Dave Storey​

closed 2023-06-19 by AlCalzone