zwave-js/zwave-js #5318: Zooz ZEN32 - add parameter 24

webmogul1, 2023-01-10:
A new parameter was added with firmware 10.30 (released 7/2022)

Firmware: 10.30
Added parameter 24 to disable scene control for the relay button and reduce the physical press delay for that button with scene control disabled

Parameter 24: Enable or disable scene control on the relay button. If scene control is disabled, the slight delay for physical press will be reduced but the relay button will no longer send central scene events to the hub. It's recommended to disable the feature only if using the relay button for direct electrical control of the connected load without any additional remote control functions.

Values: 0 – scene control enabled (default). 1 – scene control disabled.

Size: 1 byte dec

webmogul1, 2023-01-13:
I think I did it right.

https://github.com/zwave-js/node-zwave-js/pull/5329

AlCalzone, 2023-01-14:
Let's keep this open until the PR is merged

closed 2023-03-08 by AlCalzone