zwave-js/zwave-js #5322: Zooz ZSE40 - add parameters

webmogul1, 2023-01-11:
ZSE40 700 (released 3/2022)
Firmware: 1.10

Firmware version changed to 1.10 to follow new naming convention for device firmware files

Added parameters 9 and 10 for temperature and humidity offset settings 

Basic Set Reports
Parameter 8: Disable Basic Set reports sent to the hub when motion is detected to reduce Z-Wave activity from the sensor.

Values: 

0 – Basic Set reports disabled.

1 – Basic Set reports enabled (default).

Size: 1 byte dec

Temperature Offset
Parameter 9: Set the reporting offset on your temperature sensor.

Values: 0 - 200 (degrees Fahrenheit, where value 0 equals -10 degrees, value 100 equals 0 degrees, and value 200 equals +10 degrees).

Default: 100.

Size: 1 byte dec

Humidity Offset
Parameter 10: Set the reporting offset on your humidity sensor.

Values: 0 - 200 (%, where value 0 equals -10%, value 100 equals 0%, and value 200 equals +10%). Default: 0.

Default: 100.

Size: 1 byte dec

webmogul1, 2023-01-13:
https://github.com/zwave-js/node-zwave-js/pull/5330

AlCalzone, 2023-01-14:
Let's keep this open until the PR is merged

closed 2023-01-26 by AlCalzone