zwave-js/zwave-js #5328: Device Minoston MP21Z - Unable to set param 2 -

robertwynkoop, 2023-01-13:
Home Assistant reports "The parameter has been updated." However refresh, param is still 0.

Param 2 is: Auto-Off Timer

# log for the interaction
```
2023-01-13T16:30:51.819Z SERIAL » 0x011d001303169f03e50094b8302c49cda9bdffaa5772967d795c1a73250bc4    (31 bytes)
2023-01-13T16:30:51.819Z DRIVER » [Node 003] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      11
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 229
                                    └─[SupervisionCCGet]
                                      │ session id:      54
                                      │ request updates: true
                                      └─[ConfigurationCCSet]
                                          parameter #:      2
                                          reset to default: false
                                          value size:       2
                                          value format:     UnsignedInteger
                                          value:            60
2023-01-13T16:30:51.823Z SERIAL « [ACK]                                                                   (0x06)
2023-01-13T16:30:51.828Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2023-01-13T16:30:51.829Z SERIAL » [ACK]                                                                   (0x06)
2023-01-13T16:30:51.829Z DRIVER « [RES] [SendData]
                                    was sent: true
2023-01-13T16:30:51.841Z SERIAL « 0x010500130b00e2                                                     (7 bytes)
2023-01-13T16:30:51.842Z SERIAL » [ACK]                                                                   (0x06)
2023-01-13T16:30:51.842Z DRIVER « [REQ] [SendData]
                                    callback id:     11
                                    transmit status: OK
2023-01-13T16:30:51.855Z SERIAL « 0x011700040003119f031d001824fa0144abd196cdb9e1311da9                (25 bytes)
2023-01-13T16:30:51.857Z SERIAL » [ACK]                                                                   (0x06)
2023-01-13T16:30:51.858Z DRIVER « [Node 003] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 29
                                    └─[SupervisionCCReport]
                                        session id:          54
                                        more updates follow: false
                                        status:              Fail
                                        duration:            0s
2023-01-13T16:30:52.872Z SERIAL » 0x01160013030f9f03e6002c38bb5771f76d098d1aaa250c82                  (24 bytes)
2023-01-13T16:30:52.873Z DRIVER » [Node 003] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      12
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 230
                                    └─[ConfigurationCCGet]
                                        parameter #: 2
2023-01-13T16:30:52.876Z SERIAL « [ACK]                                                                   (0x06)
2023-01-13T16:30:52.881Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2023-01-13T16:30:52.882Z SERIAL » [ACK]                                                                   (0x06)
2023-01-13T16:30:52.882Z DRIVER « [RES] [SendData]
                                    was sent: true
2023-01-13T16:30:52.894Z SERIAL « 0x010500130c00e5                                                     (7 bytes)
2023-01-13T16:30:52.895Z SERIAL » [ACK]                                                                   (0x06)
2023-01-13T16:30:52.895Z DRIVER « [REQ] [SendData]
                                    callback id:     12
                                    transmit status: OK
2023-01-13T16:30:52.906Z SERIAL « 0x011a00040003149f031e00e44ba98da0ccb907de3d7115e3811a13c1          (28 bytes)
2023-01-13T16:30:52.908Z SERIAL » [ACK]                                                                   (0x06)
2023-01-13T16:30:52.911Z CNTRLR   [Node 003] [~] [Configuration] 2: 0 => 0
2023-01-13T16:30:52.912Z DRIVER « [Node 003] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 30
                                    └─[ConfigurationCCReport]
                                        parameter #: 2
                                        value size:  4
                                        value:       0

```

# Full log
[zwave_current.log](https://github.com/zwave-js/node-zwave-js/files/10413429/zwave_current.log)

AlCalzone, 2023-01-30:
Looks like the device doesn't implement Supervision properly. I'll disable this CC for the MP21Z.

Aside from that, HA doesn't seem to tell you when a supervised Configuration Set failed, I've raised a core issue for that.

closed 2023-01-30 by AlCalzone

AlCalzone, 2023-07-24:
Turns out we had an incorrect config file for that device - the value size was wrong, so the device was correct in not accepting the change:
https://github.com/zwave-js/node-zwave-js/pull/6056

PR #5377 (2023-01-30): fix(config): disable Supervision for Minoston MP21Z
fixes: #5328