zwave-js/zwave-js #5334: Zooz Zen77 state not updated when toggled from physically connected 3-way switch on firmware 10.30.2

pletch, 2023-01-14:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Zen77 devices upgraded to most recent Zooz Firmware (10.30.2) do not correctly update state on controller when on/off state changed via physically connected 3-way switch (as observed in zwave-js-ui).  Downgrading the Zen 77 firmware to the prior version (10.20.1) restores the expected behavior.  

I have two of these devices wired in a 3-way configuration and both exhibit the same behavior.

Contacted Zooz about issue and they indicated that the issue should be raised with zwave-js project.

### Device information

Manufacturer: Zooz
Model name: Zen77
Node ID in your network: 32

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [X] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js-ui: 8.6.2.faa2543
zwave-js: 10.3.1

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Upgraded Zen77 switches to latest available Zooz Firmware (10.30.2).

Change Notes for Zooz Firmware VER. 10.30 (released 10/2022 as OTA)
- Adjusted Direct Association behavior for Group 3 for full sync and smooth dimming when two or more Z-Wave dimmers are associated together in a virtual 3-way or 4-way set-up
- The value for the double tap setting now applies to devices associated with the Z-Wave dimmer in Groups 2-4
- The Night Light Mode setting now applies to devices associated with the Z-Wave dimmer in Groups 2-4
- Added parameter 7 to choose Direct Association reports sent to devices associated with the Z-Wave dimmer in Group 3. [See all settings here.](https://www.support.getzooz.com/kb/article/550-zen77-s2-dimmer-700-advanced-settings/)
- Added parameter 31 to enable central scene commands (scene control) from the mechanical switch connected to the Z-Wave dimmer in a direct 3-way set-up. [Seel all settings here.](https://www.support.getzooz.com/kb/article/550-zen77-s2-dimmer-700-advanced-settings/)

Might possibly be related to association reports and updates to parameter 7 in device config (advanced settings note these are for firmware 10.20 but I believe this is an error and the advanced settings listed are actually for 10.30).  Change notes above imply this is only for group 3 devices but the Advanced Settings summary implies this is for any associated devices.

**Association Reports**
Parameter 7: Choose which physical and Z-Wave triggers should prompt the dimmer to send a status change report to associated devices.

Values: 0 – none; 1 – physical tap on ZEN77 only; 2 – physical tap on connected 3-way switch only; 3 – physical tap on ZEN77 or connected 3-way switch; 4 – Z-Wave command from hub; 5 – physical tap on ZEN77 or Z-Wave command from hub; 6 – physical tap on connected 3-way switch or Z-Wave command from hub; 7 – physical tap on ZEN77 / connected 3-way switch or Z-Wave command from hub; 8 – timer only; 9 – physical tap on ZEN77 or timer; 10 – physical tap on connected 3-way switch or timer; 11 – physical tap on ZEN77 / connected 3-way switch or timer; 12 – Z-Wave command from hub or timer; 13 – physical tap on ZEN77, Z-Wave command from hub, or timer; 14 – physical tap on ZEN77 / connected 3-way switch, Z-Wave command from hub, or timer; 15 – all of the above. Default: 15.
Size: 1 byte dec

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Yes, the expected behavior occurs on Zen77 firmware 10.20.

### Attach Driver Logfile

- Performed Interview of Node 32.

- At end, set switch state to off (Target Value = 0) via zwave-js-ui.    

- Physically switched state via wired 3-way switch.  Light is on in field but zwave-js-ui does not indicate new current value.

[zwavejs_2023-01-14.log](https://github.com/zwave-js/node-zwave-js/files/10418677/zwavejs_2023-01-14.log)

AlCalzone, 2023-01-30:
AFAIK parameter 7 is only meant for the devices in group 2/3, not the lifeline.

Unless there was a command lost/dropped (there seems to be some filtering going on in the log, you may want to redo the toggle test without any log filters to be sure), I would assume this to be a bug in the device.

pletch, 2023-02-04:
Thank you for looking at this.  My interpretation looking at the initial log is that there is no status update sent from the switch when toggled via the connected 3-way switch.  

Here is another log with no node filtering:
2023-02-04T14:25:47     Zen77 Node 32 switched on as part of central scene.
2023-02-04T14:25:54     Zen77 Node 32 switched off via physical switch on Zen77 device
        A few seconds later I walked over and switched the light back on via the connected 3-way switch.  Zwave-JS UI still shows the light as off.  The log shows no indication that status update command was sent.
2023-02-04T14:27:48     Zen77 Node 32 commanded on to 75% via Zwave JS UI.

If my interpretation is indeed correct, I will loop back with Zooz and tell them this appears to be a definite bug in their latest firmware for this device.

[zwavejs_2023-02-04.log](https://github.com/zwave-js/node-zwave-js/files/10608895/zwavejs_2023-02-04.log)

pletch, 2023-02-19:
I reached back out to Zooz and their support explained that there was an error with the version of the 10.30 firmware that was initially uploaded to the server at the time I performed the update.  I was instructed to re-download the currently posted version of 10.30.2 and reflash.  After doing so, I reinterviewed and this did indeed fix the issue.

pletch, 2023-02-19:
This was related to firmware bug in the initially posted Zen77 firmware version 10.30.2

closed 2023-02-19 by pletch