zwave-js/zwave-js #5335: Second Thermostat Setpoint not being found for Intermatic PE653 z-wave pool controller

bluemantwo, 2023-01-16:
I have an Intermatic PE653 pool controller.  Product Manufacturer: 0x0005, Product Identifiers: 0x5045:0x0653 

I had it included on my ISY944i and was able to access both Thermostat setpoints: one for the Pool which was type Heating and one for the spa which was type Cooling (though Intermatic says it should be Type Furnace).  When I included the device in my Home Assistant zwave-js, it only finds the 1 setpoint for Heating.  Attached is the log created by doing a re-interview of the device (Device ID 19).  Inclusion starts at 1:45:36 Zulu in the log.  It seem to not make any reference to the other setpoint.  Any ideas?  My PE653 is running firmware v39.
[zwavejs_2023-01-15.log](https://github.com/zwave-js/node-zwave-js/files/10422003/zwavejs_2023-01-15.log)

AlCalzone, 2023-01-30:
I'm afraid they do in fact not implement the Command Class specification correctly. Granted, there was so much confusion around the setpoints in earlier versions that the specifications explicitly call this out and mention how to deal with the two possible interpretations, but the information returned by the device matches neither.

[image]

In this table, `1.2` means first byte (1-based), bit 2 (0-based). The device returns a value of `0x43` for the first byte, which is `0100 0011` in binary. Acoording to the table, this corresponds to either of these setpoints:
a) N/A, Heating, Auto Changeover (this is what zwave-js makes of this)
b) N/A, Heating, N/A

I think the return value is shifted by one bit from what it was supposed to be (`0x86 = 1000 0110`), which would correspond to these setpoints:
- Heating, Cooling, Furnace.

bluemantwo, 2023-01-30:
@AlCalzone , thanks for the great and helpful reply.  The odd thing is that the ISY this device was connected to had no support at all for the PE653.  It just happened to work.  Also odd is that while Intermatic goes out of their way to say that setpoint #2 if Furnace, it showed up as Cooling in the ISY.  They definitely messed something up.  Your idea that they shifted the return value by 1 bit does seem very plausible.  

I am sort of new to this, but is there a way I can write a custom driver that would adjust for this? It is such a 'one off' product (doubt there are more than 1000 or so in the wild) that I might have to just do this one myself somehow (?)

bluemantwo, 2023-01-31:
@kpine You did it!!!!!  All I really need is the ability to set the Furnace Setpoint, and using the service call you listed, it worked!!!  I did have to use endpoint 0 (as opposed to 1), but it worked!!!!!

It would also be nice, though not critical, to be able to also read the setpoint value for Furnace (7).  Is there a way to set a variable to that value by doing a 'get' type of command?  

THANK YOU!!!!!!!

bluemantwo, 2023-01-31:
@kpine , you freaking genius!!!!!  I just realized that after doing the call to set and then get the value for the Setpoint that was not showing up, it now shows up with an entity id in Home Assistant!!!!!!!
[image]

bluemantwo, 2023-01-31:
I just did a power down of the entire system, rebooted, and the entity still is there and active.  I imagine that if I do a re-interview of the device then it will go away.  Just I can just keep a script to run a set/get command to get it back.  Easy enough.  Thanks again!!

AlCalzone, 2023-01-31:
> why don't the interview logs here show the setpoint interpretation selection since the CC is v1?

Probably because the endpoints are set up differently for Multi Channel CC v1 and that information is lost in translation.

bluemantwo, 2023-04-08:
@AlCalzone , I had a follow up question related to the Clock setting on my pool controller.  I am not able to read or set the clock for time of day.  It tends to gain about 30 seconds per day so much be reset regularly.  Right now I have to do it using the z-wave remote control that came with the system.  I would like to be able to set it automatically from Home Assistant.  The attached log shows that the Clock is discovered (command class 0x81/129) and during discovery it does seem to report its current time, but when I try to issue a clock set or get command, I see nothing in the logs in terms of response from the pool controller.  

This is the command I am sending it:

service: zwave_js.invoke_cc_api
data:
  method_name: get
  endpoint: "1"
  command_class: "129"
  parameters:
    - 1
target:
  device_id: dea02e849740113a6c5c426a9b4d5eb1

I was not sure what parameter to use and could not find any documentation on parameters with a Clock. I also tried endpoint 0 with no joy either.  Any thoughts on what else I might try?  
[Pool Node Interview Log.txt](https://github.com/zwave-js/node-zwave-js/files/11183623/Pool.Node.Interview.Log.txt)

bluemantwo, 2023-04-08:
Well, hell.  @AlCalzone , just putting my thoughts together to ask you the question allowed me to figure it out!  I decided rather than doing a get, I would do a set, with parameters hours and minutes.  And sure enough, it works!  It set the clock!  Sorry for bothering you, but thanks for being a sounding board to help me get my thoughts in order.  

Here is the command that works:

service: zwave_js.invoke_cc_api
data:
  method_name: set
  command_class: "129"
  parameters:
    - 10
    - 09
target:
  device_id: dea02e849740113a6c5c426a9b4d5eb1

That sets clock to 10:09am.  Yay!!!

bluemantwo, 2023-04-09:
Sorry, @kpine , I accidently tagged the above 2 items to AlCalzone.  I updated them to direct the to you.  Problem is now solved, but I wanted to make sure you knew I appreciated your earlier directions which helped me solve it!

AlCalzone, 2023-04-11:
maybe this api is simpler?
https://zwave-js.github.io/node-zwave-js/#/api/node?id=setdateandtime

AlCalzone, 2023-04-11:
That API is specifically built to support all possible time related CCs a device may use.

JDogg016, 2023-05-08:
Ok.  This thread is way beyond my intelligence level.  I have FW 3.4 installed and have 5 switches, an air temperature sensor and a number entity (basic) available.

I am looking to discover the thermostat entity and possibly control my variable speed pool pump and I find it very interesting that you are writing service calls into HA to control some entities that may not otherwise be exposed.

Any assistant you can lend would be great as I could then use Node Red to build automations.

AlCalzone, 2023-07-20:
@bluemantwo #6044 should fix this issue by overriding the setpoint query with the correct values. I'm unable to test though due to not having the device.
When 11.5.3 lands in the application you're using, can you re-interview the device and see if that fixes the setpoints?

bluemantwo, 2023-07-20:
I will do.  Thank you so much for the exceptional work you are doing for the community.

bluemantwo, 2023-07-23:
@AlCalzone , 11.5.3 did not seem to resolve the issue for the second thermostat setpoint.  To recap, there are 2 setptoints.  Setpoint 1 (heating, for pool) and setpoint 7 (furnace, for spa).  I did a re-interview (log attached) and it found Setpoint 1 as always, but not setpoint 7.  

However, since I can do a custom call to the specific address of Setpoint 7 to read and write the data, I do not consider this a serious issue.  OK to mark closed in my view.  And thank you so much for all you are doing for z-wave.  It is amazingly great now.  
[PE653 interview log.txt](https://github.com/zwave-js/node-zwave-js/files/12138007/PE653.interview.log.txt)

AlCalzone, 2023-07-24:
@bluemantwo unfortunately that log is of no help. I need a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) on loglevel **debug**.
I'd still like to fix this, since the device seems somewhat popular and reducing friction here is good.

bluemantwo, 2023-07-24:
Is there a more detailed log I can enable during interview?  Let me know how I can help. 

I own 2 of these devices.  One running the more prevalent 3.4 firmware, which is what most users are on, which I have just as a spare and do not use.  The other is running 3.9 firmware, which is a special firmware developed by Intermatic for property managers which has (supposedly) improved z-wave compliance.  Version 3.9 is the one I am using. 

The reason I am reluctant to have you or anyone spend much time on this device is that the volumes are low, and then there are 3 firmware versions used: 3.3 (20% of users), 3.4 (70% of users), and 3.9 (just a handful).   All those permutations and the fact that Intermatic did not follow z-wave standards at all, makes this device a PITA to support.  

As for the thermostats (Heater-1 and Furnace-7), I am not even sure they are handled the same way across firmwares.  

So these devices kind of belong in the category of "better than nothing" support.  ;)

bluemantwo, 2023-07-26:
@AlCalzone , here is the detailed log for the interview of the PE653.  Node 19.  
[PE653 interview (Node 19).txt](https://github.com/zwave-js/node-zwave-js/files/12176001/PE653.interview.Node.19.txt)

Thanks for looking into this.

AlCalzone, 2023-07-26:
Hmm I think this is a combination of a handful of semi-related issues:
1. I suspect that firmware 3.9 also only supports Multi Channel CC v1, but it reports support for V2 like firmware 3.4 where we forced the use of v1. As a result a lot of queries for the endpoints fail.
2. Your device seems to have bad connectivity. I see several communication attempts falling back to 9.6 kbit/s and many RSSI readings below -90 dBm which is terribly close to background noise.
3. Lots of responses to endpoint functionality queries are delayed and/or repeated (probably due to the above), and Z-Wave JS currently fails to check that the received one matches the queried one, so there is a bit of chaos in the interview (which also fails a couple of times).

I suspect these problems are also the reason why the compat flag does not get applied. We shouldn't actually see the thermostat setpoint query go out, but it does.

You'll need to look into 2., I'll have to fix 1. and 3. before we can retry.

bluemantwo, 2023-07-26:
Yes, 2. is probably correct and understandable.  It is quite a distance from my indoor z-wave controller and the outdoor pool house where the pool controller is.  Pool controller is also one of the few 300 series devices left on my network, if that matters.  I have a z-wave repeater mid way between the z-wave controller and pool controller, but for some reason it always wants to go direct.

AlCalzone, 2023-07-26:
> but for some reason it always wants to go direct

The latest Z-Wave JS UI lets you configure priority (return) routes which are always attempted first.

AlCalzone, 2023-08-17:
@bluemantwo did you manage to improve the connectivity? Problems 1 and 3 above are fixed, so we should be able to continue.

bluemantwo, 2023-08-17:
@AlCalzone , do you recall which release has the latest fix for this?  I will make sure I am updated and try today.  Thanks as always.

AlCalzone, 2023-08-17:
Driver version `11.7.0`, which should be included in all latest addon/application versions

bluemantwo, 2023-08-24:
@AlCalzone , still no luck finding the second thermostat setpoint.  Currently running 11.13.0.  Log attached.  
[Node 19 Thermostat.log](https://github.com/zwave-js/node-zwave-js/files/12434128/Node.19.Thermostat.log)

AlCalzone, 2023-08-25:
I think this is the problem:
> 2023-08-24T23:34:59.165Z CNTRLR   [Node 019] User-provided device config loaded

You still have an old custom config file which gets preferred over the one we're trying to test.

bluemantwo, 2023-08-25:
DOH!!!!!!  Let me remove that and give it a try.  Maybe the problem for the past several attempts.

bluemantwo, 2023-08-25:
@AlCalzone, I am an idiot.  It works perfectly.  I had written my own custom parameter definitions for firmware v39 which is why I had a custom file.   I have to still look, but it appears the new version is giving me all of the custom parameters I needed (and more!).   This broke some of my old settings and automations, but I will fix those to adjust to your new config file.  

THANKS so much for your work on this.

and SORRY for forgetting I was using a custom config file.  Really sorry for that as it probably wasted your time on some of these issues. 

 YAY!!!!!!!!

bluemantwo, 2023-08-25:
@AlCalzone , just another follow up.  Everything is working 100% for me on firmware v3.9.  I can finally drop that custom config file I had been using.  This is just perfect!!!

bluemantwo, 2023-08-26:
@AlCalzone ,

Still one issue with the thermostat after all, which I did not realize until now.  While both the setpoints now show up, the 7-furnace setpoint is not active.  I did not realize because I use a script that sends commands directly to it which works fine.  But the standard interface does not work.  I included new log, which includes this comment at the end:

2023-08-26T17:44:37.195Z CNTRLR   [Node 019] Interviewing Thermostat Setpoint...
2023-08-26T17:44:37.195Z CNTRLR » [Node 019] retrieving supported setpoint types...
2023-08-26T17:44:37.196Z CNTRLR   [Node 019] Thermostat Setpoint bitmap interpretation is unknown, assuming B fo
                                  r now
2023-08-26T17:44:37.196Z CNTRLR » [Node 019] querying current value of setpoint Heating...
2023-08-26T17:44:37.337Z CNTRLR « [Node 019] received current value of setpoint Heating: 39 °F
2023-08-26T17:44:37.337Z CNTRLR » [Node 019] querying current value of setpoint Furnace...
2023-08-26T17:44:37.478Z CNTRLR « [Node 019] received current value of setpoint Furnace: 39 °F

Sorry for not catching this before.  I am thrilled by the new parameters that are now included and the fact that I no longer need to use my custom config file.  I just lost site of the issue with the second thermostat setpoint.  
[PE653 Interview using 11.13.0.txt](https://github.com/zwave-js/node-zwave-js/files/12446497/PE653.Interview.using.11.13.0.txt)

bluemantwo, 2023-08-26:
More follow up.  Picture below shows that the settings in the JS-UI interface work perfectly for both pool (heater) and spa (furnace).  In the past, only pool was working.  I can set the spa setpoint, and it shows up in my pool system as having changed to the correct value.  I can also change the value from my pool system, and do a refresh in the js-ui and it reads correctly. 

However, the entity created for spa climate does not allow changing the spa setpoint.  Pool works fine.  
[image]

AlCalzone, 2023-08-27:
> However, the entity created for spa climate does not allow changing the spa setpoint. Pool works fine.

I'm a bit tired, so I'm not 100% sure I understood you:
- In Z-Wave JS UI, both setpoints work
- In Home Assistant itself, only one of the setpoint works

Correct?

bluemantwo, 2023-08-27:
Yes.  In the picture above, in the JS UI, I am able to read and write both setpoints.  Verified they work by using hand control (PE953) to confirm settings.  

But in HA, the SPA setting does not work.    See screen shot below.  Spa does not show setpoint value and when you try to change it give the error message shown at the bottom below.

[image]

AlCalzone, 2023-08-28:
@marcelveldt, this might be another case for you :)

bluemantwo, 2023-08-28:
I really appreciate the work of whoever created all the parameter values for the PE653.  I thought There were several that I never knew existed, and a couple I had guessed wrong on.  HUGE thanks to whoever did all the research and detective work to get those parameters.  My experience has been that Intermatic has been terrible at documenting their pool controller in the past.

AlCalzone, 2023-08-28:
That would be @philh30 in https://github.com/zwave-js/node-zwave-js/pull/5822
Sorry for the tag, but credit where credit is due :)

bluemantwo, 2023-08-28:
Thanks @philh30 !!!  You da man! 

And @AlCalzone ,  could the issue with my second setpoint be due to the endpoint being wrong?  This is the put put from diagnostic request done in the HA interface (not JS-UI):

note: Pool_3 with endpoint: 1 works, Pool_2 with endpoint: 0 does not.  

    "domain": "climate",
        "entity_id": "climate.pool_3",
        "original_name": "",
        "original_device_class": null,
        "disabled": false,
        "disabled_by": null,
        "hidden_by": null,
        "original_icon": null,
        "entity_category": null,
        "supported_features": 1,
        "unit_of_measurement": null,
        "value_id": "19-67-1-setpoint-1",
        "primary_value": {
          "command_class": 67,
          "command_class_name": "Thermostat Setpoint",
          "endpoint": 1,
          "property": "setpoint",
          "property_name": "setpoint",
          "property_key": 1,
          "property_key_name": "Heating"
      
      
      domain": "climate",
              "entity_id": "climate.pool_2",
              "original_name": "",
              "original_device_class": null,
              "disabled": false,
              "disabled_by": null,
              "hidden_by": null,
              "original_icon": null,
              "entity_category": null,
              "supported_features": 1,
              "unit_of_measurement": null,
              "value_id": "19-67-0-setpoint-7",
              "primary_value": {
                "command_class": 67,
                "command_class_name": "Thermostat Setpoint",
                "endpoint": 0,
                "property": "setpoint",
                "property_name": "setpoint",
                "property_key": 7,
          "property_key_name": "Furnace"

bluemantwo, 2023-08-28:
[zwave_js-3a152d2a5cd6e4dafe5146f53abf5151-Pool-dea02e849740113a6c5c426a9b4d5eb1.json (1).txt](https://github.com/zwave-js/node-zwave-js/files/12458155/zwave_js-3a152d2a5cd6e4dafe5146f53abf5151-Pool-dea02e849740113a6c5c426a9b4d5eb1.json.1.txt)

bluemantwo, 2023-08-29:
@kpine , thanks for the follow up.  As to the current temperatures, there is only 1 water temperature probe.   The probe is placed BEFORE the pool/spa diverter valve so that it always gives the temperature of the water that is flowing through the pump.  So that fact that both setpoints will use the same current temperature works perfectly.

AlCalzone, 2023-10-19:
Closing as this is solved

PR #6044 (2023-07-20): fix: override supported setpoint types for Intermatic PE653
fixes: #5335
closed 2023-10-19 by AlCalzone